vshymanskyy/TinyGSM #314: Help on running MqttClient example on wemos D1 mini

gusantor, 2019-08-01:
Hello and thanks !

## What type of issues is this?
[x] Bug or issue with library functionality (ie, sending data over TCP/IP)

### What are you working with?
Main processor board: ESP8266, on a wemod D1 mini
Modem: SIM800l
TinyGSM version: 0.7.9
Code: MqttClient Example, with Software serial Rx->D2, Tx->D1, Baudrate 9600

### Scenario, steps to reproduce
I'm trying to run the example, and stablish a connection with test.mosquitto.org
Just installed the example.
Enable TINY_GSM_MODEM_SIM800, 
Enable DUMP_AT_COMMANDS,
Define SerialAT as software serial and baudrate to 9600 in both serial comunication channels
Define apn credentials
Not using WiFi connection

### Expected result
I'd like to get connection to test.mosquitto.org

### Actual result
The script advance sort of well, get connection to network but get stuck at AT+CIPRXGET=4,1, giving ERROR for this command and others forward, like:
AT+CIPSTATUS=1
AT+CIPCLOSE=1,1
AT+CIPSTART=1,"TCP","test.mosquitto.org",1883

### AT command log
Just afte to reset wemos, I get :
Wait...
Initializing modem...
AT
AT
OK
AT+CLTS=1
AT+CLTS=1
OK
AT&W
AT&W
OK
AT+CFUN=0
AT+CFUN=0
+CPIN: NOT READY

OK
AT+CFUN=1,1
AT+CFUN=1,1
OK
[11511] ### TinyGSM Version: 0.7.9
AT
�[11718] ### Unhandled: �
AT
AT
OK
AT&FZ
AT&FZ
OK
ATE0
ATE0
OK
AT+GMM

SIMCOM_SIM800L

OK
[11923] ### Modem: SIMCOM SIM800L
[11923] ### Modem: SIMCOM SIM800L
AT+CPIN?

ERROR
AT+CPIN?

+CPIN: READY

OK
ATI

SIM800 R14.18

OK
Modem: SIM800 R14.18
Waiting for network...AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,1

OK
 OK
AT+CREG?

+CREG: 0,1

OK
Network connected
AT+CIPRXGET=4,1

ERROR
[18902] ### Available: 0 on 1
AT+CIPSTATUS=1

ERROR
=== MQTT NOT CONNECTED ===
Connecting to test.mosquitto.orgAT+CIPRXGET=4,1

ERROR
[20972] ### Available: 0 on 1
AT+CIPSTATUS=1

ERROR
AT+CIPCLOSE=1,1

ERROR
AT+CIPSSL=0

OK
AT+CIPSTART=1,"TCP","test.mosquitto.org",1883

ERROR
 fail
AT+CIPRXGET=4,1

ERROR
[23272] ### Available: 0 on 1
AT+CIPSTATUS=1

ERROR

Please any advice on how to get the example to run is very appreciated

Thanks in advance

SRGDamia1, 2019-08-01:
Oops. There's an error in that example.

Comment out line 167 and 176 (https://github.com/vshymanskyy/TinyGSM/blob/bbc429da1a89fdffe9f78016824dc9610fa7e2dc/examples/MqttClient/MqttClient.ino#L167)

gusantor, 2019-08-01:
Thank you @SRGDamia1

After your comment I've tried that, but can't compile, I receive:

'gprsUser' was not declared in this scope

So, I've changed gprsUser by user and gprsPass by pass, resulting in line #170

if (!modem.gprsConnect(apn, user, pass)) {

After that, code compile and debugging messages changed, among others I received

AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK

and, even when I first receive some "=== MQTT NOT CONNECTED ===", later I can see

AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","MY.GIVEN.IP.HERE","1883","CONNECTED"

OK

So, great !!, it seem to be connected, I'll now move on to connect, publish and subscribe to my own mqtt server , THANKS !!!

closed 2019-08-01 by gusantor