vshymanskyy/TinyGSM #315: Problem with TinyGsmClient when there's no connection

tkralj, 2019-08-01:
Hello all, i need some assistance from you smart people :). I am having a trouble when trying to send http request and there's no gprs connection. modem.isGprsConnected() returns true but i have deliberately disconnected connection on the gsm operators portal so i know there's no connection even though method returns true. What i want to achieve is to automatically reconnect to the gprs in the case of connection drop for what ever reason it may be. Right now there's no way i can find out that there's no connection prior to calling http and receive a timeout. i want to establish a connection prior to http call if it is possible. What do you think would it be ok solution if i always connect to gprs prior to call and disconnect from it after the call. would it be too much of overhead ?

SRGDamia1, 2019-08-02:
What modem?

tkralj, 2019-08-02:
hello sim800l

SRGDamia1, 2019-08-02:
I'm not really sure how to help you.  If ```isGprsConnected()``` returns true, then the SIM800 thinks it is still attached to GPRS and still has an ip address.  I don't know how to "tell" it otherwise.

Could you post an AT log showing what's happening?

tkralj, 2019-08-02:
hi. that is exactly the problem. sim thinks its connected but it isn't (isNetworkConnected behaves the same) and if you issue http request in this state everything freezes until the timeout . is there any reliable method of checking sim connectivity ? i would like to test it before issuing http requests so i can reconnect in advance if necessary.

SRGDamia1, 2019-08-02:
I don't know of any way.

marcosdepo, 2020-01-24:
try to send AT+CIPSTATUS to gsm module, if its disconnected will response with ERROR or IP INITIAL

kaustubh-ke, 2023-01-10:
@marcosdepo  same I got   # Unhandled: STATE: IP INITIAL , PLEASE help me to solve.