vshymanskyy/TinyGSM #316: Timeout for reading data from stream should abort read operation after first timeout (?)

sabbay, 2019-08-09:
Hello,
While testing my solution I came across weird situation. While interrupting SIM800L <-> ESP32 communication I noticed that stream`s **_timeout** only skips the character that is being currently read, then tries to read another character in function **modemRead**.

My quickfix is:
```c++
  size_t modemRead(size_t size, uint8_t mux) {
#ifdef TINY_GSM_USE_HEX
    sendAT(GF("+CIPRXGET=3,"), mux, ',', size);
    if (waitResponse(GF("+CIPRXGET:")) != 1) {
      return 0;
    }
#else
    sendAT(GF("+CIPRXGET=2,"), mux, ',', size);
    if (waitResponse(GF("+CIPRXGET:")) != 1) {
      return 0;
    }
#endif
    streamSkipUntil(','); // Skip Rx mode 2/normal or 3/HEX
    streamSkipUntil(','); // Skip mux
    size_t len_requested = stream.readStringUntil(',').toInt();
    //  ^^ Requested number of data bytes (1-1460 bytes)to be read
    size_t len_confirmed = stream.readStringUntil('\n').toInt();
    // ^^ Confirmed number of data bytes to be read, which may be less than requested.
    // 0 indicates that no data can be read.
    // This is actually be the number of bytes that will be remaining after the read
    for (size_t i=0; i<len_requested; i++) {
      uint32_t startMillis = millis();
#ifdef TINY_GSM_USE_HEX
      while (stream.available() < 2 && (millis() - startMillis < sockets[mux]->_timeout)) {
        DBG("### 1");
        TINY_GSM_YIELD();
      }
      char buf[4] = { 0, };
      buf[0] = stream.read();
      buf[1] = stream.read();
      char c = strtol(buf, NULL, 16);
#else
      while (!stream.available() && (millis() - startMillis < sockets[mux]->_timeout)) {
        TINY_GSM_YIELD();
      }
      if(millis() - startMillis > sockets[mux]->_timeout){
        DBG("TIMEOUT, ABORTING READ.");
        break;
      }
      char c = stream.read();
#endif
      sockets[mux]->rx.put(c);
    }
    // sockets[mux]->sock_available = modemGetAvailable(mux);
    sockets[mux]->sock_available = len_confirmed;
    waitResponse();
    return len_requested;
  }
```

Was skipping byte by byte intended? It seems weird to me.

SRGDamia1, 2019-08-14:
The stream ```_timeout``` value usually is per-character.

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1