vshymanskyy/TinyGSM #318: MQTT not connected- MQTT example

mbc99, 2019-08-12:
## What type of issues is this?
I'm trying to get the MQTT example working.
Before attempting that I ran the diagnostics example. No problem there as you can see:
```
Wait...
Initializing modem...[11076] ### TinyGSM Version: 0.7.9
[11131] ### Modem: AT+GMM

SIMCOM SIM808
[11132] ### Modem: AT+GMM

SIMCOM SIM808
 [OK]
Modem: ATI  SIM808 R14.18
Waiting for network... [OK]
Connecting to telefonica.es [OK]
Local IP: _deleted for security_
Connecting to vsh.pp.ua [OK]
[25035] ### Available: 0 on 1
.....[25631] ### Available: 1 on 1

[25665] ### READ: 1 from 1
[25964] ### READ: 511 from 1
[27044] ### Available: 121 on 1
[27142] ### READ: 121 from 1
[27663] ### Available: 0 on 1
Server disconnected
GPRS disconnected

************************
 Received: 121 bytes
 Test:     PASSED
************************

```

### What are you working with?

Main processor board: Arduino Mega
Modem: SIM808
TinyGSM version: 0.9.7
Code: 
```
/**************************************************************
 *
 * For this example, you need to install PubSubClient library:
 *   https://github.com/knolleary/pubsubclient
 *   or from http://librarymanager/all#PubSubClient
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 * For more MQTT examples, see PubSubClient library
 *
 **************************************************************
 * Use Mosquitto client tools to work with MQTT
 *   Ubuntu/Linux: sudo apt-get install mosquitto-clients
 *   Windows:      https://mosquitto.org/download/
 *
 * Subscribe for messages:
 *   mosquitto_sub -h test.mosquitto.org -t GsmClientTest/init -t GsmClientTest/ledStatus -q 1
 * Toggle led:
 *   mosquitto_pub -h test.mosquitto.org -t GsmClientTest/led -q 1 -m "toggle"
 *
 * You can use Node-RED for wiring together MQTT-enabled devices
 *   https://nodered.org/
 * Also, take a look at these additional Node-RED modules:
 *   node-red-contrib-blynk-ws
 *   node-red-dashboard
 *
 **************************************************************/

// Select your modem:
//#define TINY_GSM_MODEM_SIM800
 #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

// See all AT commands, if wanted
 //#define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// Range to attempt to autobaud
#define GSM_AUTOBAUD_MIN 9600
#define GSM_AUTOBAUD_MAX 38400

// Add a reception delay, if needed
#define TINY_GSM_YIELD() { delay(2); }

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(7, 8); // RX, TX

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "telefonica.es";
const char user[] = "telefonica";
const char pass[] = "telefonica";

// MQTT details
const char* broker = "test.mosquitto.org";

const char* topicLed = "GsmClientTest/led";
const char* topicInit = "GsmClientTest/init";
const char* topicLedStatus = "GsmClientTest/ledStatus";

#include <TinyGsmClient.h>
#include <PubSubClient.h>

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else

TinyGsm modem(SerialAT);
#endif
TinyGsmClient client(modem);
PubSubClient mqtt(client);

#define LED_PIN 13
int ledStatus = LOW;

long lastReconnectAttempt = 0;

void setup() {

  // Set console baud rate
  SerialMon.begin(115200);
  delay(10);

  // Set your reset, enable, power pins here
  pinMode(LED_PIN, OUTPUT);

  SerialMon.println("Wait...");

  // Set GSM module baud rate
  SerialAT.begin(19200);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  modem.restart();
  // modem.init();

  String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem: ");
  SerialMon.println(modemInfo);

  // Unlock your SIM card with a PIN
  //modem.simUnlock("1234");

#if TINY_GSM_USE_WIFI
  SerialMon.print(F("Setting SSID/password..."));
  if (!modem.networkConnect(wifiSSID, wifiPass)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");
#endif

#if TINY_GSM_USE_GPRS && defined TINY_GSM_MODEM_XBEE
  // The XBee must run the gprsConnect function BEFORE waiting for network!
  modem.gprsConnect(apn, gprsUser, gprsPass);
#endif

  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork(240000L)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  if (modem.isNetworkConnected()) {
    SerialMon.println("Network connected");
  }

#if TINY_GSM_USE_GPRS && defined TINY_GSM_MODEM_HAS_GPRS
    SerialMon.print(F("Connecting to "));
  SerialMon.print(apn);
    if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    SerialMon.println(" fail");
      delay(10000);
      return;
  }
  SerialMon.println(" OK");
#endif

  // MQTT Broker setup
  mqtt.setServer(broker, 1883);
  mqtt.setCallback(mqttCallback);
}

boolean mqttConnect() {
  SerialMon.print("Connecting to ");
  SerialMon.print(broker);

  // Connect to MQTT Broker
  boolean status = mqtt.connect("GsmClientTest");

  // Or, if you want to authenticate MQTT:
  //boolean status = mqtt.connect("GsmClientName", "mqtt_user", "mqtt_pass");

  if (status == false) {
    SerialMon.println(" fail");
    return false;
  }
  SerialMon.println(" OK");
  mqtt.publish(topicInit, "GsmClientTest started");
  mqtt.subscribe(topicLed);
  return mqtt.connected();
}

void loop() {

  if (!mqtt.connected()) {
    SerialMon.println("=== MQTT NOT CONNECTED ===");
    // Reconnect every 10 seconds
    unsigned long t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) {
        lastReconnectAttempt = 0;
      }
    }
    delay(100);
    return;
  }

  mqtt.loop();
}

void mqttCallback(char* topic, byte* payload, unsigned int len) {
  SerialMon.print("Message arrived [");
  SerialMon.print(topic);
  SerialMon.print("]: ");
  SerialMon.write(payload, len);
  SerialMon.println();

  // Only proceed if incoming message's topic matches
  if (String(topic) == topicLed) {
    ledStatus = !ledStatus;
    digitalWrite(LED_PIN, ledStatus);
    mqtt.publish(topicLedStatus, ledStatus ? "1" : "0");
  }
}
```
### Scenario, steps to reproduce
Impossible to establish a connection with the MQTT server

### Expected result

### Actual result
This is the output of the serial with the MQTT example:
```
Wait...
Initializing modem...
[6744] ### TinyGSM Version: 0.7.9
[6801] ### Modem: AT+GMM

SIMCOM SIM808
[6802] ### Modem: AT+GMM

SIMCOM SIM808
Modem: ATI  SIM808 R14.18
Waiting for network... OK
Network connected
[14365] ### Available: 0 on 1
[15381] ### Unhandled: =1

ERROR
=== MQTT NOT CONNECTED ===
Connecting to test.mosquitto.org[16405] ### Available: 0 on 1
[17420] ### Unhandled: =1

ERROR
 fail
[18625] ### Available: 0 on 1
[19641] ### Unhandled: =1

ERROR
=== MQTT NOT CONNECTED ===
[20763] ### Available: 0 on 1
[21779] ### Unhandled: =1

```

### AT command log
This is the result of the MQTT example but with the DUMP_AT_COMMANDS enabled:

```
Wait...
Initializing modem...
[6342] ### TinyGSM Version: 0.7.9
[6400] ### Modem: AT+GMM

SIMCOM SIM808
[6401] ### Modem: AT+GMM

SIMCOM SIM808
Modem: ATI  SIM808 R14.18
Waiting for network... OK
Network connected
[13418] ### Available: 0 on 1
[14434] ### Unhandled: =1

ERROR
=== MQTT NOT CONNECTED ===
Connecting to test.mosquitto.org[15457] ### Available: 0 on 1
[16473] ### Unhandled: =1

ERROR
 fail
[17678] ### Available: 0 on 1
[18695] ### Unhandled: =1

ERROR
=== MQTT NOT CONNECTED ===
[19816] ### Available: 0 on 1
[20831] ### Unhandled: =1

ERROR
=== MQTT NOT CONNECTED ===
[21952] ### Available: 0 on 1
[22967] ### Unhandled: =1

ERROR
=== MQTT NOT CONNECTED ===
[24089] ### Available: 0 on 1
[25103] ### Unhandled: =1

ERROR
=== MQTT NOT CONNECTED ===
Connecting to test.mosquitto.org[26125] ### Available: 0 on 1
[27140] ### Unhandled: =1

```

SRGDamia1, 2019-08-12:
You don't have any AT commands in your log.  Did you maybe paste in the same thing twice.  It's really hard to see what's happening without that.

mbc99, 2019-08-12:
You are totally right, I didn't pase correctly.
I've edited my original message to avoid confussion, I think that all the AT commands are showed

mbc99, 2019-08-14:
Looking at the AT+CREG comand it seems like the first number of the response indicates the registration status. 0 is for not registered; could be this the problem?

mbc99, 2019-08-14:
To see if it was MQTT's faul I tested the HTTP test with similar results.
``Wait...
Initializing modem...
AT
AT
AT

OK
AT+CLTS=1
AT+CLTS=1
AT+CLTS=1

OK
AT&W
AT&W
AT&W

OK
AT+CFUN=0
AT+CFUN=0
AT+CFUN=0

+CPIN: NOT READY

OK
AT+CFUN=1,1
AT+CFUN=1,1
AT+CFUN=1,1

OK
AT
AT
AT

OK
AT&FZ
AT&FZ
AT&FZ

OK
ATE0
ATE0
ATE0

OK
AT+CPIN?
AT+CPIN?

ERROR
AT+CPIN?

RDY

+CFUN: 1

+CPIN: SIM PIN
AT+CPIN?

+CPIN: SIM PIN

OK
ATI
ATI

SIM808 R14.18

OK
Modem: ATI  SIM808 R14.18
Waiting for network...AT+CREG?
AT+CREG?
[... 1595 lines cut ...]
AT+CREG?
AT+CREG?

+CREG: 0,0

OK
AT+CREG?
AT+CREG?

+CREG: 0,0

OK
AT+CREG?
AT+CREG?

+CREG: 0,0

OK
AT+CREG?
AT+CREG?

+CREG: 0,0
``

Note that just before running this sketch I successfully ran the diagnostics test. And if I recall corretly this test uses an internet connection; so I don't understand why on the other tests it can't connect to the internet

SRGDamia1, 2019-08-14:
It looks like you're not being able to register on the network.  Are you sure your SIM is ok?  Can you get registered on the network outside of this library?

SRGDamia1, 2019-08-14:
When you modified the MQTT example as you posted above, you removed the line ```#define TINY_GSM_USE_GPRS true``` (line 75ish in the example).  Without that define, the code wasn't running the function to connect to GPRS, which is why your connection failed.  It took me a few passes over your code to catch that.

In the most recent post, as you saw, you're not successfully registering on the network.  That part should happen automatically even before connecting to GPRS.  It did happen in the output you posted from the MQTT test (you got a ```CREG = 0,1``` in that).  I don't know why it would register on the network in one test and not in another.  Are you in a poor service area?  Are your power and antenna connections solid?  

Your post from the HTTP example didn't look suspicious (other than not connecting) but the MQTT output you posted has something odd going on with your module.  Even though the echo should be turned off, it was echoing all commands.  Some of them are echoed multiple times.

mbc99, 2019-08-14:
@SRGDamia1 I see that you have deleted your message. Is this because I can register to the network?
I will try with another library (and another SIM card) but I supposed that if the diagnostics test was able to download something that my hardware was fine

mbc99, 2019-08-14:
> When you modified the MQTT example as you posted above, you removed the line `#define TINY_GSM_USE_GPRS true` (line 75ish in the example). Without that define, the code wasn't running the function [...]

It's not that I deleted. This line doesn't appear on the MQTT example. In Arduino IDE I've openend that example sketch and using Control+F didn't show that line. Furthermore, adding this line to the code throws this error:
```
src\main.cpp:169:33: error: 'gprsUser' was not declared in this scope
src\main.cpp:169:43: error: 'gprsPass' was not declared in this scope
```
It's refering to this lines (which makes sense now that GPRS is set to true):
```
#if TINY_GSM_USE_GPRS && defined TINY_GSM_MODEM_HAS_GPRS
    SerialMon.print(F("Connecting to "));
  SerialMon.print(apn);
    if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    SerialMon.println(" fail");
      delay(10000);
      return;
  }
  SerialMon.println(" OK");
#endif
```

As for the signal I'm trying this in a populated area. Running the all_functions.ino gives a signal quality of 7

The antena is attached via a SMA connector and I'm using a 5V 2A power supply (the module I'm using has voltage regulator).

By the way, the all_functions example does say I'm connected via GPRS and gives me an IP. However, the CCID and IMEI are not provided by the serial (don't know if it's normal)

mbc99, 2019-08-14:
It's working. The error was simply corrected by changing the variable names of user to gprsUser and pass to gprsPass.
However, I didn't change that so I supose the example is incorrect

SRGDamia1, 2019-08-14:
What?

But that line *is* there: https://github.com/vshymanskyy/TinyGSM/blob/f0dd33ffe32ae46b9378efa02112e7d75e0739d1/examples/MqttClient/MqttClient.ino#L75

And the variable names *are* correct: https://github.com/vshymanskyy/TinyGSM/blob/f0dd33ffe32ae46b9378efa02112e7d75e0739d1/examples/MqttClient/MqttClient.ino#L84

mbc99, 2019-08-14:
> What?
>
> But that line _is_ there:
>
> https://github.com/vshymanskyy/TinyGSM/blob/f0dd33ffe32ae46b9378efa02112e7d75e0739d1/examples/MqttClient/MqttClient.ino#L75
>
> And the variable names _are_ correct: [...]

Wow, sorry. My library isn't like that but I supose (as it was updated 13 days ago) that I don't have the library updated.

Edit: yes, I can confirm that my library wasn't updated. 

Thank you very much

SRGDamia1, 2019-08-14:
Your print out shows you have TinyGSM version 0.9.7.  When did you download the examples?

SRGDamia1, 2019-08-14:
You know.. I think the Arduino IDE pulls from the "Releases" and there hasn't been a release in a while.  The example *was* wrong in the last release.  I seems like the IDE pulled the most recent library version, but not the most current examples.  I will put out a new release so hopefully this doesn't happen again.

mbc99, 2019-08-14:
> Your print out shows you have TinyGSM version 0.9.7. When did you download the examples?

I would say a month ago. But I downloaded from the Arduino IDE. 
For the next time I will always download libraries from github directly just in case :)

> You know.. I think the Arduino IDE pulls from the "Releases" and there hasn't been a release in a while. The example _was_ wrong in the last release. I seems like the IDE pulled the most recent [...]
Great!!

One last question to make sure it's working. Using the test.mosquitto.org broker should I expect callbacks? Bceause this is what I recieve over and over with the MQTT example:
```
Wait...
Initializing modem...
[6343] ### TinyGSM Version: 0.9.7
[6401] ### Modem: AT+GMM

SIMCOM SIM808
[6402] ### Modem: AT+GMM

SIMCOM SIM808
Modem: ATI  SIM808 R14.18
Waiting for network... OK
Network connected
Connecting to telefonica.es OK
[19704] ### Available: 0 on 1
=== MQTT NOT CONNECTED ===
Connecting to test.mosquitto.org[20515] ### Available: 0 on 1
[20719] ### Got Data: 1
[20756] ### Available: 4 on 1
[20789] ### READ: 4 from 1
 OK
[21056] ### Available: 0 on 1
[21154] ### Got Data: 1
[21193] ### Available: 5 on 1
[21230] ### READ: 5 from 1
[21556] ### Available: 0 on 1
[22057] ### Available: 0 on 1
[22560] ### Available: 0 on 1
[23062] ### Available: 0 on 1
[23564] ### Available: 0 on 1
[24065] ### Available: 0 on 1
[24566] ### Available: 0 on 1
[25068] ### Available: 0 on 1
[25570] ### Available: 0 on 1
[26072] ### Available: 0 on 1
[26572] ### Available: 0 on 1
[27078] ### Available: 0 on 1
[27578] ### Available: 0 on 1
[28077] ### Available: 0 on 1
[28580] ### Available: 0 on 1
[29082] ### Available: 0 on 1
[29584] ### Available: 0 on 1
[30083] ### Available: 0 on 1
[30586] ### Available: 0 on 1
[31088] ### Available: 0 on 1
[31589] ### Available: 0 on 1
[32091] ### Available: 0 on 1
[32591] ### Available: 0 on 1
[33095] ### Available: 0 on 1
[33597] ### Available: 0 on 1
[34098] ### Available: 0 on 1
[34599] ### Available: 0 on 1
[35099] ### Available: 0 on 1
[35602] ### Available: 0 on 1
[36104] ### Available: 0 on 1
[36581] ### Got Data: 1
[36622] ### Available: 2 on 1
[36658] ### READ: 2 from 1
[37121] ### Available: 0 on 1
```

SRGDamia1, 2019-08-14:
You'll only activate the call-back if you use another tool to publish a 0 or 1 to GsmClientTest/led, then you should see the LED change and the a new message published with the current status at GsmClientTest/ledStatus.

I've also had much better luck with broker.hivemq.com than with test.mosquitto.org.  If you use broker.hivemq.com you can use their web tool http://www.hivemq.com/demos/websocket-client/ to send the messages to your board and see the call back.

SRGDamia1, 2019-08-14:
Also, if everything is working, could you close this issue?

closed 2019-08-14 by mbc99