vshymanskyy/TinyGSM #323: get Time

jimmisitepu88, 2019-08-28:
hii TinyGSM

I use sim5320, and try to get time, 
String gsmTime = modem.getGSMDateTime(DATE_TIME);
String gsmDate = modem.getGSMDateTime(DATE_DATE);

and when i compile, I get error like this
HttpClient:214:28: error: 'TinyGsm {aka class TinyGsmSim5360}' has no member named 'getGSMDateTime'

     String gsmTime = modem.getGSMDateTime(DATE_TIME);

                            ^

HttpClient:216:28: error: 'TinyGsm {aka class TinyGsmSim5360}' has no member named 'getGSMDateTime'

     String gsmDate = modem.getGSMDateTime(DATE_DATE);

thanks

jimmisitepu88, 2019-09-04:
ini my library, we add this function

String sim5320::gTime(){
    String Twkt; int len; char buf[55];
    delay(500);
    sentAT("AT+CTZR=1", "OK", 1000);
    //wRespon(2000);
    delay(500);
    sentAT("AT+CTZU=1", "OK", 1000);
    //wRespon(2000);
    
    delay(1500);
    serialAT.println("AT+CCLK?");
    nMillis = millis();
    oMillis = nMillis;
    while(nMillis - oMillis < 3000){
        nMillis = millis();
        while (serialAT.available())
        {
            Twkt =  serialAT.readString();
        }  
    }
    len = Twkt.length();
    memset(buf, '\0', len);
    Twkt.toCharArray(buf, len);
    Twkt = "";
    for(int j = 21; j<len-11 ; j++){
        Twkt += String(buf[j]);
    }
    Twkt = String("20") + Twkt;
    //serialMON.println(Twkt);
    oMillis = nMillis;
    return Twkt;
}

SRGDamia1, 2019-09-04:
I'm sorry.  I don't have time do this.  If you want this feature included, please submit a clean pull request not a link to another library with the feature.

eabase, 2019-09-23:
If you look at my closed issue for SIM5360, you will find the fix there.

SRGDamia1, 2020-04-15:
This feature was added for the SIM5xxx in the 0.10 update.

closed 2020-04-15 by SRGDamia1