vshymanskyy/TinyGSM #324: Quectel M95 Error Connecting GPRS

perca23, 2019-08-29:
Hello!
I am trying to use Quectel M95 to connect GPRS.
I am using Diagnostics to understand the reason for not connecting.
The process usually occurs until the command QIMUX = 1, which always returns Error.
I have already reset the module to default as indicated, but the problem persists.
The Cell Chip works because I could test with SIM800L.
I am using with ESP32 as follows:
Hardware Serial SerialAT (1);
SerialAT.begin (19200, SERIAL_8N1, 5, 18, false);
Can someone help me?

[quectel_diagnostic.txt](https://github.com/vshymanskyy/TinyGSM/files/3556622/quectel_diagnostic.txt)

SRGDamia1, 2019-08-29:
Please, please, follow the issue template.

I don't have one of those units, so I can help some as I have time but I can't debug it myself.

Could you please try the version here:  https://github.com/EnviroDIY/TinyGSM/
Then re-post your AT log.

perca23, 2019-08-30:
Hello,
thanks for helping.
Still no solution.
Follows log AT.
[diagnostic_9.10.txt](https://github.com/vshymanskyy/TinyGSM/files/3559941/diagnostic_9.10.txt)

SRGDamia1, 2019-08-30:
I made some really minor changes to the ordering.  You can see if it makes a difference.  Can you try sending AT commands directly to the unit to see what order you can get it to accept the commands in?

perca23, 2019-09-02:
Answers to AT + QIMUX =? + QIMUX: (0.1)
For AT + QIMUX? + QIMUX: 1
When I change TinyGsmClientM95.h to
     sendAT (GF ("+ QICSGP = 1, \" "), apn, GF (" \ ", \" "), user, GF (" \ ", \" "), pwd, GF (" \ "") );
     if (waitResponse ()! = 1) {
       return false;
     }
     sendAT (GF ("+ QIMODE = 0"));
     if (waitResponse ()! = 1) {
       return false;
     }
         // Enable multiple TCP / IP connections
     sendAT (GF ("+ QIMUX = 1"));
     if (waitResponse ()! = 1) {
       return false;
     }

     // Start TCPIP Task and Set APN, User Name and Password
     // sendAT ("+ QIREGAPP = \" ", apn," \ ", \" ", user," \ ", \" ", pwd," \ "");
     sendAT ("+ QIREGAPP = \" ", apn," \ ", \" ", user," \ ", \" ", pwd," \ "");
     if (waitResponse ()! = 1) {
       return false;
     }
The command QIMODE = 0 and QIMUX = 1 return ok, but returns error in QIOPEN as per attachment.
[diagnostics-alter.txt](https://github.com/vshymanskyy/TinyGSM/files/3564098/diagnostics-alter.txt)

SRGDamia1, 2019-09-03:
Ugh.  There was a comma missing in the TCP open command.  Sorry.  I fixed it - 58ff6b42dddb450890bb48e6383000106d7ed4ea

perca23, 2019-09-03:
I made the change as you specified.
The connection was established, but soon after + QISACK returned error.
[Diagnostic-03.txt](https://github.com/vshymanskyy/TinyGSM/files/3572084/Diagnostic-03.txt)

SRGDamia1, 2019-09-04:
Well, all the examples show using this, but we can comment it out.  Also, I was looking for QIND instead of QINDI.  I changed both.  Please update to 0.9.13 and try again.

I'm sorry about all of the iteration.  I don't have this module myself so I hadn't tested it.

perca23, 2019-09-06:
Hello, there is no problem for me regarding interaction. Thank you for helping to solve the problem.
Using 0.9.13 I realized that there should be an error because QIMODE and QIMUX are after QIREGAPP which causes error in QIMODE.

Bringing them in before QIREGAPP the code continues to work until QIRDI cannot proceed.
[Diagnostics-05.txt](https://github.com/vshymanskyy/TinyGSM/files/3585163/Diagnostics-05.txt)

SRGDamia1, 2019-09-06:
I think I found the issue with QIRDI.  Could you try again:  c83063fa5dba1f5b75dc605c19ca215843d25f35

perca23, 2019-09-06:
Unfortunately the same situation persists.
+ QIRDI: 0.1.1
[24422] ### Got Date: 1
........................

SRGDamia1, 2019-09-06:
It just cuts out after saying "Got Data"?  Nothing else happens?  It locks up?  Or then errors start appearing?

perca23, 2019-09-06:
After "Got Data" does not lock, is printing "." "." On the same line
Only ends with a physical reset[24422] 

### Got Data: 1
......................................................................................................

SRGDamia1, 2019-09-06:
Oh.  Found the issue.  It registers that data has arrived but doesn't think anything is available.  Give me a few more minutes on it.

SRGDamia1, 2019-09-06:
The problem was that the module tells us data is available, but we have no way of knowing how much.  We can assume the buffer is full each time and then adjust downward if we can't really read that much data.

perca23, 2019-09-09:
Hello!
It really worked that way.
I just had to remove the new line in sendAT (GF ("+ QIOPEN ="), mux, GF (", \" "), GF (" TCP "), GF (" \ ", \" "), host, GF ("\", "), port);
When there is a new line after "host", QIOPEN returns error and the code does not proceed.
Thank you very much for the contribution.

SRGDamia1, 2019-09-09:
that's strange; I don't know why that line mattered.

Would you mind sharing the output from the successful run?

perca23, 2019-09-09:
Of course not.
I am using for MQTT connection and all ok.
Follows output of the diagnosis.
Thank you again!
[diagnostic-ok.txt](https://github.com/vshymanskyy/TinyGSM/files/3592797/diagnostic-ok.txt)

PR #327 (2019-09-09): Updates
M95 _confirmed_ working (closes #324)
Fix ESP8266 sock status
Lots of minor fixes to XBee's
Put sim unlock in all init's
Made it a little bit harder to choose the wrong options for the modems in the examples
closed 2019-09-09 by SRGDamia1