vshymanskyy/TinyGSM #330: Header values messed up with URL Query string parameters

tkralj, 2019-09-25:
Hello, i am using SIM800 on ESP32 with TinyGSM and ArduinoHttpClient to send GET and POST requests to my server. Everything works OK for a while (day or two) and after that suddenly my server receives incorrect header values. For example i have api-key header which is essentiali GUID of the device. And after a while my server receives, instead of GUID, date and time from URL query string. Is it possible problem with TinyGSM (fifo maybe) or should i look somewhere else.

Here is sample code:

  TinyGsmClient clientHttp(modem,1);
  HttpClient https(clientHttp, "http://myserver/webapi/", 80);

    https.beginRequest();
  
    https.post("TransNotify?tid=UUID&time=2019-09-2620%17:00:00&amount=0&rsnd=0&code=0");
    https.sendHeader("Content-Type", "application/x-www-form-urlencoded");
    https.sendHeader("Content-Length", 0);
    https.sendHeader("api-key", cfg.sn); 
    https.sendHeader("Authorization", AuthString);
   
    https.endRequest();   

  int statusCode = https.responseStatusCode();
  res = https.responseBody();

  https.stop();

As i said, this code runs correct for a while but after come time it messes up header values with data from URL. Can someone help??

SRGDamia1, 2019-09-25:
The FIFO is used for incoming data, not outgoing. There's no send buffer.

I don't have any ideas what would cause your issue. Sorry!

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1