vshymanskyy/TinyGSM #332: HttpClient specific url error

Ecancan, 2019-09-29:
I used the example directly and just changed the site address and this is the result? What is the cause of this problem?

Code : 
`/**************************************************************
 *
 * This sketch connects to a website and downloads a page.
 * It can be used to perform HTTP/RESTful API calls.
 *
 * For this example, you need to install ArduinoHttpClient library:
 *   https://github.com/arduino-libraries/ArduinoHttpClient
 *   or from http://librarymanager/all#ArduinoHttpClient
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 * For more HTTP API examples, see ArduinoHttpClient library
 *
 * NOTE: This example may NOT work with the XBee because the
 * HttpClient library does not empty to serial buffer fast enough
 * and the buffer overflow causes the HttpClient library to stall.
 * Boards with faster processors may work, 8MHz boards will not.
 **************************************************************/

// Select your modem:
#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
//#define SerialAT Serial1

// or Software Serial on Uno, Nano
#include <SoftwareSerial.h>
SoftwareSerial SerialAT(11, 12); // RX, TX

// Increase RX buffer to capture the entire response
// Chips without internal buffering (A6/A7, ESP8266, M590)
// need enough space in the buffer for the entire response
// else data will be lost (and the http library will fail).
#define TINY_GSM_RX_BUFFER 650

// See all AT commands, if wanted
//#define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon
//#define LOGGING  // <- Logging is for the HTTP library

// Range to attempt to autobaud
#define GSM_AUTOBAUD_MIN 9600
#define GSM_AUTOBAUD_MAX 115200

// Add a reception delay - may be needed for a fast processor at a slow baud rate
//#define TINY_GSM_YIELD() { delay(2); }

// Define how you're planning to connect to the internet
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials, if any
const char apn[]  = "internet";
const char gprsUser[] = "wap";
const char gprsPass[] = "500";

// Your WiFi connection credentials, if applicable
const char wifiSSID[]  = "YourSSID";
const char wifiPass[] = "YourWiFiPass";

// Server details
const char server[] = "termobil.ecancan.com";
const char resource[] = "/reportgsm";
const int  port = 80;

#include <TinyGsmClient.h>
#include <ArduinoHttpClient.h>

// Just in case someone defined the wrong thing..
#if TINY_GSM_USE_GPRS && not defined TINY_GSM_MODEM_HAS_GPRS
#undef TINY_GSM_USE_GPRS
#undef TINY_GSM_USE_WIFI
#define TINY_GSM_USE_GPRS false
#define TINY_GSM_USE_WIFI true
#endif
#if TINY_GSM_USE_WIFI && not defined TINY_GSM_MODEM_HAS_WIFI
#undef TINY_GSM_USE_GPRS
#undef TINY_GSM_USE_WIFI
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
#endif

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

TinyGsmClient client(modem);
HttpClient http(client, server, port);

void setup() {
  // Set console baud rate
  SerialMon.begin(115200);
  delay(10);

  // !!!!!!!!!!!
  // Set your reset, enable, power pins here
  // !!!!!!!!!!!

  SerialMon.println("Wait...");

  // Set GSM module baud rate
  // TinyGsmAutoBaud(SerialAT,GSM_AUTOBAUD_MIN,GSM_AUTOBAUD_MAX);
  SerialAT.begin(9600);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  modem.restart();
  // modem.init();

  String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem Info: ");
  SerialMon.println(modemInfo);

#if TINY_GSM_USE_GPRS
  // Unlock your SIM card with a PIN if needed
  if ( GSM_PIN && modem.getSimStatus() != 3 ) {
    modem.simUnlock(GSM_PIN);
  }
#endif
}

void loop() {

#if TINY_GSM_USE_WIFI
  // Wifi connection parameters must be set before waiting for the network
  SerialMon.print(F("Setting SSID/password..."));
  if (!modem.networkConnect(wifiSSID, wifiPass)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" success");
#endif

#if TINY_GSM_USE_GPRS && defined TINY_GSM_MODEM_XBEE
  // The XBee must run the gprsConnect function BEFORE waiting for network!
  modem.gprsConnect(apn, gprsUser, gprsPass);
#endif

  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork()) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" success");

  if (modem.isNetworkConnected()) {
    SerialMon.println("Network connected");
  }

#if TINY_GSM_USE_GPRS
  // GPRS connection parameters are usually set after network registration
    SerialMon.print(F("Connecting to "));
    SerialMon.print(apn);
    if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
      SerialMon.println(" fail");
      delay(10000);
      return;
    }
    SerialMon.println(" success");

    if (modem.isGprsConnected()) {
      SerialMon.println("GPRS connected");
    }
#endif

  SerialMon.print(F("Performing HTTP GET request... "));
  int err = http.get(resource);
  if (err != 0) {
    SerialMon.println(F("failed to connect"));
    delay(10000);
    return;
  }

  int status = http.responseStatusCode();
  SerialMon.print(F("Response status code: "));
  SerialMon.println(status);
  if (!status) {
    delay(10000);
    return;
  }

  SerialMon.println(F("Response Headers:"));
  while (http.headerAvailable()) {
    String headerName = http.readHeaderName();
    String headerValue = http.readHeaderValue();
    SerialMon.println("    " + headerName + " : " + headerValue);
  }

  int length = http.contentLength();
  if (length >= 0) {
    SerialMon.print(F("Content length is: "));
    SerialMon.println(length);
  }
  if (http.isResponseChunked()) {
    SerialMon.println(F("The response is chunked"));
  }

  String body = http.responseBody();
  SerialMon.println(F("Response:"));
  SerialMon.println(body);

  SerialMon.print(F("Body length is: "));
  SerialMon.println(body.length());

  // Shutdown

  http.stop();
  SerialMon.println(F("Server disconnected"));

#if TINY_GSM_USE_WIFI
    modem.networkDisconnect();
    SerialMon.println(F("WiFi disconnected"));
#endif
#if TINY_GSM_USE_GPRS
    modem.gprsDisconnect();
    SerialMon.println(F("GPRS disconnected"));
#endif

  // Do nothing forevermore
  while (true) {
    delay(1000);
  }
}`

Result:

`Wait...
Initializing modem...
[8676] ### TinyGSM Version: 0.9.17
[8881] ### Unhandled: ⸮
[9092] ### Modem: SIMCOM SIM800L
[9093] ### Modem: SIMCOM SIM800L
Modem Info: SIM800 R14.18
Waiting for network... success
Network connected
Connecting to internet success
GPRS connected
Performing HTTP GET request... [27201] ### Available: 0 on 1
[31329] ### Available: 0 on 1
[32377] ### Unhandled: 
[34429] ### Available: 0 on 1
[35478] ### Unhandled: 
[37530] ### Available: 0 on 1
[38578] ### Unhandled: 
[40630] ### Available: 0 on 1
`

SRGDamia1, 2019-09-29:
Could you uncomment the '#define DUMP_AT_COMMANDS' and post that log? You'll also need to install StreamDebugger if you don't have it.

Ecancan, 2019-09-29:
> Wait...
> Initializing modem...
> AT
> 
> OK
> AT+CLTS=1
> 
> OK
> AT&W
> 
> OK
> AT+CFUN=0
> 
> +CPIN: READY
> 
> +CPIN: NOT READY
> 
> OK
> AT+CFUN=1,1
> 
> Call Ready
> 
> ⸮⸮⸮⸮⸮R
> OK
> [7770] ### TinyGSM Version: 0.9.17
> AT
> ⸮[7975] ### Unhandled: ⸮
> AT
> 
> OK
> AT&FZ
> 
> OK
> ATE0
> ATE0
> 
> OK
> AT+GMM
> 
> SIMCOM_SIM800L
> 
> OK
> [8177] ### Modem: SIMCOM SIM800L
> [8178] ### Modem: SIMCOM SIM800L
> AT+CPIN?
> 
> ERROR
> AT+CPIN?
> 
> RDY
> 
> +CFUN: 1
> 
> ERROR
> AT+CPIN?
> 
> ERROR
> AT+CPIN?
> 
> ERROR
> AT+CPIN?
> 
> +CPIN: READY
> 
> +CPIN: READY
> 
> OK
> ATI
> 
> SIM800 R14.18
> 
> OK
> Modem Info: SIM800 R14.18
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [13377] ### Unhandled: 
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [15388] ### Unhandled: 
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [17399] ### Unhandled: 
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [19410] ### Unhandled: 
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [21421] ### Unhandled: 
> Waiting for network...AT+CREG?
> 
> +CREG: 0,1
> 
> OK
>  success
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> Network connected
> Connecting to internetAT+CIPSHUT
> 
> SHUT OK
> AT+CGATT=0
> 
> OK
> AT+SAPBR=3,1,"Contype","GPRS"
> 
> OK
> AT+SAPBR=3,1,"APN","internet"
> 
> OK
> AT+SAPBR=3,1,"USER","wap"
> 
> OK
> AT+SAPBR=3,1,"PWD","500"
> 
> OK
> AT+CGDCONT=1,"IP","internet"
> 
> OK
> AT+CGACT=1,1
> 
> OK
> AT+SAPBR=1,1
> 
> OK
> AT+SAPBR=2,1
> 
> +SAPBR: 1,1,"10.207.190.147"
> 
> OK
> AT+CGATT=1
> 
> OK
> AT+CIPMUX=1
> 
> OK
> AT+CIPQSEND=1
> 
> OK
> AT+CIPRXGET=1
> 
> OK
> AT+CSTT="internet","wap","500"
> 
> OK
> AT+CIICR
> 
> OK
> AT+CIFSR;E0
> 
> 10.207.190.147
> 
> OK
> AT+CDNSCFG="8.8.8.8","8.8.4.4"
> 
> OK
>  success
> AT+CGATT?
> 
> +CGATT: 1
> 
> OK
> AT+CIFSR;E0
> 
> 10.207.190.147
> 
> OK
> GPRS connected
> Performing HTTP GET request... AT+CIPCLOSE=1,1
> 
> ERROR
> AT+CIPSSL=0
> 
> OK
> AT+CIPSTART=1,"TCP","termobil.ecancan.com",80
> 
> OK
> 
> 1, CONNECT OK
> AT+CIPSEND=1,3
> 
> >GET 
> DATA ACCEPT:1,3
> AT+CIPSEND=1,1
> 
> >  
> DATA ACCEPT:1,1
> AT+CIPSEND=1,8
> 
> >/reports 
> DATA ACCEPT:1,8
> AT+CIPSEND=1,9
> 
> > HTTP/1.1 
> DATA ACCEPT:1,9
> AT+CIPSEND=1,2
> 
> >
>  
> DATA ACCEPT:1,2
> AT+CIPSEND=1,6
> 
> >Host:  
> DATA ACCEPT:1,6
> AT+CIPSEND=1,20
> 
> >termobil.ecancan.com 
> DATA ACCEPT:1,20
> AT+CIPSEND=1,2
> 
> >
>  
> DATA ACCEPT:1,2
> AT+CIPSEND=1,10
> 
> >User-Agent 
> DATA ACCEPT:1,10
> AT+CIPSEND=1,2
> 
> >:  
> DATA ACCEPT:1,2
> AT+CIPSEND=1,13
> 
> >Arduino/2.2.0 
> DATA ACCEPT:1,13
> AT+CIPSEND=1,2
> 
> >
>  
> DATA ACCEPT:1,2
> AT+CIPSEND=1,10
> 
> >Connection 
> DATA ACCEPT:1,10
> AT+CIPSEND=1,2
> 
> >:  
> DATA ACCEPT:1,2
> AT+CIPSEND=1,5
> 
> >close 
> DATA ACCEPT:1,5
> AT+CIPSEND=1,2
> 
> >
>  
> DATA ACCEPT:1,2
> AT+CIPSEND=1,2
> 
> >
>  
> DATA ACCEPT:1,2
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [29797] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","CONNECTED"
> 
> OK
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 1,1
> 
> +CIPRXGET: 4,1,338
> 
> OK
> [30960] ### Available: 1 on 1
> AT+CIPRXGET=2,1,1
> 
> +CIPRXGET: 2,1,1,337
> H
> OK
> [31982] ### Unhandled: 
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,337
> 
> OK
> [32037] ### Available: 337 on 1
> AT+CIPRXGET=2,1,337
> 
> +CIPRXGET: 2,1,337,0
> TTP/1.1 200 OK
> Connection: close
> X-Powered-By: PHP/7.2.22
> Set-Cookie: PSSDc1c6a83f0eb;a=
> xr:h  o11820G
> aeor:otench sradePaanch
> nnTettlnhs=F8Cttet 
> t n2S 1145GT
> e
> K[33061] ### Unhandled: 
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [33115] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","REMOTE CLOSING"
> 
> OK
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [34283] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","REMOTE CLOSING"
> 
> OK
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [35465] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> Call Ready
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","REMOTE CLOSING"
> 
> OK
> AT+CIPRXGET=4,1
> 
> 1, C⸮5⸮
> +CIPRXGET: 4,1,0
> 
> OK
> [36635] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","CLOSED"
> 
> OK
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [37793] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","CLOSED"
> 
> OK
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [38950] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","CLOSED"
> 
> OK
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [40109] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","CLOSED"
> 
> OK
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [41266] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","CLOSED"
> 
> OK
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [42423] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","CLOSED"
> 
> OK
> AT+CIPRXGET=4,1
> 
> +CIPRXGET: 4,1,0
> 
> OK
> [43580] ### Available: 0 on 1
> AT+CIPSTATUS=1
> 
> +CIPSTATUS: 1,0,"TCP","109.232.216.139","80","CLOSED"
> 
> OK
> 

I'm using arduino nano. So I am using software serial.

Ecancan, 2019-09-29:
I also encounter the following problem when I submit a request like this.

```
// Server details
const char server[] = "termobil.ecancan.com";
const char resource[] = "/reports/?securekey=xxxxxxxx&temp=40&hum=55.34&systemtime=0:0:0:2:36:18&device_token=xxxxxxxxxx&userkey=xxxxxx";
const int  port = 80;
```

Result

> Wait...
> Initializing modem...
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> ATI
> 
> SIM800 R14.18
> 
> OK
> [14027] ### Unhandled: 
> Modem Info: 
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [15039] ### Unhandled: 
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [17050] ### Unhandled: 
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [19061] ### Unhandled: 
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [21072] ### Unhandled: 
> AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> [23083] ### Unhandled: 
> Waiting for network...AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
>  fail
> Waiting for network...AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
> AT+CREG?
> 
> +CREG: 0,1
> 
> OK
>  fail
>

SRGDamia1, 2019-09-29:
It looks like a serial issue; the right commands are going back and forth, but they're not being understood. You can use AltSoftSerial on the Nano; it's likely to help. 

Other things to try:

- Try uncommenting the '#define TINY_GSM_YIELD() { delay(2); }' to see if adding a little delay helps to keep responses from being spliced. 
- Try the web client example instead of the http client. The HTTPClient library is convenient, but it can be slow.

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1

juanccidonncha, 2023-06-30:
> #ifdef DUMP_AT_COMMANDS #include <StreamDebugger.h> StreamDebugger debugger(SerialAT, SerialMon); TinyGsm modem(debugger); #else TinyGsm modem(SerialAT); #endif
> 
> TinyGsmClient client(modem); HttpClient http(client, server, port);

It appears an error when i use the TinyGsm modem(SerialAT) line and sais that identifier "TinyGsm" is undefined but i don't know how to solve that. I have tried to solve it through the libraries but I always get that error