vshymanskyy/TinyGSM #334: SIM900 + Arduino Uno wiring

Kepete, 2019-10-07:
Not an issue but have been struggling with it and the chat seems to be not do active. 

I'm stuck with the correct wiring for SIM900 and Arduino Uno combo.
I tried https://github.com/vshymanskyy/TinyGSM/blob/master/tools/AT_Debug/AT_Debug.ino
and it is unsuccessful. Probably a wiring issue as I have seen many varations of how to connect uno and SIM900.
Currently tried this wiring setup:
[image]
Changed the pins in `SoftwareSerial SerialAT(8, 7); // RX, TX` but without success. 
Also tried many different combinations but no luck.

SRGDamia1, 2019-10-07:
Did you make sure to use the button to turn the sim900 on or implement the wire and power on function?

Kepete, 2019-10-07:
> Did you make sure to use the button to turn the sim900 on or implement the wire and power on function?

Yes I did power it on from the button. Also I have been successfully using this module with other libraries/code. 

Is the wiring correct?

SRGDamia1, 2020-04-15:
I'm sorry; I know this is very old.  There shouldn't be a wiring issue; it's a shield made for an uno.  I think the issue was non-library code and I'm going to close this issue.  If you still need help with this, you can re-open it.

closed 2020-04-15 by SRGDamia1