vshymanskyy/TinyGSM #335: STM32duino + UBLOX Sara-U201

arminsalcin, 2019-10-10:
Hello to all , 

I have a question about using the arduino or stm32duino to communicate with ublox sarau-201 module , as i google it i didn't found any example or someone using it together .

Is it okay to use it , and why someone didn't try to use it together before ?

SRGDamia1, 2019-10-10:
I don't see any reason they wouldn't work together.

arminsalcin, 2019-10-10:
Thank you for response @SRGDamia1 ,

Me neither , just was curious bcz i tought it need some more processing but as i see ublox also use at commands so everything should be okay :D

abhaysbharadwaj, 2020-01-26:
Hi,
I can confirm that this library works with STM32Duino.
I have tested it with STM32F103C8T6 and a Sim800C.
Used Roger Clark's STM32Duino Arduino Core.
Tested HTTP + MQTT, as well as HTTPS + MQTTS.

closed 2020-04-15 by SRGDamia1