vshymanskyy/TinyGSM #337: Any plans for adding SMS Receive for SIM800L?

MornyDK, 2019-10-16:
Hi

Are there any plans for adding the possibility to receive SMS for the SIM800L, and if so what are the time frame for the implementation?

Kind Regards
MornyDK

SRGDamia1, 2019-10-16:
I don't currently have any plans to implement this and think it is very, very unlikely I will ever make such plans.  The 2G network is vanishing in the US and nothing I'm working on uses SMS in any way.

I don't know if @vshymanskyy has any different plans.

emigbur, 2020-04-10:
Hi, I'm not totally agree with you, first, this great library is not just used on US, I use SMS a lot to controls my IOT devices, GSM is going to stay for long time, for sure NB IOT , CAT M will be the future for this devices, i'm working with this too, but GSM still been more used.
Please take this in count.

SRGDamia1, 2020-04-10:
You're welcome to submit a pull request.