vshymanskyy/TinyGSM #341: modemConnect timeout

nikost000, 2019-10-31:
It seems that the timeout for modemConnect is hard coded to 75 seconds, set in TinyGsmCommon.h:

`
 virtual int connect(const char *host, uint16_t port) { \
    return connect(host, port, 75); \
  } \
  virtual int connect(IPAddress ip, uint16_t port) { \
    return connect(ip, port, 75); \
  }
`

So when the host doesn't respond it takes a full 75 seconds until it fails, which drains a lot of power.

SRGDamia1, 2019-10-31:
You can use put in any timeout you want.  The 75 that you're seeing in the common is just an overload.

Just use ```client.connect(server, port, timeout_s)``` and it will honor your timeout.  If you use ```client.connect(server, port)``` then it will use a 75 second timeout.

SRGDamia1, 2019-10-31:
But be aware that if you're using pointers to a Arduino ```Client``` object (instead of to a ```TinyGsmClient ``` object) you will not be able to use the custom timeout because that method isn't part of the lower level ```Client`` object.  Depending on what you're trying to do, it will either fail to compile or give you *undefined behavior* because of object slicing.

If you're not passing around pointers or calling the base Client, change your timeout at will.

SRGDamia1, 2020-04-15:
I'm closing this because I think it's been addressed.  If you're still having problems, please re-open.

closed 2020-04-15 by SRGDamia1