vshymanskyy/TinyGSM #343: BG96 getLocalIP() not working

beegee-tokyo, 2019-11-22:
## What type of issues is this?

[X] Bug or issue with library functionality (checkl if GPRS is connecte)

On BG96 the function isGprsConnected always fails. The function tries to check if a valid IP address is set with **`AT+QILOCIP`**, but the BG96 doesn't support this AT command.

### What are you working with?

Main processor board: ESP32
Modem: Quectel BG96
TinyGSM version: 0.9.17

### Code
```cpp
	if (modem.isGprsConnected())
	{
		DBG("GPRS connected");
	}
	else
	{
		DBG("GPRS NOT connected");
	}

```
### Scenario, steps to reproduce
I want to check if GPRS is still connected regularly after gprsConnect() was successfull

### Expected result
isGprsConnected should return true if the connection is active and only return true if the connection is broken ( ==> no IP address)

### Actual result
isGprsConnected returns always false because AT command **`AT+QILOCIP`** is not a valid command for the BG96

### AT command log
```
[19909] Connecting to APN >>  internet
AT+QIDEACT=1

OK
AT+QICSGP=1,1,"internet","",""

OK
AT+QIACT=1

OK
AT+CGATT=1

OK
[24647] - success
AT+CGATT?

+CGATT: 1

OK
AT+QILOCIP

ERROR
AT+QILOCIP

ERROR
```

SRGDamia1, 2019-11-22:
Huh.  You are quite correct.  It should be using ```AT+QIACT?``` which returns ```+QIACT: <contextID= 1>, <context_state (1=active)>, <context_type (1=IPV4/2=IPV6)>, <IP_address>```

If I don't get to fixing it soon, could you submit a PR with the fix?

beegee-tokyo, 2019-11-23:
Found another weird bug in the BG96. **'AT+QISTATE=1,1'** did not work.

In **'modemGetConnected()'** I had to change 
'''if (waitResponse(GF("+QISTATE:")))'''
to 
'''if (waitResponse(GF("+QISTATE:")) != 1)'''
to get it right.
AT debug looks the same, but **'modemGetConnected()'** returned false without the change
'''
AT+QISTATE=1,1

+QISTATE: 1,"TCP","52.7.124.212",1883,26195,2,1,1,0,"uart1"

OK
'''

Preparing a pull request now.

SRGDamia1, 2020-04-15:
I think this has been addressed.

closed 2020-04-15 by SRGDamia1

beegee-tokyo, 2020-04-15:
Thanks for merging finally.