vshymanskyy/TinyGSM #346: Code architecture, DRY

AlexIII, 2019-11-29:
Hi!
I wanted to make a pull request adding functionality to send SMS in utf8 (which is native to Arduino IDE) in efficient way (which does not require to compose and prepare the whole SMS string before sending it). Essentially it just provides a Print interface to the user which converts subset of utf8 to utf16 on the fly and sends it to SIMxxx module.
The problem is, I really confused as to where I should add my code in a way that not breaks your library architecture!
Essentially I need to add to `TinyGsm` class three methods:
```
bool sendSMS_UTF8_begin(const char* const number);
Print& sendSMS_UTF8_stream();
bool sendSMS_UTF8_end();
```
Which are going to be used like this:
```
modem.sendSMS_UTF8_begin(number);
auto s = sendSMS_UTF8_stream();
s.print(F("UTF-8 chars: АБВГД, var="));
int var = -195;
s.print(var);
modem.sendSMS_UTF8_end();
```
In my project I just inherited from `TinyGsm` and added my new methods in the derived class, but now I suppose I ought to add them alongside sendSMS_UTF16() method. And there's 6 of them exactly the same in  different files!
So, the question is, should I also follow that pattern or maybe there's a better way to do that?

SRGDamia1, 2019-12-02:
The code base is a bit ugly, isn't it.  :-/  The repetition comes from a combination of many modems with often very similar but not actually the same commands and a desire to keep the overall implementation small.  (#280)  If it weren't for the v-table memory issue, some straightforward virtual classes would be so very much better.

The pre-processor macros reduce the repetition somewhat but they are just ugly and seem to confuse everyone.  I would really like to improve the whole setup, but I don't have time right now.

As for your functions, the simplest thing would be to implement them just for the specific modem you're interested in and leave everything else alone.  That is - continue on the trend that gives the SIM800 more functions than any other modem.  If you're sure the other SIMCOM or other modems that have the same methods are really exactly the same, you could consolidate into a macro in the common file and call that macro in the individual modem files.

AlexIII, 2019-12-03:
Ok, thank you for your answer. The methods I want to add are based on existing `sendSMS_UTF16()` which is exactly the same across the different modems. I will try to use CRTP as I suggested earlier for these few methods. Maybe you'll like it, or, if not, I will move the common code to a macro.
Btw, I realize it is very hard to keep the balance between functionality, efficiency and beauty of the code on such a restricted platform. The code base is not _ugly_ it just could use a little refactor. (=

SRGDamia1, 2020-03-02:
Check it out now.

AlexIII, 2020-03-18:
Cool! I'll test it with my project.