vshymanskyy/TinyGSM #348: speed of FileDownload test is slow

danielsafrigin, 2019-12-11:
## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[x] Bug or issue with library functionality (ie, sending data over TCP/IP)

### What are you working with?

Main processor board: ESP32
Modem: sim7000e
TinyGSM version: 0.7.9
Code: FileDownload

### Scenario, steps to reproduce

I try to download the file 100k file, tried with 115200, 230400, 921600
`const char resource[]  = "/TinyGSM/test_100k.bin";`
### Expected result
I expect to receive 100k bytes/s

### Actual result
I receive 5k bytes/s

SRGDamia1, 2019-12-13:
Do you get faster downloads via AT commands?

furqon2710, 2023-07-21:
I am having the same problem. downloading 900kb Files. more than an hour the download is not finished yet. and then it stop on 95%.  here is the log

Connecting to server OK
Reading header
 81.84%
stop client
gprs disconnect

File Size 938256
Length:  767892
Calculated. CRC32:    0x0
Expected CRC32:    0xAF968D0E
Downloaded in:       3702.79s

if i calculated, then my download speed only  200 bytes/second