vshymanskyy/TinyGSM #352: Clarify that this library does not "just work" - it must only be used for experimentation and relies on undefined behvior/has serious known bugs

BracketJohn, 2019-12-23:
## What type of issues is this?

Bug or issue with library functionality.

### What are you working with?

TinyGSM version: 0.3.5
Hardware: SIM868 R14.18, Arduino Uno for prototyping, atmega328p for deployment

### Scenario, steps to reproduce

Base scenario is #255. However, this is a more holistic issue: I think that at least in the implementation of the SIM800/SIM808 modules - I have not checked any other modules code - there are bigger issues with heap allocations, casting and possible memory leaks.

#### Heap allocations

There are heap allocations all over the place, e.g., every time a `String` or some operation deriving from it is used. This, in itself, is not necessarily a bug. It is however generally considered bad style (heap allocations) and the usage of `String` is largely frowned upon by the community. See [here](https://hackingmajenkoblog.wordpress.com/2016/02/04/the-evils-of-arduino-strings/), [here](https://forum.arduino.cc/index.php?topic=364772.0) and [here](https://forum.arduino.cc/index.php?topic=444696.0) for blogs and threads on this.

#### Casting

Casting happens all over the place, e.g., there's four uses of Arduino's `toFloat` and more uses of `toInt` than that in the SIM800 and SIM808 code. `toFloat` has the problem that it implicitly truncated, [see here for more  (official arduino docs)](https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/tofloat/). `toInt` will:

> stop performing the conversion.

whenever it received invalid chars. It is not specified what this means. [See here for the official docs on this](https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/toint/)

#### Memory Leaks

The aforementioned points and the just overall Arduino `String` heavy implementation of this library (see for example the `waitResponse` method) lead to large amounts of memory fragmentation. Over time, this will lead your MCU to being out of RAM and to then drift into undefined behavior (UB).

This is terrible to debug (I've been there a couple of times and it took me a long time to realize this) and even harder to fix.

### Actual result

I decided to write this up, after a discussion I had with @djcobylevy  in #255. There, the behavior of this lib has lead them to believe, that the underlying function just works, but since it casts using `toFloat`, it truncated. They then use a call to `String(aFloatVar, 6)` in order to cast to `String` before printing. As the Arduino std lib relies on UB when there's less than 6 bits, they also receive a 6 decimal place float which they believe to function. However, what really happened, is that they cast un-initialized memory.

### Expected result

That the above should not happen/that is should be more explicitly documented in the `README`. Currently, the `README` states:

> A small Arduino library for GSM modules, that just works.

I do not believe this to be true. I think it should be more explicitly stated, that this library relies on undefined behavior and might behave in unexpected ways, i.e., I think it should be more clearly stated, that this in an alpha or something to experiment with 

This is, implicitly stated further down with:
> Now, you have more space for your experiments.

But after the previously described exchange, I don't think this is enough. Before said exchange, I though that this was just an issue for advanced use cases. After said exchange, I think that this is an issues for users that are just starting out to use this library as well. For this reason, there should be a big focus on these issues from the beginning.

If there's even more time to improve the `README`, some example of unexpected behavior should be given, so that people know what to expect when using this library.

Note: In no way am I blaming the author of this library or requesting any actual improvements or work; I know that OSS sucks and am very thankful that this library exists. However, it should be clear to anyone that this cannot be used for anything more than experiments. Personally, I've started the development of an actual hardware product for a company using this library. It was amazing and we got a fast prototype. But, once you let stuff run for more than an hour or do more complicated operations (e.g., multiple GPS Positioning requests, multiple HTTP GET requests), this library lead our MCU to crash 90 % of the time. It was hard to debug and in the end we wrote our own subset of this library. It is only for the SIM8XX series and far from being useful for anyone else than us (which makes it arguably worse than this lib) - but I intend to publish the subset once I find time to make it actually usable.

SRGDamia1, 2020-02-07:
I've done a ton of work on this library in the last week or so I've been thinking about ways to improve the library/reduce **S**tring use.  I can do away with a lot of the casts and have replaced the problem GPS toFloats() - https://github.com/EnviroDIY/TinyGSM/commit/06c31aa856c24f35d841ac84f1566527bdb1e43a - but I'm stuck on the "waitResponse()" problem.  I've scribbled out a few alternatives, but all of them feel like contortions or lead to confusions in the overloads or other problems that instantly disappear with the super convenient functionality of that String class.  So if you've come up with anything functional there, I'd love suggestions.

Please check out what I've done over on the develop branch of the EnviroDIY fork.

SRGDamia1, 2020-02-08:
Also, we have loggers using this library going on _years_ of deployment making get requests every 5 minutes. We're admittedly using a larger 1284p processor, but filling it up with libraries for storing to SD and communicating with a variety of environmental sensors. And, like I said, we have programs running for many months, not just a few hours. What other libraries are you attempting to use with this?

BracketJohn, 2020-02-10:
Thank you for getting back to me on this, I really appreciate your effort! 

I will take a peak at your code. In addition to that, I will try to extract a usable `waitResponse` from our codebase if I can get around to that. I think just replacing `waitResponse` will already be a huge gain.

QMand, 2020-05-11:
I usually have a check to see if a value was found in my returned data. 
boolean sendATcommand(char* ATcommand, char* expected_answer)
I then check for a response in my buffer
response[x] =  Serial.read();
if (strstr(response, expected_answer) != NULL){
return = true;
}

I also have a timeout etc. in my function. I'm not sure if this will help some to maybe replace the waitResponse with a non String version.

roysG, 2021-09-26:
Is there any update on this issue?