vshymanskyy/TinyGSM #357: configTime crashes, so no NTP setup on esp32

minusplusminus, 2020-01-18:
Hi,

I'm using the Esp32 combined with Sim800L. Everything works perfect, except NTP. 

In TinyGSM 0.7.9 I try to set NTP time by using `configTime` function of arduino. In other examples, when using Wifi, it's done after the Client is initialized. But when I do this with TinyGSM, it's not the case. I could not get a backtrace working yet. Here's my code:

```
/**************************************************************

   This sketch connects to a website and downloads a page.
   It can be used to perform HTTP/RESTful API calls.

   TinyGSM Getting Started guide:
     https://tiny.cc/tinygsm-readme

 **************************************************************/

// Your GPRS credentials (leave empty, if missing)
const char apn[]      = "portalmmm.nl"; // Your APN
const char gprsUser[] = ""; // User
const char gprsPass[] = ""; // Password
const char simPIN[]   = ""; // SIM card PIN code, if any

// TTGO T-Call pin definitions
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to the module)
#define SerialAT  Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb

// Define the serial console for debug prints, if needed
//#define TINY_GSM_DEBUG SerialMon
//#define DUMP_AT_COMMANDS

#include <Wire.h>
#include <TinyGsmClient.h>
#include "utilities.h"

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

// Server details
const char server[] = "vsh.pp.ua";
const char resource[] = "/TinyGSM/logo.txt";

TinyGsmClient client(modem);
const int  port = 80;

// Configuration for NTP
const char* ntp_primary = "pool.ntp.org";
const char* ntp_secondary = "time.nist.gov";

void setupNTP() {
  Serial.println("Syncing time...");
  configTime(0, 0, ntp_primary, ntp_secondary);
  Serial.println("Waiting on time sync...");
  while (time(nullptr) < 1510644967) {
    delay(10);
  }
}

void setup() {
  // Set console baud rate
  SerialMon.begin(115200);
  delay(10);

  // Keep power when running from battery
  Wire.begin(I2C_SDA, I2C_SCL);
  bool   isOk = setPowerBoostKeepOn(1);
  SerialMon.println(String("IP5306 KeepOn ") + (isOk ? "OK" : "FAIL"));

  // Set-up modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);

  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);

  // Set GSM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  modem.restart();
  // Or, use modem.init() if you don't need the complete restart

  String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem: ");
  SerialMon.println(modemInfo);

  // Unlock your SIM card with a PIN if needed
  if (strlen(simPIN) && modem.getSimStatus() != 3 ) {
    modem.simUnlock(simPIN);
  }
}

void loop() {
  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork(240000L)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  if (modem.isNetworkConnected()) {
    SerialMon.println("Network connected");
  }

  SerialMon.print(F("Connecting to APN: "));
  SerialMon.print(apn);
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  SerialMon.print("Connecting to ");
  SerialMon.print(server);
  if (!client.connect(server, port)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  setupNTP();
}
```

Is it possible to set the Client somewhere or am I missing something.

SRGDamia1, 2020-01-18:
I'm sorry, I'm not at all familiar with that function. NTP is a UDP based protocol, so if that function is trying to do something with UDP, it won't work. This library only supports TCP.

minusplusminus, 2020-01-20:
Thanks for the reply. 

Is there an other way to configure the time of the esp32?

SRGDamia1, 2020-01-20:
I think you're confusing the ESP32's on-board _WiFi_ functionality with the extra _cellular_ functionality this library can provide.  This library uses the serial port of the ESP32 to talk with the serial port of the SIM800 via AT commands and wraps it up nicely so that connection functions like an Arduino TCP 'Client' instance.  My really quick Google search on that ```configTime``` function shows it's part of the ESP32 core and expects the ESP32's own WiFI to be connected for it to work.  Then it uses UDP on WiFi to sync with NTP.  If you're connecting a SIM800 to the ESP32, I'm guessing it's because you _don't_ have any WiFi connection and all of the ESP32's WiFi related functionality is non-functional.  You'll have to write your own function that can use the TinyGSM Client's TCP connection to get the time from a TCP source (like the NIST daytime server, but _not_ NTP) and then set the onboard clock manually based on the time returned.

closed 2020-01-20 by SRGDamia1

ejri, 2021-03-23:
@minusplusminus what was your workaround for this? Did you manage to set configTime without relying on the esp32's wifi connection?

capedra, 2021-11-19:
This is how I'm getting the current NTP time using the SIM800C module and updating the ESP32 clock without using WiFi and incrementing the SIM800C epoch time by 1 second, since it's the offset that I've been facing:
```
modem.sendAT(GF("+CNTPCID=1"));
modem.waitResponse();

modem.sendAT(GF("+CNTP=\"200.160.7.186\",-12"));
modem.waitResponse();

modem.sendAT(GF("+CNTP"));
modem.waitResponse(3000, GF("+CNTP: 1" GSM_NL));

modem.sendAT(GF("+CCLK?"));
modem.waitResponse(GF(GSM_NL));
String cclk = modem.stream.readStringUntil('\n');
modem.waitResponse();
Serial.print("\r\nCCLK: ");
Serial.println(cclk);
String currentDt = cclk.substring(cclk.indexOf("\"")+1, cclk.lastIndexOf("\""));
currentDt.trim();
Serial.print("\r\ncurrentDt: ");
Serial.println(currentDt);

int hh, mm, ss, yy, mon, day, tz;
struct tm when = {0};
time_t epoch = 0;

sscanf(currentDt.c_str(), "%d/%d/%d,%d:%d:%d-%d", &yy, &mon, &day, &hh, &mm, &ss, &tz);
when.tm_hour = hh;
when.tm_min = mm;
when.tm_sec = ss;
when.tm_year = 2000 + yy - 1900;

when.tm_mon = mon - 1;
when.tm_mday = day;
when.tm_isdst = -1;
epoch = mktime(&when);
struct timeval tv;
tv.tv_sec = epoch+1;
tv.tv_usec = 0;
settimeofday(&tv, NULL);

epoch = epoch + (60 * (tz/4) * 60);
Serial.print("CURRENT EPOCH:");
Serial.println(epoch);
```

After that, it's just a matter of querying the correct date and time using the getLocalTime function:
```
modem.sendAT(GF("+CCLK?"));
modem.waitResponse(GF(GSM_NL));
String cclk = modem.stream.readStringUntil('\n');
modem.waitResponse();
String currentDt = cclk.substring(cclk.indexOf("\"")+1, cclk.lastIndexOf("\""));
currentDt.trim();
getLocalTime(&timeinfo);
char s[51];
strftime(s, 50, "%A, %B %d %Y %H:%M:%S", &timeinfo);

Serial.print("\r\nCCLK: ");
Serial.println(cclk);

Serial.print("\r\ncurrentDt from SIM800: ");
Serial.println(currentDt);

Serial.print("\r\ncurrentDt from ESP32: ");
Serial.println(String(s));

```

Ki-csen, 2022-09-14:
> This is how I'm getting the current NTP time using the SIM800C module and updating the ESP32 clock without using WiFi and incrementing the SIM800C epoch time by 1 second, since it's the offset that [...]

Thank you a lot, your comment saved me. 
Just a note: The "%d/%d/%d,%d:%d:%d-%d" is not 100% correct. The timezone can be positive and in this case, the `epoch = epoch + (60 * (tz/4) * 60);` also can be fixed.

aprodrive, 2023-01-18:
Hey @capedra  your comment is very helpful .
But this code returns 12 hour format time , will anyone please  help I want time format of 24 hour ,