vshymanskyy/TinyGSM #358: Arduino Mega - SIM800L - MqttClient.ino

thecrow159, 2020-01-28:
## What type of issues is this?

[ x ] Bug or issue with library functionality (ie, sending data over TCP/IP)
       --> Modem is **not able to connect to the Mqtt Broker**

### What are you working with?

Main processor board: **Arduino Mega**
Modem: **SIM800 R14.18**
TinyGSM version: **0.7.9**
Code: **MqttClient.ino**

### Scenario, steps to reproduce
The modem can't connect to the mqtt broker, giving message **=== MQTT NOT CONNECTED ===**

### Expected result
Connection done with the broker

### Actual result
The Diagnostics.ino sketch exit with PASSED result.
Modem in factory default.
https://gist.github.com/thecrow159/51ec214276e2629838ebd277ef0c9ce3

### AT command log
Read above

SRGDamia1, 2020-01-28:
Connecting to MQTT on the SIM800 works fine - but the test.mosquitto.org broker often doesn't. It's a broker problem, not a TinyGSM one.  Try a different test broker or the broker you actually will use.  I usually have much better luck with broker.hivemq.com than test.mosquitto.org.

closed 2020-01-28 by SRGDamia1

thecrow159, 2020-01-29:
Hi SRGDamia1, here is the log with broker.hivemq.com. Thank you for replying

`13:28:44.201 -> Initializing modem...
13:28:44.201 -> AT
13:28:44.201 -> 
13:28:44.201 -> Call Ready
13:28:44.201 -> 
13:28:44.201 -> SMS Ready
13:28:44.201 -> 
13:28:44.201 -> OK
13:28:44.201 -> AT+CLTS=1
13:28:44.234 -> 
13:28:44.234 -> OK
13:28:44.234 -> AT&W
13:28:44.234 -> 
13:28:44.234 -> OK
13:28:44.234 -> AT+CFUN=0
13:28:44.599 -> 
13:28:44.599 -> +CPIN: NOT READY
13:28:49.280 -> 
13:28:49.280 -> OK
13:28:49.280 -> AT+CFUN=1,1
13:28:49.280 -> 
13:28:49.280 -> OK
13:28:52.301 -> [11080] ### TinyGSM Version: 0.7.9
13:28:52.301 -> AT
13:28:52.301 -> ⸮⸮[11283] ### Unhandled: ⸮⸮
13:28:52.566 -> AT
13:28:52.600 -> 
13:28:52.600 -> OK
13:28:52.600 -> AT&FZ
13:28:52.600 -> 
13:28:52.600 -> OK
13:28:52.600 -> ATE0
13:28:52.600 -> ATE0

13:28:52.600 -> OK
13:28:52.600 -> AT+GMM
13:28:52.600 -> 
13:28:52.600 -> SIMCOM_SIM800L
13:28:52.600 -> 
13:28:52.600 -> OK
13:28:52.600 -> [11412] ### Modem: SIMCOM SIM800L
13:28:52.600 -> [11412] ### Modem: SIMCOM SIM800L
13:28:52.633 -> AT+CPIN?
13:28:52.633 -> 
13:28:52.633 -> ERROR
13:28:53.630 -> AT+CPIN?
13:28:53.630 -> 
13:28:53.630 -> RDY
13:28:53.630 -> 
13:28:53.630 -> +CFUN: 1
13:28:53.630 -> 
13:28:53.630 -> ERROR
13:28:54.625 -> AT+CPIN?
13:28:54.625 -> 
13:28:54.625 -> ERROR
13:28:55.632 -> AT+CPIN?
13:28:55.632 -> 
13:28:55.632 -> +CPIN: READY
13:28:55.665 -> 
13:28:55.665 -> +CPIN: READY
13:28:55.665 -> 
13:28:55.665 -> OK
13:28:55.665 -> ATI
13:28:55.665 -> 
13:28:55.665 -> SIM800 R14.18
13:28:55.665 -> 
13:28:55.665 -> OK
13:28:55.665 -> Modem: SIM800 R14.18
[... 160 lines cut ...]
13:29:15.639 -> ERROR
13:29:15.639 -> [34419] ### Available: 0 on 1
13:29:15.639 -> AT+CIPSTATUS=1
13:29:15.639 -> 
13:29:15.639 -> ERROR
13:29:17.632 -> === MQTT NOT CONNECTED ===
13:29:17.731 -> AT+CIPRXGET=4,1
13:29:17.764 -> 
13:29:17.764 -> ERROR
13:29:17.764 -> [36537] ### Available: 0 on 1
13:29:17.764 -> AT+CIPSTATUS=1
13:29:17.764 -> 
13:29:17.764 -> ERROR
13:29:19.756 -> === MQTT NOT CONNECTED ===
13:29:19.855 -> AT+CIPRXGET=4,1
13:29:19.855 -> 
13:29:19.855 -> ERROR
13:29:19.855 -> [38657] ### Available: 0 on 1
13:29:19.855 -> AT+CIPSTATUS=1
13:29:19.889 -> 
13:29:19.889 -> ERROR
13:29:21.879 -> === MQTT NOT CONNECTED ===
13:29:21.879 -> Connecting to broker.hivemq.comAT+CIPRXGET=4,1
13:29:21.879 -> 
`

thecrow159, 2020-01-29:
I sorted it out. I had same issue of #318.
I installed the library directly from the IDE a week ago.

thecrow159, 2020-01-29:
In the MqttClient.ino example sketch (the last this time ..) I can see in the top description:

>  * Subscribe for messages:
>  *   mosquitto_sub -h test.mosquitto.org -t GsmClientTest/init -t GsmClientTest/ledStatus -q 1
>  * Toggle led:
>  *   mosquitto_pub -h test.mosquitto.org -t GsmClientTest/led -q 1 -m "toggle"

while in the sketch the broker is "broker.hivemq.com".

So there's a broker host mismatch

ortegafernando, 2020-01-30:
top description are comments. Don't worry.
I have the same error as you. May be anyone could help us.

thecrow159, 2020-01-30:
Yes, I'm using a mosquitto local server with auth login and everything is working now. Thank you