vshymanskyy/TinyGSM #361: TinyGSM Sending too much data to Serial

Nauman3S, 2020-02-17:
I am using TinyGSM for quite a while and it is great. The only issue that I am facing is, the library is sending too much information to my primary serial. I have commented out DUMP_AT_COMMANDS but still there is too much data that I can see on my serial monitor.
For example, in one of my projects I am using PubSubClient with TinyGSM and TinyGSM shows message number, subscribers, and clients etc.
I want to close serial interaction of TinyGSM altogether, is it something that is achievable?

SRGDamia1, 2020-02-17:
Have you defined ```TINY_GSM_DEBUG```?  Remove/comment out that definition if you have.  Without that defined it shouldn't print anything; with it you may get a fair amount, depending on which module you're using.  Defining ```DUMP_AT_COMMANDS``` will give you extensive output.

closed 2020-04-15 by SRGDamia1

Nauman3S, 2020-04-15:
I had TINY_GSM_DEBUG and DUMP_AT_COMMANDS commented out but still, there was a fair amount of data related to MQTT PUB/SUB, on SIM800L, that was being received. For that, I had to manually remove all the debug statements from lib.

SRGDamia1, 2020-04-15:
If you don't define TINY_GSM_DEBUG somewhere, the debug statements do nothing.  You must have it defined somewhere else in your code or your build flags.