vshymanskyy/TinyGSM #364: Using beginPublish() endpublish(9 from pubsubclient disconnect MQTT

hardmax, 2020-02-27:
Hi,

I send a large mqtt mensaje with pubsubclient library using beginPublish() endpublish() function. This function send directly data over tcp, without have to use a variable to storage the info.

When I use with wifi (ESP286, ESP32),  I dont have problems, but with use TinyGSM over a SIM800L, the data that send over MQTT is corrutc, and the mqtt connection get down.

If i use mqtt publish method, I dont have any problem, but for the size of data, I can´t use that method to send message.

I think that was a problem of:
#define MQTT_MAX_TRANSFER_SIZE 

But I try with diferents values in the library MQTT, but all still the same.

Regards

hwfloyd, 2020-06-08:
Hi @hardmax, I'm having problems trying to publish MQTT through the SIM800 of my ESP32 TTGO T-CALL. Do you know how to get it to work?

hardmax, 2020-06-08:
I have it working, I use tinyGSM and all are ok. What problem do you have?

hwfloyd, 2020-06-09:
Thank you for your reply. Im also using TinyGSM and PubSubClient. I was able to connect to the APN, but then failed to connect to the MQTT broker. I was able to do it through WiFi, but not with the SIM. I'm not sure if maybe the GPSR connection is lost or if I'm making a mistake with the MQTT client. Is there any chance you could share your code with me? or could you see what might be wrong with mine? Thank you!

Also might be important to add that soon as I connect to the APN, the led (which shows the connection status) starts blinking very fast and I don't know what that could mean. I only know that when it first connects to the network is blinks every 3 seconds, which means it's working (this is happening with mine, but only until it connects to the APN).