vshymanskyy/TinyGSM #366: SIM5320 doesn't work in TinyGSM

robg3987, 2020-03-08:
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
Arduino Mega
SIM5320A
TinyGSM version=0.9.17

TinyGSM_SIM800_SIM900
add #define TINY_GSM_MODEM_SIM5320  
//#define TINY_GSM_MODEM_SIM800
AT commands are for SIM800 to enable TCP connection, therefore it fails when I use SIM5320A.
These commands don't exist in the AT command list for SIM5320,, but the library is still trying to execute them (AT+CIPSHUT, CSTT, CIICR, CIFSR)
Proper commands for SIM5320A are 
AT+CGSOCKCONT=1,"IP","APN"
AT+CSOCKSETPN=1
AT+CIPMODE=0
AT+NETOPEN
AT+IPADDR
AT+NETCLOSE

Need library updated to include these
*****Tried changing to SIM5360,, and doesn't work either

FROM AT Command Tester:
**************************************************************************************************
utilizing the wrong AT commands from SIM800 :

> OK

Successfull response for AT query..
AT+CGMM
SIMCOM_SIM5320A
OK
Model Number : SIMCOM_SIM5320A
AT+CGMI
SIMCOM INCORPORATED
OK
Manufacturer : SIMCOM INCORPORATED
Checking registration status...
AT+CREG?
+CREG: 0,1
OK
The device is registered in home network.
AT+CGREG?
+CGREG: 0,1
OK
The device is registered in home network.
AT+CMEE=1
OK
Checking if device is already connected...
AT+CGACT?
+CGACT: 1,0
+CGACT: 2,0
OK
Close any previous connections...
AT+CIPSHUT
ERROR
Attaching to network...
AT+CGATT=1
OK
Device is successfully attached to the network
Setting up APN for TCP connection...
AT+CSTT="rogers-core-appl1.apn"
ERROR
Unable to set APN for TCP/UDP connection...
**********************************************************************************************
log from using the PROPER AT command set for SIM5320:

AT+CGSOCKCONT=1,"IP","rogers-core-appl1.apn"
OK
AT+CSOCKSETPN=1
OK
AT+CIPMODE=0
OK
AT+NETOPEN
OK
+NETOPEN: 0
AT+IPADDR
+IPADDR: 25.161.17.11
OK
AT+NETCLOSE
TCP connection is closed by remote server
OK
+NETCLOSE: 0
TCP connection is closed by remote server

SRGDamia1, 2020-03-08:
Um, those *are* the commands being used for the 5320/5360/5300/7100:  https://github.com/vshymanskyy/TinyGSM/blob/d28e654dfdbd46382821c0d2ac64e74573a4dacd/src/TinyGsmClientSIM5360.h#L287

You must be defining you modem as a SIM800 somewhere.

robg3987, 2020-03-08:
that's what I was thinking too,,, but I don't see where,,,,
this is from the TinyGSM_SIM800_SIM900 examples sketch:
`/* Comment this out to disable prints and save space */
#define BLYNK_PRINT Serial
#define BLYNK_DEBUG // Optional, this enables lots of prints

// Select your modem:
#define TINY_GSM_MODEM_SIM5320      //changed name from 5360 to 5320
//#define TINY_GSM_MODEM_SIM5360
//#define TINY_GSM_MODEM_SIM800
//#define TINY_GSM_MODEM_SIM900
//#define TINY_GSM_MODEM_M590
//#define TINY_GSM_MODEM_A6
//#define TINY_GSM_MODEM_A7
//#define TINY_GSM_MODEM_BG96
//#define TINY_GSM_MODEM_XBEE

// Default heartbeat interval for GSM is 60
// If you want override this value, uncomment and set this option:
//#define BLYNK_HEARTBEAT 30

#include <TinyGsmClient.h>
#include <BlynkSimpleTinyGSM.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "DuyrJ5LuYMjHx2i_XN8gjHcu79RuZkmo";

// Your GPRS credentials
// Leave empty, if missing user or pass
char apn[]  = "rogers-core-appl1.apn";
char user[] = "";
char pass[] = "";

// Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3); // RX, TX

TinyGsm modem(SerialAT);

void setup()
{
  // Debug console
  Serial.begin(115200);

  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(115200);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  Serial.println("Initializing modem...");
  modem.restart();

  // Unlock your SIM card with a PIN
  //modem.simUnlock("1234");

  Blynk.begin(auth, modem, apn, user, pass);
}

void loop()
{
  Blynk.run();
}`

SRGDamia1, 2020-03-08:
Huh, it's been a long time since I played with Blynk, but quickly checking the sketch and the Blynk adapter it looks like it should be fine.  Are you using the Arduino IDE or PlatformIO?  Are you sure you didn't set a build flag in the platformio.ini?  In either IDE, can you run a "clean" to remove all pre-compiled archives and then rebuild and see if you get the same thing?

robg3987, 2020-03-08:
hi,, I am using the Arduino IDE.
I am using the default example TinyGSM_SIM800_SIM900 that comes with the Blynk library.
I only modified the modem selection (added #define TINY_GSM_MODEM_SIM5320) and the APN.
I have recompiled and loaded clean sketches over and over with the same results, I've tried entirely new sketches from the Blynk editor,, I've tried other examples that all use the 
#define TINY_GSM_MODEM_SIM5320    //changed name from 5360 to 5320

#include <TinyGsmClient.h>
#include <BlynkSimpleTinyGSM.h>, instances,, with all the same results.
 so my only conclusion was that the library and class SIM5320 aren't quite right

SRGDamia1, 2020-03-08:
Could you update your library to the latest (0.10.1) and run the web client example with debug on and dump AT commands and post the output? I don't think there are any changes in the commands for the 5360 in the newer version, but I did a *lot* of refactoring before v0.10 and I'd rather test for bugs against the latest.

robg3987, 2020-03-09:
ok, will do. BTW, when I checked the IDE library manager for updates,, it only shows TinyGSM update as 0.7.9 ! So,I downloaded and installed from GH and double checked that version is 0.10.1

SRGDamia1, 2020-03-09:
The Arudino IDE goes by release numbers, and there hadn't been a release in ages.  I just put out a release, so the IDE should detect it - though I don't know how often it scans, it might not be right away.

robg3987, 2020-03-10:
will do... Do you want me to change modem class to 5320 or leave as 5360 ??

SRGDamia1, 2020-03-10:
It doesn't matter which you use.  They are in the same module in TinyGSM.

robg3987, 2020-03-13:
ok, here you go. Hopefully this contains the data your looking for.
Wait...
[9] Trying bau1024] Modem responded at rate 115200
Wait...
[9] Trying baud rate 115200 ...
[1024] Modem responded at rate 115200
Initializing modem...
AT

OK
AT+REBOOT

OK
[7057] ### TinyGSM Version: 0.10.1
[7057] ### TinyGSM Compiled Module:  TinyGsmClientSIM5360
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+CGMM

SIMCOM_SIM5320A

OK
[7090] ### Modem: SIMCOM SIM5320A
[7091] ### Modem: SIMCOM SIM5320A
AT+CTZR=0

OK
AT+CTZU=1

OK
AT+CPIN?

+CPIN: READY

OK
ATI

Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM5320A
Revision: SIM5320A_V1.5
IM:7059
A+SDE
K[8120] ### Unhandled: Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM5320A
Revision: SIM5320A_V1.5
IM:7059
A+SDE
K
Modem Info: 
AT+CPIN?

+CPIN: READY

OK
Waiting for network...AT+CGREG?

+CGREG: 0,1

OK
 success
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to rogers-core-appl1.apnAT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
AT+NETCLOSE

+NETCLOSE: 2

ERROR
AT+CGDCONT=1,"IP","rogers-core-appl1.apn","0.0.0.0",0,0

OK
AT+CGSOCKCONT=1,"IP","rogers-core-appl1.apn"

OK
AT+CSOCKSETPN=1

OK
AT+CIPMODE=0

OK
AT+CIPSENDMODE=0

OK
AT+CIPCCFG=10,0,0,0,1,0,75000

OK
AT+CIPTIMEOUT=75000,15000,15000

OK
AT+NETOPEN

OK

+NETOPEN: 1

START

+CPIN: READY

+PACSP1

OPL DONE

PNN DONE

SMS DONE

+VOICEMAIL: INIT_STATE, 0, 0

PB DONE
[83266] ### Unhandled: OK

+NETOPEN: 1

START

+CPIN: READY

+PACSP1

OPL DONE

PNN DONE

SMS DONE

+VOICEMAIL: INIT_STATE, 0, 0

PB DONE
 fail
Waiting for network...AT+CGREG?

+CGREG: 0,1

OK
 success
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to rogers-core-appl1.apnAT+CIPCLOSE=0

+IP ERROR: Network not opened

ERROR
AT+NETCLOSE

+IP ERROR: Network is already closed

ERROR
AT+CGDCONT=1,"IP","rogers-core-appl1.apn","0.0.0.0",0,0

OK
AT+CGSOCKCONT=1,"IP","rogers-core-appl1.apn"

OK
AT+CSOCKSETPN=1

OK
AT+CIPMODE=0

OK
AT+CIPSENDMODE=0

OK
AT+CIPCCFG=10,0,0,0,1,0,75000

OK
AT+CIPTIMEOUT=75000,15000,15000

OK
AT+NETOPEN

OK

+NETOPEN: 0 success
AT+NETOPEN?

+NETOPEN: 1,1

OK
AT+IPADDR

+IPADDR: 25.160.98.140

OK
GPRS connected
Connecting to vsh.pp.ua
AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
AT+CIPRXGET=1

OK
AT+CIPOPEN=0,"TCP","vsh.pp.ua",80

OK

+CIPOPEN: success
Performing HTTP GET request...
 0,0
[96305] ### Unhandled: 0,0
AT+CIPSEND=0,32

>GET /TinyGSM/logo.txt HTTP/1.1

OK

+CIPSEND: 0,32,32

AT+CIPSEND=0,17

>Host: vsh.pp.ua

OK

+CIPSEND: 0,17,17

AT+CIPSEND=0,21

>Connection: close

OK

+CIPSEND: 0,21,21

AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK

RECV FROM:212.237.59.104:80

+RECEIVE,0,633:

+CIPRXGET: 1,0

+IPCLOSE: 0,1
[96994] ### Closed:  0
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,633

OK
AT+CIPRXGET=2,0,633

+CIPRXGET: 2,0,633,0
HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Fri, 13 Mar 2020 11:16:00 GMT
Content-Type: text/plain; charset=UTF-8
Content-Length: 121
Connection: close
X-DNS-Prefetch-Control: off
X-Frame-Options: SAMEORIGIN
Strict-Transport-Security: max-age=15552000; includeSubDomains
X-Download-Options: noopen
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
Accept-Ranges: bytes
Cache-Control: public, max-age=0
Last-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936080"

  _____            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

OK
HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Fri, 13 Mar 2020 11:16:00 GMT
Content-Type: text/plain; charset=UTF-8
Content-Length: 121
Connection: close
X-DNS-Prefetch-Control: off
X-Frame-Options: SAMEORIGIN
Strict-Transport-Security: max-age=15552000; includeSubDomains
X-Download-Options: noopen
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
Accept-Ranges: bytes
Cache-Control: public, max-age=0
Last-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936080"

  _____            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
Server disconnected
AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
AT+NETCLOSE

+NETCLOSE: 9

ERROR
GPRS disconnected

SRGDamia1, 2020-03-13:
So that looks like it mostly worked, except for the close/disconnect. I'm not sure what's going on there. But you successfully got the website.

robg3987, 2020-03-15:
not sure either,, I've tested with my other Blynk sketches and it seems to work now. Thank you for your assistance!

closed 2020-03-15 by robg3987

SRGDamia1, 2020-03-16:
I'm glad it works!

AmalGuibene, 2023-05-15:
Hello, I am seeking assistance in connecting the FONA 3G SIM5320 module to the cloud via TCP connect to establish an MQTT connection. I have successfully connected the module using GPRS connectivity by executing the following AT commands: AT+CGATT=1, AT+CGSOCKCONT="IP', APN, and AT+NETOPEN="tcp",1883.

However, I am facing difficulties when attempting to establish a TCP connection using the command: AT+TCPCONNECT="test.mosquitto.org",1883. Unfortunately, my attempts have been unsuccessful thus far.

I would greatly appreciate any guidance or support you can offer in resolving this issue. If you have experience or expertise in working with the SIM5320 module and MQTT connectivity, your insights would be invaluable.

Please let me know if there are any additional details or information I can provide to assist in troubleshooting the problem. I am eager to make progress on this project and any help would be greatly appreciated.

Thank you for your time and consideration.

Best regards,
Amal