vshymanskyy/TinyGSM #371: MC60 is not working with this library, Can you please fix this issue?

shridattdudhat, 2020-03-15:
GPRS and GPS in MC60 module is not working with this library 

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)

### What are you working with?

Main processor board: 
Modem: 
TinyGSM version: 
Code: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### AT command log

SRGDamia1, 2020-03-16:
With that level of detail given with respect to "not working," no, I cannot fix it.

If there's something specific that isn't working, please fill out the template and explain in detail what the problem is and if possible what you believe the solution to be.

SRGDamia1, 2020-04-15:
I'm closing this due to lack of information.  Please re-open it with more information if you're still having issues.

closed 2020-04-15 by SRGDamia1

galizki, 2020-09-06:
I have found several issues with MC60 which make it not working for HTTP requests.
According to docs, the following is required:

1) It is needed to specify for Modem that it should work as 'client' in gprsConnectImpl:

```
    // Modem is used as a client
    sendAT(GF("+QISRVC=1"));
    if (waitResponse() != 1) { return false; }
```

2) In modemConnect, before opening the connection, it is required to specify how 'host' parameter should be treated: as a host of as an IP address:

```
    // By default, MC60 expects IP address as 'host' parameter. 
    // If it is a domain name, "AT+QIDNSIP=1" should be executed.
    // "AT+QIDNSIP=0" is for dotted decimal IP address.
    IPAddress addr;
    sendAT(GF("+QIDNSIP="), (addr.fromString(host)?0:1));
    if (waitResponse() != 1) { return false; }
```

3) During sending process 'mux' should be passed to +QISACK if multiply connections used:

`      sendAT(GF("+QISACK="), mux); // If 'mux' is not specified, MC60 returns 'ERRROR' (for QIMUX == 1)
`

4) And it is a problem with capturing '+QIRDI' in waitResponse. Responce contains additional parameter. So the proper code will be:

```
        } else if (data.endsWith(
                       GF(GSM_NL "+QIRDI:"))) {  // TODO(?):  QIRD? or QIRDI?
          // +QIRDI: <id>,<sc>,<sid>,<num>,<len>,< tlen>
          streamSkipUntil(',');  // Skip the context
          streamSkipUntil(',');  // Skip the role
          // read the connection id
          int8_t mux = streamGetIntBefore(',');
          // read the number of packets in the buffer
          int8_t num_packets = streamGetIntBefore(',');
          // read the length of the current packet
          streamSkipUntil(','); // Skip the length of the current package in the buffer
          int16_t len_total = streamGetIntBefore('\n'); // Total length of all packages
          if (mux >= 0 && mux < TINY_GSM_MUX_COUNT && sockets[mux] &&
              num_packets >= 0 && len_total >= 0) {
            sockets[mux]->sock_available = len_total;
          }
          data = "";
          // DBG("### Got Data:", len_total, "on", mux);
        } // <...>

```

I have tested the modified code (attached) with D-IOT 2560 PRO based on Arduino Mega 2560 with integrated MC60 and examples/WebClient/WebClient.ino works fine for me by using Serial3 to communicate with MC60 ('#define SerialAT Serial3') 

[ATTENTION!] BUT with one change in source code: it is required to comment out setting "Connection: close" header line. For some reason MC60 is not waiting data to and closes the connection immediately. So, one trink and the example works fine:

`//client.print("Connection: close\r\n\r\n");`

[TinyGsmClientMC60.h.zip](https://github.com/vshymanskyy/TinyGSM/files/5179805/TinyGsmClientMC60.h.zip)