vshymanskyy/TinyGSM #372: how is tinyGSM dealing with low signal strength

maromme, 2020-03-17:
Hi,
thanks for the nice lib. It's working fine for me at home, but I'm wondering how it's working with bad / changing signal strength. The place I'm using my project is known to me difficult GSM coverage.

Basically I'm using the modem just to send some status SMS.

My reduced code:
` modem.restart();
 String smsMessage = "test message";
    if (modem.sendSMS(SMS_TARGET, smsMessage)) {
      SerialMon.println(smsMessage);
    }
    else {
      SerialMon.println("SMS failed to send");
`
-> How long is it trying to send the SMS?
-> Are there timeouts in the TinyGSM lib?

I was considering adding the following after the modem.restart() to wait until there is a signal at all. I found out that it can take some time to get a signal at all (especially at bad places). Calling the modem.getSignalQuality() straight before the modem.sendSMS() results in a "0" after a couple of seconds in some value (depending on the signal strength.

`long curMillis = 0;                                   
  long preMillis = millis();                             
  int csq = modem.getSignalQuality();
  int max_wait = 6000;

  // wait for singnal strength or abort after max_wait
  while ((csq < 2) && (curMillis - preMillis < max_wait))
  {
    curMillis = millis();
    csq = modem.getSignalQuality();
    Serial.println(csq);
  }`

What do you think. Is that enough or shall I consider something else?

Thanks a lot,
Markus

SRGDamia1, 2020-03-18:
The sendSMS function doesn't do any "checking" to make sure you're connected to the network before attempting to send the message.  That function will make one attempt to send the message, and if it fails there is no retry.  You'd have to write up that logic yourself.  After each individual AT command sent to the modem there is a 'timeout' while waiting for a response for the modem - how long and whether there's a retry depends on the function, the manual/user guide/specifications, and some experimentation.  But for most commands the wait for a response from the modem is 1s and the command is not retried.

Have you used the```waitForNetwork()``` function?  It will do just that - hold everything and keep asking the modem for its connection status every 250ms until the SIM is successfully registered on the network.  With no argument it defaults to 60s, but you can add an argument with the number of ms to wait and set it to whatever you want (```waitForNetwork(120000L)```. 

It's actually possible to be registered on the network, send a message, and go back to sleep all while the signal being reported by ```getSignalQuality()``` (AT+CSQ) is still 0 - but waiting for a good value isn't a bad idea if you're more worried about getting the message out successfully than preserving battery.  Conversely, though, just checking for good signal is not enough - stellar signal does nothing if you're not registered on the network first.  Sometimes, especially with a new SIM or new location the network registration can take _forever_ (like.. 20+ minutes with a u-blox SARA R410M).  

Unfortunately... to figure out exactly what series of waits and queries is going to get the best success while wasting the least battery you'll probably have to do some testing with your real modules/SIM's/locations.  Most of what I've done is with the XBee's/u-blox R410 and the SIM800 using Hologram brand SIMs in the USA.

maromme, 2020-03-18:
Hi, thanks for your answer, I really appreciate that!
After your suggestions I'll try the `waitForNetwork()` and if there is a positiv result within a reasonalble time, I'll try the sendSMS() until success / manually implemented timeout. The last step ends with a not successfully send message anyway if the signal is too poor. There is then no point in checking the signal strength.
Thanks,
Markus

closed 2020-03-20 by maromme