vshymanskyy/TinyGSM #374: SIM800 modem.waitForNetwork never proceeds

ebenl, 2020-03-24:
## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
**[x ] Bug or issue with library functionality (ie, sending data over TCP/IP)**

### What are you working with?

**Main processor board: <SAMD21>
Modem: < SIM800 R14.18>
TinyGSM version: <0.10.1>
Code: < if (!modem.waitForNetwork(35000))>**

### Scenario, steps to reproduce

With Library 0.7.9 this function works correctly. 
With latest 0.10.1 this function loops

### Expected result
```
This is the result I get using 0.7.9 : ( this is the debug output )

Initializing modem...
Attempt no: 1
AT
?AT
ÿAT
AT
AT
AT
AT
AT
ÿ
OK
AT&FZ

OK
ATE0
ATE0
OK
AT+CPIN?

ERROR
AT+CPIN?

RDY

+CFUN: 1

ERROR
AT+CPIN?

+CPIN: READY

Call Ready

+CPIN: READY

OK
ATI

SIM800 R14.18

OK
Modem: SIM800 R14.18
Waiting for network...
AT+CREG?

SMS Ready

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,5

OK
success in connecting!!!
AT+CCLK?

+CCLK: "04/01/01,00:00:07+08"00:00:07+08
AT+CCLK?

OK

```

### Actual result
This is the result using 0.10.1 : (debug output)

```
Initializing modem...
Attempt no: 1
AT
?AT
ÿAT
AT
AT
AT
AT
AT
ÿ
OK
ATE0

OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

ERROR
AT+CPIN?

RDY

+CFUN: 1

+CREG: 0

ERROR
AT+CPIN?

+CPIN: READY

+CREG: 2

Call Ready

+CPIN: READY

OK
ATI

SIM800 R14.18

OK
Modem: SIM800 R14.18
Waiting for network...
AT+CREG?

SMS Ready

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

+CREG: 2,2

OK
AT+CREG?

*PSUTTZ: 2020,3,20,17,38,43,"+8",0

DST: 0

+CREG: 2,2

OK
AT+CREG?

+CREG: 5,"2E30","61EF"

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 5,"2E30","61EF"

+CREG: 5,"2E30","61EF"

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?

+CREG: 2,5,"2E30","61EF"

OK
AT+CREG?
```

.....looping forever....

SRGDamia1, 2020-03-25:
Huh.  That's odd.  Does it make a difference if you set CREG back to 0?  Unfortunately, I don't have a SIM800 home with me and will be working from home for the foreseeable future.

ebenl, 2020-03-25:
Thanks for the reply.

Setting CREG to 0 (or 1) right after modem.init() fixes the issue in 0.10.1

closed 2020-03-25 by ebenl

SRGDamia1, 2020-03-25:
Ok, good to know.  Thanks!