vshymanskyy/TinyGSM #375: TTGO T-SIM7000G problem with TinyGSM v0.10.1

Dr-PV, 2020-03-27:
Hi,

I am using TinyGSM with TTGO T-Call (SIM800) and very satisfied with the library. The library is a very nice works.

Recently, I tried TinyGSM v0.7.9 on TTGO T-SIM7000G board without any problems connecting to GSM network with the example sketch "AllFunctions.ino". However, when I updated the library to v0.10.1, the same sketch could not connect to the GSM network and kept looping on trying to connect.

Please check if this is the issue with the library v0.10.1 or any modifications needed to apply to the sketch.

Thank you.

SRGDamia1, 2020-04-15:
I think this is probably the same issues as #381

SRGDamia1, 2020-04-15:
I've updated the SIM7000 to check both EPS and GPRS status (0.7.9 used only GPRS, 0.10.1 only EPS).  I think that may solve your issue.  Can you try again and close the issue if it worked?

Dr-PV, 2020-04-15:
> I've updated the SIM7000 to check both EPS and GPRS status (0.7.9 used only GPRS, 0.10.1 only EPS). I think that may solve your issue. Can you try again and close the issue if it worked?

Thank you very much.
The network connection in version 0.10.1 works OK.

(I do not know if I should start a new issue or not for the following.)

However, there are two new problems in version 0.10.1:
1.   String MODEMInfo = MODEM.getModemInfo () ; 
      sometimes returns a null string (no MODEM name returned),
2.   MODEM.restart () ;
      takes a very long time to initialize the MODEM, about 63 seconds. The version 0.7.9 takes only 8 seconds to initialize the MODEM.

Again, I do not know if the problems are specific to TTGO T-SIM7000G or some special codes are needed for this particular board. Anyway, TTGO T-SIM7000G board seems to be a very nice board and very easy to work with because several necessary components are integrated on one board.

Please look into the new issues above.

Thank you.

SRGDamia1, 2020-04-15:
1 - If the modem doesn't respond.. it doesn't respond.  The command hasn't changed.  If you are seeing a response in your AT log and the library isn't parsing it, please post an example AT log.

2 - The library now waits for the modem to completely finish booting and spit out an "SMS Ready" message.  Previously it just waited a few seconds after the modem rebooted and then went on whether the SIM7000 was ready or not.  Maybe you're not getting the ready message.  Please post an AT log of the reboot.

Dr-PV, 2020-04-19:
These are the AT dumps from v0.7.9 and v0.10.1. Both of them were executed on TTGO T-SIM7000G board which shows the big differences in total network connecting time: 15 seconds for 0.7.9 vs. 63 seconds for v0.10.1.

Thnk you.

SRGDamia1, 2020-04-19:
Yeah, you're modem just isn't saying "ready" at the end (or, rather, I'm guessing it's saying it at a different baud rate given the junk you are seeing as it boots).  The library is waiting a whole minute for that "ready", hence the long delay.  Strange things happen if you try to command the modem before it's ready, but a full minute is probably overkill.  I'll drop that wait down to 10s, which I think is enough, with the caveat that I might bump it up again if someone else has issues with theirs not being ready.  The 10s is still longer than the 3s that 0.7.9 had, but, as you saw from occasional strange things happening in your attempts with that, 3s wasn't enough.

SRGDamia1, 2020-04-19:
The issue with the modem name didn't appear in your logs - are you still getting it regularly?

Dr-PV, 2020-04-25:
The library is working beautifully.

Thank you.

closed 2020-04-25 by Dr-PV