vshymanskyy/TinyGSM #378: No location with ESP32 and SIM800L

albertomercurio, 2020-04-06:
## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ x] Bug or issue with library functionality (ie, sending data over TCP/IP)

### What are you working with?

Main processor board:  ESP32
Modem:  SIM800L
TinyGSM version:  0.10.1
Code:  AllFunctions

### Scenario, steps to reproduce

I'm using the TTGO T-CALL board, with ESP32 and SIM800L integrated. The function modem.getGsmLocation() give always false if i set the TINY_GSM_TEST_SSL as true. But if if set SSL to false my location is fixed correctly.
Why this happen? 
I'm Using the Things Mobile simcard.

SRGDamia1, 2020-04-06:
No clue.  I don't have a SIM800 model that supports SSL. Post the AT log.

albertomercurio, 2020-04-06:
How can i do that?
However i noticed that sometimes also with ssl setted to false it doesn't take the location. Is it normal?

SRGDamia1, 2020-04-07:
Look at any of the examples and use the dump AT code.

albertomercurio, 2020-04-08:
```
[42] Wait...
[6042] Initializing modem...
AT
AT

OK
AT&W
AT&W

OK
AT+CFUN=0
AT+CFUN=0

+CPIN: NOT READY

OK
AT+CFUN=1,1
AT+CFUN=1,1

OK
[9312] ### TinyGSM Version: 0.10.1
[9312] ### TinyGSM Compiled Module:  TinyGsmClientSIM800
AT
⸮AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+GMM

SIMCOM_SIM800L

OK
[9627] ### Modem: SIMCOM SIM800L
[9627] ### Modem: SIMCOM SIM800L
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CME ERROR: CFUN state is 0 or 4
AT+CPIN?

+CPIN: READY

OK
AT+GMM

SIMCOM_SIM800L

OK
[10654] ### Modem: SIMCOM SIM800L
[10654] Modem Name: SIMCOM SIM800L
ATI

SIM800 R14.18

OK
[10661] Modem Info: SIM800 R14.18
AT+CPIN?

+CPIN: READY

OK
AT+CPIN="1503"

+CME ERROR: operation not allowed
[10676] Waiting for network...
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,5

OK
AT+CREG?

+CREG: 0,5

OK
[17319] Network connected
[17319] Connecting to 
AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN",""

OK
AT+CGDCONT=1,"IP",""

OK
AT+CGACT=1,1

+CTZV: +8,1
[22508] ### Network time zone updated.

*PSUTTZ: 2020,4,8,9,47,3,"+8",1
[22513] ### Network time and time zone updated.

DST: 1
[22514] ### Daylight savings time state updated.

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.193.17.168"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

10.193.17.168

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

10.193.17.168

OK
[23192] GPRS status: connected
AT+CCID

8944502312190250939f

OK
[24193] ### Unhandled: 8944502312190250939f

OK
[24193] CCID: 
AT+GSN

863576047331942

OK
[24200] IMEI: 863576047331942
AT+CIMI

234500027025093

OK
[24205] IMSI: 234500027025093
AT+COPS?

+COPS: 0,0,"Wind Telecom SpA"

OK
[24213] Operator: Wind Telecom SpA
AT+CIFSR;E0

10.193.17.168

OK
[24221] Local IP: 10.193.17.168
AT+CSQ

+CSQ: 20,0

OK
[24227] Signal quality: 20
AT+CMGF=1

OK
AT+CSCS="HEX"

OK
AT+CUSD=1,"*111#"

OK

+CUSD: 2
[31005] Balance (USSD): 
AT+CMGF=1

OK
AT+CSCS="HEX"

OK
AT+CUSD=1,"*161#"

OK

+CUSD: 2
[37794] Phone number (USSD): 
[37794] Connecting to  vsh.pp.ua
AT+CIPCLOSE=1,1

+CME ERROR: operation not allowed
AT+CIPSSL=1

OK
AT+CIPSTART=1,"TCP","vsh.pp.ua",443

OK

1, CLOSE OK
[51861] ... failed
[51861] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 2

OK
[53851] Couldn't get GSM location, retrying in 15s.
[68852] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 3

OK
[70031] Couldn't get GSM location, retrying in 15s.
[85031] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 3

OK
[86231] Couldn't get GSM location, retrying in 15s.
[101231] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 3

OK
[102430] Couldn't get GSM location, retrying in 15s.
[117430] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 3

OK
[118607] Couldn't get GSM location, retrying in 15s.
[133607] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 0,15.763547,38.255439,400,20/04/08,11:48:56

OK
[135516] Latitude: 15.76354694 	Longitude: 38.25543976
[135516] Accuracy: 400.00
[135518] Year: 2020 	Month: 4 	Day: 8
[135521] Hour: 11 	Minute: 48 	Second: 56
[135525] Retrieving GSM location again as a string
AT+CLBS=1,1

+CLBS: 0,15.763547,38.255439,400

OK
[135538] GSM Based Location String: 15.763547,38.255439,400
[135540] Requesting current network time
AT+CCLK?

+CCLK: "20/04/08,11:48:56+08"

OK
[135553] Year: 2020 	Month: 4 	Day: 8
[135553] Hour: 11 	Minute: 48 	Second: 56
[135555] Timezone: 2.00
[135557] Retrieving time again as a string
AT+CCLK?

+CCLK: "20/04/08,11:48:56+08"

OK
[135568] Current Network Time: 20/04/08,11:48:56+08
AT+CIPSHUT

SHUT OK
AT+CGATT=0

+SAPBR 1: DEACT

OK
AT+CGATT?

+CGATT: 0

OK
[140875] GPRS disconnected
AT+CBC

+CBC: 0,77,4021

OK
[140880] Battery charge state: 0
[140880] Battery charge 'percent': 77
[140881] Battery voltage: 4.02
[140881] End of tests.

```

SRGDamia1, 2020-04-08:
It looks like you did get the location, just not on the first try. I think that's normal; it takes a while to "settle" on a location which is why that program tries several times with a wait between.

I'm not sure what went wrong with you TCP connection in there, though. It looks like it closed before getting the response. Has that been happening with other sites?

albertomercurio, 2020-04-08:
I precise that PCP connection works, SSL doesn't work, also with another server:
```
[37918] Connecting to  albertomercurio.it
AT+CIPCLOSE=1,1

+CME ERROR: operation not allowed
AT+CIPSSL=1

OK
AT+CIPSTART=1,"TCP","albertomercurio.it",443

OK

1, CLOSE OK
[40064] ... failed

```

SRGDamia1, 2020-04-08:
Hm.  It's like it's closing the connection as soon as it is open.  Unfortunately, I just don't have a SIM800 variant with SSL support myself to test so the best I can do is ask you to make changes and keep posting the AT log for me.

First thing it to try removing the "stop" from the secure connect function.  Open TinyGsmClientSIM800.h and comment out line **_135_** ```stop();``` in the GsmClientSecureSim800 connect function.  (Make sure you're in the client secure section!)  The intent or that particular ```stop()``` is to always make sure the socket is closed and thus available to make a connection before attempting the connection.  But I've seen it happen with u-blox modules that if you call stop() when there wasn't actually an open connection it waits until you've made a connection later and then "helpfully" immediately stops that connection.  It looks like your SIM800 might be doing something similar.  

Are you ok with the location issue?

albertomercurio, 2020-04-08:
Failed again. 
```
[24369] Operator: Wind Telecom SpA
AT+CIFSR;E0

10.193.162.143

OK
[24377] Local IP: 10.193.162.143
AT+CSQ

+CSQ: 21,5

OK
[24382] Signal quality: 21
AT+CMGF=1

OK
AT+CSCS="HEX"

OK
AT+CUSD=1,"*111#"

OK

+CUSD: 2
[31263] Balance (USSD): 
AT+CMGF=1

OK
AT+CSCS="HEX"

OK
AT+CUSD=1,"*161#"

OK

+CUSD: 2
[38145] Phone number (USSD): 
[38145] Connecting to  vsh.pp.ua
AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=0

OK
AT+CIPSTART=0,"TCP","vsh.pp.ua",80

OK

0, CONNECT OK
AT+CIPSEND=0,32

>GET /TinyGSM/logo.txt HTTP/1.0

DATA ACCEPT:0,32
AT+CIPSEND=0,17

>Host: vsh.pp.ua

DATA ACCEPT:0,17
AT+CIPSEND=0,21

>Connection: close

DATA ACCEPT:0,21
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","212.237.59.104","80","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","212.237.59.104","80","CONNECTED"

OK

+CIPRXGET: 1,0
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,633

OK
AT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,570
HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Wed, 08 A
OK
HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Wed, 08 AAT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,507
pr 2020 20:53:03 GMT
Content-Type: text/plain; charset=UTF-8

OK
pr 2020 20:53:03 GMT
Content-Type: text/plain; charset=UTF-8
AT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,444
Content-Length: 121
Connection: close
X-DNS-Prefetch-Control:
OK
Content-Length: 121
Connection: close
X-DNS-Prefetch-Control:AT+CIPRXGET=4,0

+CIPRXGET: 4,0,444

OK
AT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,381
 off
X-Frame-Options: SAMEORIGIN
Strict-Transport-Security: m
OK
 off
X-Frame-Options: SAMEORIGIN
Strict-Transport-Security: mAT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,318
ax-age=15552000; includeSubDomains
X-Download-Options: noopen

OK
ax-age=15552000; includeSubDomains
X-Download-Options: noopen
AT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,255

X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=blo
OK

X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=bloAT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,192
ck
Accept-Ranges: bytes
Cache-Control: public, max-age=0
Las
OK
ck
Accept-Ranges: bytes
Cache-Control: public, max-age=0
LasAT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,129
t-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936
OK
t-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936AT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,66
080"

  _____            _____  _____  _____
    |  | |\ | \
OK
080"

  _____            _____  _____  _____
    |  | |\ | \AT+CIPRXGET=2,0,63

+CIPRXGET: 2,0,63,3
_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  
OK
_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  AT+CIPRXGET=2,0,3

+CIPRXGET: 2,0,3,0
|

OK
|

AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","212.237.59.104","80","REMOTE CLOSING"

OK
AT+CIPCLOSE=0,1

0, CLOSE OK
[41421] Connecting to  vsh.pp.ua
AT+CIPSSL=1

OK
AT+CIPSTART=1,"TCP","vsh.pp.ua",443

OK

1, CLOSE OK
[53817] ... failed
[53817] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 1

OK
[55849] Couldn't get GSM location, retrying in 15s.
[70850] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 1

OK
[72889] Couldn't get GSM location, retrying in 15s.
[87889] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 2

OK
[120206] Couldn't get GSM location, retrying in 15s.
[135207] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 3

OK
[136487] Couldn't get GSM location, retrying in 15s.
[151487] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 3

OK
[152788] Couldn't get GSM location, retrying in 15s.
[167788] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 1

OK
[169809] Couldn't get GSM location, retrying in 15s.
[184809] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 3

OK
[186089] Couldn't get GSM location, retrying in 15s.
[201089] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 3

OK
[202367] Couldn't get GSM location, retrying in 15s.
[217367] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 2

OK
[249683] Couldn't get GSM location, retrying in 15s.
[264683] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 3

OK
[265985] Couldn't get GSM location, retrying in 15s.
[280985] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 2

OK
[313301] Couldn't get GSM location, retrying in 15s.
[328301] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 1

OK
[330349] Couldn't get GSM location, retrying in 15s.
[345350] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 1

OK
[347389] Couldn't get GSM location, retrying in 15s.
[362389] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 2

OK
[394705] Couldn't get GSM location, retrying in 15s.
[409705] Requesting current GSM location
AT+CLBS=4,1

+CLBS: 2

OK
[442023] Couldn't get GSM location, retrying in 15s.
[457022] Retrieving GSM location again as a string
AT+CLBS=1,1

+CLBS: 3

OK
[458326] GSM Based Location String: 
[458326] Requesting current network time
AT+CCLK?

+CCLK: "20/04/08,23:00:00+08"

OK
[458334] Year: 2020 	Month: 4 	Day: 8
[458334] Hour: 23 	Minute: 0 	Second: 0
[458336] Timezone: 2.00
[458338] Retrieving time again as a string
AT+CCLK?

+CCLK: "20/04/08,23:00:00+08"

OK
[458352] Current Network Time: 20/04/08,23:00:00+08
AT+CIPSHUT

SHUT OK
AT+CGATT=0

+SAPBR 1: DEACT

OK
AT+CGATT?

+CGATT: 0

OK
[463711] GPRS disconnected
AT+CBC

+CBC: 0,79,4032

OK
[463717] Battery charge state: 0
[463717] Battery charge 'percent': 79
[463717] Battery voltage: 4.03
[463717] End of tests.

```
Do i remove the comment on `stop();`?
This time location didn't work. Sometimes it work, but sometimes not.

SRGDamia1, 2020-04-08:
Boo, removing that stop doesn't seem to have made the slightest difference.  You might as well put it back.

For location, I think the module just needs to be connected to the network for a fairly long time before the location works.  I honestly haven't had much success with it.  2G is dieing in the US and where I am I didn't have many places where I could get much signal to test with even before we were all restricted to stay-at-home.

I don't have a next suggestion; I'll have to think on it and google around.  Are you absolutely sure your module has the right firmware for SSL?  On mine I get "ERROR" back for any SSL commands I attempt, so I'd guess the fact anything happens for you might indicate it works. but I know most people's success with SSL on the SIM800 is very firmware dependent.

albertomercurio, 2020-04-08:
I should have the latest version 14.18. What sim module do you recommend?

SRGDamia1, 2020-04-09:
I'm sorry, I don't have any recommendations ready.  I would suggest you google around for other people's results with that particular firmware.  The library is using the documented commands to use SSL on the TCP connection; I really have no idea why it's not working.

gorghino, 2021-05-21:
Ciao @albertomercurio , did you eventually find a solution?

ilgarbenli, 2023-12-06:
could you solve this problem. I'm fcing with same problem. Need help

kakka0903, 2023-12-11:
> could you solve this problem. I'm facing with same problem. Need help

I never got this board to function as I desired. If I recall correctly, there are issues with the board design which should be documented in the issues section of this repo. I bought the [Arduino MKR GSM 1400](https://store.arduino.cc/products/arduino-mkr-gsm-1400) instead and that was way more reliable.