vshymanskyy/TinyGSM #379: HttpClient request does not work for SIM5320 (Fatal exception)

hubaksis, 2020-04-09:
## What type of issues is this?
[x ] Bug or issue with library functionality (ie, sending data over TCP/IP)

### What are you working with?

Main processor board: ESP8266 (Wemod D1 mini Pro)
Modem: SIM5320
TinyGSM version: 0.10.1
Code: TinyGSM/examples/HttpClient/

### Scenario, steps to reproduce
Change the example like the steps below
1.  #define TINY_GSM_MODEM_SIM5360
2. #include <SoftwareSerial.h>
SoftwareSerial SerialAT(D2, D4);

3.  #define DUMP_AT_COMMANDS
4. Define your apn.

### Expected result
The request is sent

### Actual result
Exception

### AT command log
```SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
Wait...
Initializing modem...
AT

OK
AT+REBOOT

OK
[6181] ### TinyGSM Version: 0.10.1
[6181] ### TinyGSM Compiled Module:  TinyGsmClientSIM5360
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+CGMM

SIMCOM_SIM5320E

OK
[6324] ### Modem: SIMCOM SIM5320E
[6325] ### Modem: SIMCOM SIM5320E
AT+CTZR=0

OK
AT+CTZU=1

OK
AT+CPIN?

+CPIN: READY

OK
ATI

Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM5320E
Revision: SVI0
+,K
[7452] ### Unhandled: Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM5320E
Revision: SVI0
+,K
Modem Info: 
AT+CPIN?

+CPIN: READY

OK
Waiting for network...AT+CGREG?

+CGREG: 0,1

OK
 success
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to warehouseAT+CIPCLOSE=0

+IP ERROR: Network not opened

ERROR
AT+NETCLOSE

+IP ERROR: Network is already closed

ERROR
AT+CGDCONT=1,"IP","warehouse","0.0.0.0",0,0

OK
AT+CGSOCKCONT=1,"IP","warehouse"

OK
AT+CSOCKSETPN=1

OK
AT+CIPMODE=0

OK
AT+CIPSENDMODE=0

OK
AT+CIPCCFG=10,0,0,0,1,0,75000

OK
AT+CIPTIMEOUT=75000,15000,15000

OK
AT+NETOPEN

OK

+NETOPEN: 1

START

+CPIN: READY

SMS DONE

+VOICEMAIL: INIT_STATE, 0, 0

PB DONE
[83094] ### Unhandled: OK

+NETOPEN: 1

START

+CPIN: READY

SMS DONE

+VOICEMAIL: INIT_STATE, 0, 0

PB DONE
 fail
Waiting for network...AT+CGREG?

+CGREG: 0,1

OK
 success
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to warehouseAT+CIPCLOSE=0

+IP ERROR: Network not opened

ERROR
AT+NETCLOSE

+IP ERROR: Network is already closed

ERROR
AT+CGDCONT=1,"IP","warehouse","0.0.0.0",0,0

OK
AT+CGSOCKCONT=1,"IP","warehouse"

OK
AT+CSOCKSETPN=1

OK
AT+CIPMODE=0

OK
AT+CIPSENDMODE=0

OK
AT+CIPCCFG=10,0,0,0,1,0,75000

OK
AT+CIPTIMEOUT=75000,15000,15000

OK
AT+NETOPEN

OK

+NETOPEN: 0 success
AT+NETOPEN?

+NETOPEN: 1,1

OK
AT+IPADDR

+IPADDR: 100.78.7.129

OK
GPRS connected
Performing HTTP GET request... AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
AT+CIPRXGET=1

OK
AT+CIPOPEN=0,"TCP","vsh.pp.ua",80

OK

+CIPOPEN: 0,0
[96488] ### Unhandled: 0,0
AT+CIPSEND=0,3

>GET
OK

+CIPSEND: 0,3,3

AT+CIPSEND=0,1

> 
OK

+CIPSEND: 0,1,1

AT+CIPSEND=0,17

>/TinyGSM/logo.txt
OK

+CIPSEND: 0,17,17

AT+CIPSEND=0,9

> HTTP/1.1
OK

+CIPSEND: 0,9,9

AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2

AT+CIPSEND=0,6

>Host: 
OK

+CIPSEND: 0,6,6

AT+CIPSEND=0,9

>vsh.pp.ua
OK

+CIPSEND: 0,9,9

AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2

AT+CIPSEND=0,10

>User-Agent
OK

+CIPSEND: 0,10,10

AT+CIPSEND=0,2

>: 
OK

+CIPSEND: 0,2,2

AT+CIPSEND=0,13

>Arduino/2.2.0
OK

+CIPSEND: 0,13,13

AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2

AT+CIPSEND=0,10

>Connection
OK

+CIPSEND: 0,10,10

AT+CIPSEND=0,2

>: 
OK

+CIPSEND: 0,2,2

AT+CIPSEND=0,5

>close
OK

+CIPSEND: 0,5,5

AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2

AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0Fatal exception 29(StoreProhibitedCause):
epc1=0x40201990, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000001c, depc=0x00000000

Exception (29):
epc1=0x40201990 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000001c depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffcf0 end: 3fffffc0 offset: 01a0
3ffffe90:  3ffee820 3ffeead8 3ffeeae8 00000000  
3ffffea0:  00000001 00000000 4bc6a7f0 00000000  
3ffffeb0:  40105b95 01d3000b 40100604 00000000  
3ffffec0:  3ffee820 3ffeead4 00000000 402028c1  
3ffffed0:  01ba84f7 3ffeec10 00000002 40204097  
3ffffee0:  40204655 00000002 00000000 00000000  
3ffffef0:  0000000d 3ffee840 3ffee820 4020296d  
3fffff00:  00000000 00017eb6 3ffee7c8 40203702  
3fffff10:  01ba828f 3ffe8986 3ffe8a0d 40204c38  
3fffff20:  40204c2c 3ffeed48 3ffe87fe 3ffee7c8  
3fffff30:  00000000 00017eb6 3ffe8a27 40203b7c  
3fffff40:  3ffe89a4 3ffeead4 3ffeed48 3ffee7c8  
3fffff50:  3ffee7c8 3ffeead4 3ffeed48 40202769  
3fffff60:  007a1200 bbe89e03 feefef00 feefeffe  
3fffff70:  00000000 0000005f 80efeffe feefef00  
3fffff80:  00000000 00000000 00000001 40100558  
3fffff90:  3fffdad0 00000000 3ffeeb74 3ffeebb4  
3fffffa0:  3fffdad0 00000000 3ffeeb74 402041b0  
3fffffb0:  feefeffe feefeffe 3ffe84f0 401015b1  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
```

SRGDamia1, 2020-04-09:
No, I've never seen that before. Google tells me that it's an ESP problem trying to access an invalid register. This library definitely doesn't touch the esp's registers directly, so I'm guessing there's a core problem with your board. Are you using the latest Arduino core? Is it possible that you have remnants of an old firmware flash hiding somewhere in your esp's memory?

Sometimes weird issues can also be caused by unstable power, particularly at peak draw moments like connecting and receiving data. Most 2G and 3G module specifications require a stable **2 amp** power supply, which is quite a bit (4x) what most Arduino style boards can supply. From my experience, you really can't get away with anything less.

hubaksis, 2020-04-10:
Thank you for the advice.
I've tried to go back to different cores: 2.4.1, 2.4.2, 2.6.2, 2.6.3 - the result is the same.

I'll try to add more debug messages to try to find the correct spot when this happens.

The power supply is ok (external, up to 5A) and everything works find with a bit different code, but I would like to try TinyGSM for my needs.

Thank you.

hubaksis, 2020-04-10:
Found the issue.

Firstly, when working with SIM5320, you need to wait for the PB DONE response after the modem restarts. Can't find it in docs, but I noticed that it can be busy and unresponsive before this response.
```
    bool restartImpl() {
        if (!testAT()) { return false; }
        sendAT(GF("+REBOOT"));
        if (waitResponse(10000L) != 1) {
            return false;
        }
        //delay(3000L);  // TODO(?):  Test this delay!

        if (waitResponse(40000L, GF(GSM_NL "PB DONE")) != 1) {
            DBG("### No PB DONE:");
            return false;
        }

        return init();
    }
```
Secondly, the issue above happens in 'modemGetConnected' method here:
```
for (int muxNo = 0; muxNo < TINY_GSM_MUX_COUNT; muxNo++) {
        // +CIPCLOSE:<link0_state>,<link1_state>,...,<link9_state>
        sockets[muxNo]->sock_connected = stream.parseInt();
}
```

As soon as I commented the line with stream.parseInt(), it started to work.

In the log above you can see that it reads only 2 numbers before the exception:
```
+CIPCLOSE: 1,0Fatal exception....
```
Maybe the stream is empty? 
Any advise? 

Thank you.

SRGDamia1, 2020-04-15:
I really don't know how this library could be causing that fatal exception.  All it is doing is using basic functions of the Arduino stream class. If those aren't working, that's a deeper problem than the library can handle.  I'll dig around more though and make sure I don't have a messed up mux number causing a null pointer or something bad of that sort.  

I've never touched a real SIM5320, just read the manual and helped other people with it, so I'll take your word for the ```PB DONE``` message and add a wait for it.

hubaksis, 2020-04-15:
I've narrowed the issue to the sockets array. Maybe it is not initialized as setting TINY_GSM_MUX_COUNT to 1 solves the issue.
I'll post an update if I find anything.

SRGDamia1, 2020-04-15:
Aha! I think it is a null pointer problem on my side.  Can you try the most recent version and see if it's fixed?

capedra, 2023-09-10:
@hubaksis is right about waiting for ```PB DONE``` or even ```PB DONEDS``` just in case you use two SIM cards in a DUAL SIM modem... because i was facing this same issue whenever I tried to subscribe to some MQTT topics just after I got the URC ```+CGEV: EPS PDN ACT 1``` and the pub/sub never happened, so, I waited for PB DONEDS and it goes alright as I'm currently testing it. For the record, I'm using this modem called A7670 just in case anyone wanna know.

capedra, 2023-10-08:
UPDATE: I actually decided to give up on MQTT and now I'm currently using UDP instead of TCP, which seems to work faster and reliable on some mobile operators in my country.