vshymanskyy/TinyGSM #380: GPRS disconected after exactly 2 minutes

adrians1990, 2020-04-12:
Im using MQTT example from TinyGSM for a SIM800L with a ESP8266. I tested with 2 SIM's of differents operators.  One of them works fine , but the other works fine during exactly 2 minutes. When I tell that works fine means that I can connect to my broker and send and receive messages.  After this time, GPRS conection is losed and the only way that I find is connect again to GPRS, but I can't undertand what is happening, why conection is disconnected after 2 minutes and for another SIM of another red operator works fine.

SRGDamia1, 2020-04-14:
Are you able to view/set/change preferences for your account tied to the mis-behaving sim?  It's possible the operator is automatically dropping the connection after a certain time period.

adrians1990, 2020-04-14:
Dear Sara, I can't modify any parameters about this SIM. I thought maybe some operators require a kind of command in order to take GPRS conection alive, but this is only my opinion.

SRGDamia1, 2020-04-15:
If there are commands needed for individual operators to keep the connection open, I don't know anything about them and this library isn't using them.  PubSubClient (the MQTT library being used in that example) sends a ping every 15s which should keep the connection alive.  You can try adjusting that by defining the keep alive to some number bigger or smaller than 15 (```#define MQTT_KEEPALIVE ##```) to see if that keeps the connection open.  If your specific operator requires more, I can't really help you with that.

Mr-HaleYa, 2020-07-29:
@adrians1990 Any news? 

@SRGDamia1 This can most likely be closed since it more of an operator error than TinyGSM. if problem comes back it can be reopened