vshymanskyy/TinyGSM #381: SIM7000 Not working in V0.10.1, TEENSY 3.6

Gigabyte1973, 2020-04-12:
I Have used TinyGSM for my SIM7000 modem for some time now, and a strange issue with the 0.10.1 Version of the lib occured.
If i configure as SIM7000 in my code I cant connect to the network, BUT if i change to SIM800 in the CODE, but still using the SIM7000 Modem hardware, IT WORKS !?!?!

**If i change back to TinyGSM lib V0.7.9, and configure as SIM7000 in the code it works, like it allways have done**

[image]

[image]
[image]

SRGDamia1, 2020-04-14:
Many of the commands for the SIM800 and the SIM7000 are similar.

You should only have one modem defined - you need to remove the line for the ESP8266.

Can you please enable debugging for TinyGSM and post the printout?
```#define TINY_GSM_DEBUG SerialMon```

It might also be helpful to look at the full AT log.  
Make sure you have [StreamDebugger](https://github.com/vshymanskyy/StreamDebugger) installed then replace ```TinyGsm modem(SerialAT);``` with 
```cpp
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
```

Gigabyte1973, 2020-04-14:
I have removed The 8266 include, did not change anything.
Log of both the WORKING SIM800 include, and the NON-WORKING SIM7000 include.

Br Rasmus

[image]
[image]

[AT modem log, set as SIM800.txt](https://github.com/vshymanskyy/TinyGSM/files/4477238/AT.modem.log.set.as.SIM800.txt)
[AT modem log set as SIM7000.txt](https://github.com/vshymanskyy/TinyGSM/files/4477239/AT.modem.log.set.as.SIM7000.txt)

SRGDamia1, 2020-04-15:
Hm.  Between the versions I switched the SIM7000 from querying for GPRS (ie 2G/EDGE) service registration to querying for EPS (ie 4G/LTE) registration status.  It looks like (whether because of the service available, the SIM card, or your board's mood) you're getting an "unknown" status for EPS connection but positive for GPRS.  That's why the SIM800's very similar (but GPRS only) commands work.  I will change the SIM7000 to accept GPRS or EPS - because it really should.  And chances are it really *is* using EPS not GPRS but is feeling moody about displaying registration status.  I haven't had that happen with a SIM7000, but it happens all the time for me with the u-blox SARA R410M.

SRGDamia1, 2020-04-15:
Please test if the newest version works and close this accordingly.

Gigabyte1973, 2020-04-15:
Hi again, thank you fot the great work, but.. New error

I have tested the new version first and then instyalled the old 0.79 both
log's is here.

0.10.1 (15-04 new version)

Initializing modem...

AT+CFUN=0

+CPIN: NOT READY

[14782] ### Unhandled: +CPIN: NOT READY

ATI

OK

Modem:

AT+CPIN="�"

+CME ERROR: SIM failure

[14880]

    ___  __          __

   / _ )/ /_ _____  / /__

  / _  / / // / _ \/  '_/

 /____/_/\_, /_//_/_/\_\

        /___/ v0.6.1 on Teensy 3.6

[14880] Modem init...

[14880] ### TinyGSM Version: 0.10.1

[14880] ### TinyGSM Compiled Module:  TinyGsmClientSIM7000

AT

OK

ATE0

OK

AT+CMEE=2

OK

AT+GMM

SIMCOM_SIM7000E

OK

[14891] ### Modem: SIMCOM SIM7000E

AT+CLTS=1

OK

AT+CBATCHK=1

OK

AT+CPIN?

+CME ERROR: SIM failure

AT+CPIN?

+CME ERROR: SIM failure

AT+CPIN?

+CME ERROR: SIM failure

AT+CPIN?

+CME ERROR: SIM failure

AT+CPIN?

+CME ERROR: SIM failure

AT+CPIN?

+CME ERROR: SIM failure

AT+CPIN?

+CME ERROR: SIM failure

AT+CPIN?

+CME ERROR: SIM failure

AT+CPIN?

+CME ERROR: SIM failure

AT+CPIN?

+CME ERROR: SIM failure

[24948] Cannot init

*NEXT PART IS WITH lib 0.79 Tested after the New test version.*

Initializing modem...

AT

OK

AT+CLTS=1

OK

AT+CFUN=0

OK

AT+CFUN=1,1

OK

[7799] ### TinyGSM Version: 0.7.9

AT

RDY

[13670] ### Unhandled: RDY

AT

AT

OK

ATE0

ATE0

OK

AT+GMM

SIMCOM_SIM7000E

OK

[13780] ### Modem: SIMCOM SIM7000E

[13780] ### Modem: SIMCOM SIM7000E

AT+CPIN?

ERROR

AT+CPIN?

+CFUN: 1

+CPIN: READY

+CPIN: READY

OK

ATI

SIM7000E R1351

OK

Modem: SIM7000E R1351

AT+CPIN="� F �;� ! H ��0F ��� F0F ��   � �"

ERROR

[14887]

    ___  __          __

   / _ )/ /_ _____  / /__

  / _  / / // / _ \/  '_/

 /____/_/\_, /_//_/_/\_\

        /___/ v0.6.1 on Teensy 3.6

[14887] Modem init...

[14887] ### TinyGSM Version: 0.7.9

AT

OK

ATE0

OK

AT+GMM

SIMCOM_SIM7000E

OK

[14894] ### Modem: SIMCOM SIM7000E

[14894] ### Modem: SIMCOM SIM7000E

AT+CPIN?

+CPIN: READY

OK

AT+CPIN?

SMS Ready

+CPIN: READY

OK

[14901] Connecting to network...

AT+CGREG?

+CGREG: 0,0

OK

AT+CGREG?

+CGREG: 0,0

OK

AT+CGREG?

+CGREG: 0,0

OK

AT+CGREG?

+CGREG: 0,0

OK

AT+CGREG?

+CGREG: 0,0

OK

AT+CGREG?

+CGREG: 0,0

OK

AT+CGREG?

+CGREG: 0,0

OK

AT+CGREG?

+CGREG: 0,0

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,5

OK

AT+COPS?

+COPS: 1,0,"3 3",0

OK

[21758] Network: 3 3

[21758] Connecting to data.tre.dk ...

AT+CIPSHUT

SHUT OK

AT+CGATT=0

DST: 1

*PSUTTZ: 20/04/15,21:00:53","+08",1

OK

AT+SAPBR=3,1,"Contype","GPRS"

OK

AT+SAPBR=3,1,"APN","data.tre.dk"

OK

AT+CGDCONT=1,"IP","data.tre.dk"

OK

AT+CGACT=1,1

DST: 1

*PSUTTZ: 20/04/15,21:00:56","+08",1

OK

AT+SAPBR=1,1

OK

AT+SAPBR=2,1

+SAPBR: 1,1,"109.56.24.70"

OK

AT+CGATT=1

OK

AT+CIPMUX=1

OK

AT+CIPQSEND=1

OK

AT+CIPRXGET=1

OK

AT+CSTT="data.tre.dk","",""

OK

AT+CIICR

OK

AT+CIFSR;E0

109.56.24.70

OK

[25771] Connected to GPRS

AT+CIPCLOSE=1

ERROR

AT+CIPCLOSE=1

ERROR

[25777] Connecting to blynk-cloud.com:80

AT+CIPCLOSE=1

ERROR

AT+CIPSTART=1,"TCP","blynk-cloud.com",80

OK

1, CONNECT OK

AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK

[26499] ### Available: 0 on 1

AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","139.59.206.133","80","CONNECTED"

OK

AT+CIPSEND=1,37

>

SRGDamia1, 2020-04-16:
Um.  Exactly what code are you using?  Something is either strange in your code or in your paste job or both.

Are you powering up the modem and then immediately sending it the restart command?  You should give it at least a few seconds (5-6) between waking it up and asking it to do anything.  Spec sheet for the SIM7000 says ~4.5s minimum.  The examples all give at least 3, which is optimistic.  The failure you saw happened because the restart asked it to go to minimum functionality before the board had finished waking up and that command appeared to fail.  But the board actually did go to low functionality a few seconds later which is why the sim card failure messages starting popping up.  At minimum functionality (CFUN=0) the SIM card is off.

What did you do to "update" your library to today's version?  If you have the most recent (ie, today), it should be printing out at least ``` ### TinyGSM Version: 0.10.3```  If you're using PlatformIO, you'll have to pull directly from Github or wait a day or two before the library crawler catches the updates from today.  If you're using the Arduino IDE's library manager, I think it goes by release numbers but I don't know how long between when a release is made and when the library manager can detect it.

Gigabyte1973, 2020-04-16:
Hi Again Sara Damiano.
I have BIG issues copy/pasting from the Serial Terminal, has to cut
multiple times to be able to get the complete log.
i'm not a great programmer at all, i'm Hardware designer but the code has
worked well the last 8 months or so with the 0.79.
Because of the Corona situation i cant get help from my good frend who is
educated SW programmer.

I know the SIM i use at the moment preffer gprs over LTE, but i have
ordered IOT specific sim cards, they will arrive next Tuesday.
The only reason i was trying the new TinyGSM lib, was to be able to use the
new  "BLYNK_GSM manager# Lib but i discovered it's only for ESP32 platform,
and i use Teensy 3.6.
So for the time beeing, i can just use the old 0.79 lib.

HW and SW used.
*Teensy 3.6 MCU* Arduino Version 1.8.12, TeensyDuino Version 1.5.1
*Modem: AND-GLOBAL SIM7000E* Link:
https://www.aliexpress.com/item/32840209812.html?spm=a2g0s.9042311.0.0.27424c4d0NNnHA
Modem Firmware R1351. Connected to HW UART4 @1152000 Baud. No jumper wires,
proper connected on a real PCB.

Br Rasmus

SRGDamia1, 2020-04-16:
Ok.  Well if you want to try an post chunks of the log where it's failing, I can look at them, but otherwise I'll leave this open until you're ready to get it going.

Mr-HaleYa, 2020-05-06:
Hi there,
 I am looking to add a Sim7000a to my line of device I am developing so we can use the Cat-M1 protocol since 2G has (mostly) died in our area. 😞 I can't seem to find *any* documentation on them or how to wire them to an Arduino. 

We were previously using a TTGO esp32 Sim800L 
[https://www.amazon.com/LILYGO-T-Call-Wireless-Antenna-SIM800L/dp/B07VD41JLN](url)

we are now hoping to use the same board as you posted above, except its the 7000a (America)
[https://it.aliexpress.com/item/32885205778.html?spm=a2g0o.detail.1000023.20.660258ddSIzAci](url)

How did you wire it to the esp8266 (we are using ESP32)? what pins when where? with the TTGO it had a doc explaining but like I said we can't find any for this. also is there anything special you had to do to get it to work? Since you said you had changed from a sim800 to the 7000 I thought you were the perfect person to ask. I also thought it was better to ask now than to not be able to get it to work and think its a bug and open a report. I heard someone say it may need a 3.7v logic shifter in between the two boards (whatever that is...)  Can you share an example of some sort or some images?

I know this is a bug report section and not really for questions but it look like you have done exactly what we are trying and I hope your knowledge saves me from opening bug reports in the future.

SRGDamia1, 2020-05-08:
I have that exact board.  I haven't actually wired it to an ESP, but I've successfully worked it with a 3.3V AVR board.  I'm looking at my notes trying to figure out what I had hooked up.  My pre-school daughter spilled a cup of Kool-Aid on my folder with my scribbled on schematic print-outs.

[image]

SRGDamia1, 2020-05-09:
Ok, hooked it up to my Feather32 (only ESP32 I own) and ran "AllFunctions" successfully.

[image]

Notes:
I hooked Rx/Tx to the ESP32's "Serial1"
I connected a LiPo (>3.7V directly to the PWRIN and one of the ground pins - other ground to the ESP32's ground.  Documentation suggests using 5V.  
Powering the module from the micro-USB jack on a "standard" 500mA port from the computer is NOT enough power.  It will talk to your board and give the impression that all is fine, but it probably won't be able to connect to the network without more power.
The PWRKEY is tied to ground, which keeps the board awake all the time.  The digital pin on the ESP32 I tried connecting to it didn't have enough umph to overcome it.  If you want to use power-down mode, you might have to be a little more creative.
I didn't test using the DTR pin for sleep.  If you want to use it, remember it's inverted via Q102

Mr-HaleYa, 2020-05-09:
Thank you for your reply! After some testing, I can confirm that the power supply needs to be much higher than any board can offer (we resorted to using a 2A wall adapter). Due to this,  we decided this would not be reliable for our project and made a fantastic discovery of the TTGO SIM7000G module. 
https://www.tindie.com/products/ttgo/lilygor-ttgo-t-sim7000g/

They have a GitHub at https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G

They also have examples using it with the TinyGSM library but it looks like they are using narrowband? 
 Our state is completely shutting down 2G very soon and we are panicking and scrambling to find an alternative. Our hope was to use Cat-M1 our project, Is that even supported with tinyGSM and how do we use it with such a board? 
Even in their example they define their GPRS. Isn't GPSR 2G? (or 2.5 as I've heard it called)

I guess my main points of question are
 1. does TinyGSM support Cat-M1
 2. is so How???

I figure it's better to ask the code developer directly that know exactly what works and why 😄

SRGDamia1, 2020-05-10:
I wouldn't get to hung up on which abbreviations someone is using. 2G, EDGE, GPRS, EPS, LTE, 4G, etc. They all have very specific technical meanings but most people mix them up so unless you're reading something in a very careful technical manual chances are they're not being used correctly. Very roughly, 2G, EDGE, and GPRS are in the same category and EPS, LTE/LTE CAT_, and 4G go together. But even those rough groupings are often used incorrectly.

TinyGSM uses "AT commands" (see https://en.m.wikipedia.org/wiki/Hayes_command_set) and implements an Arduino "Client" object - itself a subclass of a "Stream" - making it very easy to use any of the supported modules in conjunction with other Arduino libraries using that style coding. A huge variety of modules use some flavor of AT commands, which is why TinyGSM has such a variety of supported modules. In fact, Espressif supplies a firmware for using an ESP32 or ESP8266 only as a modem via AT commands - and TinyGSM supports it.  There are some standards for AT commands, but every manufacturer and module implements unique commands as well, which is why TinyGSM has so many different implementation files.

With all that background, yes, TinyGSM supports LTE-M for serveral modules where someone has coded up the AT-to-Client pathway - the BG96, and u-blox SARA R4 are two that join the SIM7000. Making any other LTE M or NB IoT work is just coding that path.