vshymanskyy/TinyGSM #382: Any future plans for China Mobile M8321 Support?

yakeer, 2020-04-19:
Wondering if I can get the M5Stack LTE Module (M8321) working with your library...

SRGDamia1, 2020-04-19:
If it works with something like an AT command syntax, it should be doable.  Submit a PR if you get it working.

captFuture, 2020-07-14:
I am currently working on the NB-IoT version of M5Stack (M5311) - probably they are pretty similar.
As the structure of TinyGSM has changed I currently need to find out how functions apart from the standards in /GPRS/TIME/SMS ... are defined.

(especially because the Modem is somehow reading GMT offset strangely [i need to divide it by four] )

on, off, restart, pin, gprs, sms... already work

SRGDamia1, 2020-07-20:
The divide by 4 is expected.  It's part of the specification for the CCLK command - see [3GPP 27.007 Ch. 8.1](https://www.etsi.org/deliver/etsi_ts/127000_127099/127007/08.15.00_60/ts_127007v081500p.pdf).  That command is implemented in the [Time ](https://github.com/vshymanskyy/TinyGSM/blob/master/src/TinyGsmTime.tpp) file.

To create a unique implementation for an inherited function, you need to make sure your module inherits from the template for the functionality you need (ie, ```class TinyGsmSim7000 : public TinyGsmModem<TinyGsmSim7000>```) and make the template class a friend class (ie, ```friend class TinyGsmModem<TinyGsmSim7000>;```).  Then within your class you can overwrite the implementation of the function to be what you need (ie, ```bool initImpl(const char* pin = NULL) { stuff here }```).  The "impl" functions should be protected.  

If the you want some unique new function, implement it as a public function right in the header for your module (as ```String setNetworkMode(uint8_t mode)``` is done in TinyGsmSim7000.h).  I recommend you check quickly through all the modules to see if one of the others implements something similar (or identical) so they can share the same name.  If you find several modules with the same implementation, we can move it up into the relevant template or create a new one if needed.

capedra, 2021-08-04:
> I am currently working on the NB-IoT version of M5Stack (M5311) - probably they are pretty similar.
> As the structure of TinyGSM has changed I currently need to find out how functions apart from the [...]

Hello @captFuture :
How is it possible to use the M5Stack M5311 NB-IOT module with TinyGSM library?
Thanks!