vshymanskyy/TinyGSM #383: MQTT TCP connection

elgarbe, 2020-04-19:
Hi, I'm testing MQTT client sketch in arduino uno. I've enabled at command dump and can see what happend with AT command.
I understand the GPRS configuration, I found it [here](https://github.com/vshymanskyy/TinyGSM/blob/8c5850ef7bdbb03e9293f39819ea87e1333704cf/src/TinyGsmClientSIM800.h#L306)
After configure the GPRS I can see the connection to the broker. I don't understand the proces used to read  brocker CONN ACK. I can see a lot of CIPSTATUS and CIPRXGET=4,0. 
Where is the code for that part of the process?

Thank!

SRGDamia1, 2020-04-19:
The MQTT example given in this library uses the PubSubClient library for the MQTT implementation.  Look in that library for the details on the broker acknowledgements and keep alive.

The CIPSTATUS and CIPRXGET you're seeing are in the "modemGetConnected" and "modemGetAvailable" functions from this library.  That is the library checking if the TCP connection is still valid and if there has been any data received, respectively.  There is nothing specific to MQTT about those commands - they are generic for any TCP connection.  You'll see them appear any time you or another library (like PubSubClient) asks about the client status (ie, via ```connected()``` or ```available()```)  When data actually comes in over the connection the modem *should* spit out a ```+CIPRXGET:1,1``` saying data is available.  If you ask for the available data, this library fetches the data with a ```+CIPRXGET=2,1,# chars available``` - that's in the "modemRead" function.

closed 2020-04-19 by SRGDamia1

elgarbe, 2020-04-19:
Thank you very much for your explanation it is very useful for me.
Could I ask one more question? I don't know much about TCP. why CIPMUX =1 is used? It's for multiple ip, but why it is needed in this context?

Again, thanks!

SRGDamia1, 2020-04-19:
This library always uses commands for multiple connections if possible.  I'm not sure what you mean by "in this context" but if you mean "within just this single example" then, yes, multiple connections aren't needed.  But within the scope of all possible uses of this library, there are plenty of places multiple connections are useful.  There is no disadvantage that I know of to using multiple connections if the module supports them.

elgarbe, 2020-04-19:
Yes I mean "within just this single example".
Is this library Arduino dependent? Or may I try to port to other architectures/toolchain with litle work?
thank!

SRGDamia1, 2020-04-19:
I haven't done much testing on non-Arduino architecture.  You're welcome to try.