vshymanskyy/TinyGSM #384: Sim7600CE not working with Tinygsm 0.10.5

tvbtrum-phuong, 2020-04-22:
Hi there,

I am working on ESP32 with SIM7600

Iam able to get an GPRS connection, after that the sketch is trying to connect to Blynk. Thats failing -> Reset ESP

Is there someone who has this setup already working?? I need some help with this…

SRGDamia1, 2020-04-22:
Please post your AT log and a better description of what the problem is.

tvbtrum-phuong, 2020-04-23:
> 
> 
> Please post your AT log and a better description of what the problem is.

I'm working with Blynk 0.6.1, not using AT command
[image]

4093] Modem init...
[4121] Connecting to network...
[4135] Network: Viettel #Hay o nha! Viettel
[4135] Connecting to v-internet ...
[4275] Connected to GPRS
[4338] Connecting to blynk-cloud.com:80
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1666  PS      : 0x00060f30  A0      : 0x800d2f0c  A1      : 0x3ffb1e50  
A2      : 0x3ffc0300  A3      : 0x3ffc0300  A4      : 0x3f40143a  A5      : 0x3ffc0308  
A6      : 0x3ffc035c  A7      : 0x00000001  A8      : 0x3ffc032c  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3f40143a  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000001c  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x400d1666:0x3ffb1e50 0x400d2f09:0x3ffb1e80 0x400d2f68:0x3ffb1eb0 0x400ebf12:0x3ffb1ed0 0x400ebf5b:0x3ffb1ef0 0x400d25f1:0x3ffb1f10 0x400d25b1:0x3ffb1f40 0x400d2ed5:0x3ffb1f70 0x400d42fb:0x3ffb1fb0 0x400885bd:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

___________________________________________________________________________________

Modem Info: Manufacturer: SIMCOM INCORPORATED Model: SIMCOM_SIM7600CE-T Revision: SIM7600M22_V1.1 IMEI: 864999042097632 +GCAP: +CGSM
[4206] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.6.1 on ESP32

[4218] Modem init...
[4255] Connecting to network...
[4274] Network: Viettel #Hay o nha! Viettel
[4274] Connecting to v-internet ...
[4439] Connected to GPRS
[4510] Connecting to blynk-cloud.com:80

It's connected toGPRS but not connect with blynk server.

I think tinygsm 0.10.5 not support for SIM7600.