vshymanskyy/TinyGSM #388: Guidance for Asynchronous https requests

ajinkyaw, 2020-04-27:
## Web Apis requests

[ x] Bug or problem compiling the library
[ x] Bug or issue with library functionality (ie, sending data over TCP/IP)

### TinyGSM Library with Sim800c and ESP32?

Main processor board: ESP32
Modem: - Sim800c
TinyGSM version: 0.10.5
Code: NA

### Scenario, steps to reproduce
My primary goal is to execute the apis ( Https Requests ) asynchronously, that means multiple requests must execute in parallel at a time.
I tried to create and run different object of HttpClient into threads but it does not work.

### Expected result
I will be able to execute multiple requests at a time.

### Actual result
Current only one request is being processed at a time.

### AT command log
NA

xvinny, 2020-04-27:
I'm currently getting the same problem. You have to follow these steps to solve it:

1 - Task synchronization: You must synchronize all the functions called through a Semaphore (this work should be done by TinyGsm developers, but....). For example:

```
class MyGsmSim800 :  public TinyGsmSim800
{
public:
   void begin()
   {
      _semaphore = xSemaphoreCreateMutex();            
   }
   bool isNetworkConnected()
   {
       xSemaphoreTake(_semaphore, timeout);
       bool result = TinyGsmSim800::isNetworkConnected();
       xSemaphoreGive(_semaphore);

       return result;
   }

private:
   SemaphoreHandle_t _semaphore;
};
```
Don't forget that you must use the same instance of MyGsmSim800 for all tasks which use GSM in order to take the advantage of the semaphore.
This feature (semaphore locking) should be developed in TinyGsmModem, but I don't know why they didn't do it yet.

2 - Mux: You have to use the Mux of your TinyGsmClient in order to separate the connections: TinyGsm doesn't provide an example or instruction on how to use it, and I'm discovering and using it by myself. Don't forget that the client calls to TinyGsm also must be synchronized.

@SRGDamia1 what do you think about putting this feature into TinyGsm backlog?

ajinkyaw, 2020-04-27:
> I'm currently getting the same problem. You have to follow these steps to solve it:
>
> 1 - Task synchronization: You must synchronize all the functions called through a Semaphore (this work should be [...]

Very first i appreciate your quick reply.
I have no idea about working on SemaPhore but i will try to learn about it and give a go ahead.
I have been studying the asynctcp lib code for reference. 
If u have more guidance please do post.
Thanks a lot.

SRGDamia1, 2020-05-05:
> 1 - Task synchronization: You must synchronize all the functions called through a Semaphore (this work should be done by TinyGsm developers, but....). 

So, giant flags like lifeguards use on the beach?

> This feature (semaphore locking) should be developed in TinyGsmModem, but I don't know why they didn't do it yet.

Because I had to ask Google what flag signals could possibly have to do with TinyGSM.   I have zero familiarity with FreeRTOS.  Sorry.

> what do you think about putting this feature into TinyGsm backlog?

I could possibly go over a pull request, but that's about it.  

> 2 - Mux: You have to use the Mux of your TinyGsmClient in order to separate the connections: TinyGsm doesn't provide an example or instruction on how to use it, and I'm discovering and using it by myself.

This, I can handle and will make a separate issue to write an example setting the mux value for multiple connections.  It's very simple, when creating the client use ```TinyGsmClient client(modem, muxNo);``` instead of  ```TinyGsmClient client(modem);``` The number of connections supported by one modem varies - XBee's only support 1 while the BG96 can handle 12.

Keep in mind that some functionality is tied to the modem itself and other to the individual TCP/SSL connections.  That is, whether there is internet access at all (isNetworkConnected(), gprsConnect(...), etc) is dependent on the modem and applies to all possible TCP connections while other functions (available(), etc) are dependent on a specific connection.  I have no idea what impact that would make on the semaphores.

xvinny, 2020-05-05:
Thanks for the answer @SRGDamia1 !!
Regarding the task synchronization, I did the following:

```
class MyGsmSim800 : public TinyGsmSim800
{
public:
    explicit VidoGsmSim800(Stream &stream) : TinyGsmSim800(stream)
    {
    }

    bool VidoGsmSim800::restart()
    {
        xSemaphoreTake(m_semaphore, portMAX_DELAY);
        bool result = TinyGsmSim800::restart();
        xSemaphoreGive(m_semaphore);

        return result;
    }
...
// the same for the other functions
```

It is working well.. now I can have two threads calling a gsm modem. Before this, if the threads executed calls to the modem at the same time, there was a concurrency problem. Imagine two threads using the same serial at the same time, one reading and one writing, it became a mess.

SRGDamia1, 2020-05-05:
I'm really glad it's working for you and someone else was able to help you through it because this was all I could think of when semaphore came up:  https://montypython.fandom.com/wiki/The_Semaphore_Version_of_Wuthering_Heights