vshymanskyy/TinyGSM #391: Any working example code for socket.io client

ilomon10, 2020-05-08:
Hi, I am new to this library, I plan to use this library for projects but I am still confused about implementing this library to communicate with the socket.io server.

Maybe you can add some sample code.

## What type of issues is this?
- [ ] Request to support a new module
    
- [ ] Bug or problem compiling the library
- [x] Bug or issue with library functionality (ie, sending data over TCP/IP)

### What are you working with?

Main processor board:  `Arduino Nano`
Modem: `SIM800l`
TinyGSM version: `0.10.6`
Code: `sorry not tried yet`

### Scenario, steps to reproduce

Trying with socket.io client library for arduino.

### Expected result

Establish a fully functional connection.

### Actual result

Not yet known, because it has never been tried, but will soon be tried.

Maybe there are any libraries provided that can run on top of this library.

Thanks

SRGDamia1, 2020-05-09:
This library just translates the AT commands used by many cellular modules into an Arduino "Client".  If you have some other library that can connect to socket.io using a "Client" implementation, then this should work to supply that.  This library does NOT use or create web-sockets or do anything else unique to any particular server type.  It also only works for out-going connections; it does not support incoming TCP connections in any way.

If you actually have trouble with the AT commands or the TCP connection, then please open an issue specific to those.  If you do get something working and you think your example would be useful to others, you can create a PR adding the example to the examples/more folder.

closed 2020-05-09 by SRGDamia1

SRGDamia1, 2020-05-09:
I'm sorry; I didn't mean to just brush you off.  If there is some problem with the way this library is working that is revealed by socket.io, please do let me know.  But you will probably need to find or create some other library for the socket.io parts of the connection - this only does raw TCP.