vshymanskyy/TinyGSM #392: client.available() not working

arangu97, 2020-05-11:
Hello,
I have a problem because the client.connect() is working (it returns true) and also the client.connected().
However client.available() is never true and then I cant read the server response. 
After 4 seconds(aprox) client.connected() changes to false so I dont know if maybe there is no enough time to get the server response.
Any idea about what is happening?
I attach part of my code: 

`if` (client.connect(server, port)) {
    // Make a HTTP request:
      SerialMon.println("Making POST request");
      SerialMon.println(jsonString);
      client.print("POST /auth HTTP/1.1");
      client.print(String("Host: ") + server + "\r\n");
      client.println("Connection: close");
      client.println("Content-Type: application/json");
      client.print("Content-Length: ");
      client.println(jsonString.length());
      client.println(jsonString);
      
      unsigned long timeout = millis();
      while (client.connected()) {
        // Print available data (HTTP response from server)
        while (client.available()) {
          char c = client.read();
          SerialMon.print(c);
        }
      }
`
Thank you

SRGDamia1, 2020-05-11:
Please post the "AT" log.  And you didn't even say which module you are using.

arangu97, 2020-05-13:
I am not using AT commands. How can I see the log?
I have this code using the TinyGSM library to make the proper connections.

The module I am using is the SIM800L integrated in the ESP32.

// SIM Module Pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22

#define SerialMon Serial
#define SerialAT Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb

#include <Wire.h>
#include <TinyGsmClient.h>
// SIM Module Pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22

#define SerialMon Serial
#define SerialAT Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb

// GPRS-GSM credentials
const char apn[]      = "TM"; // APN (example: internet.vodafone.pt) use https://wiki.apnchanger.org
const char gprsUser[] = ""; // GPRS User
const char gprsPass[] = ""; // GPRS Password

// SIM card PIN (leave empty, if not defined)
const char simPIN[] = "***"; // GPRS Password

// Server details
const char server[] ="****";
const char path[] = "*****";   // resource path, for example: /post-data.php
const int  port = 443;       // server port number

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else// Define the serial console for debug prints, if needed
  TinyGsm modem(SerialAT);
#endif

// I2C for SIM800 (to keep it running when powered from battery)
TwoWire I2CPower = TwoWire(0);

// TinyGSM Client for Internet connection
TinyGsmClient client(modem);

#define uS_TO_S_FACTOR 1000000     /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  600        /* Time ESP32 will go to sleep (in seconds) 3600 seconds = 1 hour */

#define IP5306_ADDR          0x75
#define IP5306_REG_SYS_CTL0  0x00

bool setPowerBoostKeepOn(int en){
  I2CPower.beginTransmission(IP5306_ADDR);
  I2CPower.write(IP5306_REG_SYS_CTL0);
  if (en) {
    I2CPower.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
  } else {
    I2CPower.write(0x35); // 0x37 is default reg value
  }
  return I2CPower.endTransmission() == 0;
}

void setup() {
  // Set serial monitor debugging window baud rate to 115200
  SerialMon.begin(115200);
  delay(10);

  // Start I2C communication
  I2CPower.begin(I2C_SDA, I2C_SCL, 400000);

  // Keep power when running from battery
  bool isOk = setPowerBoostKeepOn(1);
  SerialMon.println(String("IP5306 KeepOn ") + (isOk ? "OK" : "FAIL"));

  // Set modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);
  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);
  SerialMon.println("Wait...");

  // Set GSM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
  delay(3000);

  SerialMon.println("Initializing modem...");
  modem.restart();

  if (strlen(simPIN) && modem.getSimStatus() != 3 ) {
    modem.simUnlock(simPIN);
  } 

  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
}

void loop() {
  SerialMon.print("Connecting to APN: ");
  SerialMon.print(apn);
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    SerialMon.println("Failed connecting to GSM Network");
  }
  else {
    SerialMon.println("Connected to GSM Network");
    if (modem.isGprsConnected()) {
      SerialMon.println("GPRS connected");
    }
    delay(3000);
    
    SerialMon.print("Connecting to ");
    SerialMon.println(server);
    String jsonString = "{ \"email\": \"" + email + "\", \"password\": \"" + password + "\" }";
    if (client.connect(server, port)) {
    // Make a HTTP request:
      SerialMon.println("Making POST request");
      SerialMon.println(jsonString);
      client.print("POST" + resource + " HTTP/1.1");
      client.print(String("Host: ") + server + "\r\n");
      client.println("Connection: close");
      client.println("Content-Type: application/json");
      client.print("Content-Length: ");
      client.println(jsonString.length());
      client.println(jsonString);
      
      unsigned long timeout = millis();
      while (client.connected()) {
        // Print available data (HTTP response from server)
        while (client.available()) {
          char c = client.read();
          SerialMon.print(c);
        }
      }
      SerialMon.println();
    } else {
      SerialMon.println("Connection failed");
    }
    
    client.stop();
    SerialMon.println(F("Server disconnected"));
    modem.gprsDisconnect();
    SerialMon.println(F("GPRS disconnected"));
  }
 
  // Put ESP32 into deep sleep mode (with timer wake up)
  esp_deep_sleep_start();
}

SRGDamia1, 2020-05-13:
TinyGSM uses AT commands.  To enable the AT output, you add the line ```#define DUMP_AT_COMMANDS``` somewhere above the chunk of code that begins ```#ifdef DUMP_AT_COMMANDS``` and make sure that you have the [StreamDebugger](https://github.com/vshymanskyy/StreamDebugger) library installed.

You also need to add a ```modem.waitForNetwork()``` before your ```modem.gprsConnect(apn, gprsUser, gprsPass)``` in the loop.  You cannot establish a GPRS connection without first successfully registering on the network.  The registration should happen automatically once the module is at full functionality with the SIM unlocked, but it takes time and you have to wait for it.

SRGDamia1, 2020-05-13:
You need another blank line between you HTTP headers and your content: https://developer.mozilla.org/en-US/docs/Web/HTTP/Messages.

```cpp
      client.print("Content-Length: ");
      client.println(jsonString.length());
      client.println();  // <----------- ADD THIS
      client.println(jsonString);
```

arangu97, 2020-05-14:
Ok, I see... Thank you so much
Yes I connect to the GPRS network properly and also to the server(the only problem that I had is that I had to specify the IP address because with the URL is not working, maybe I should add some DHCP module or server?).

arangu97, 2020-05-14:
I get this response for the server:

```
HTTP/1.1 400 Bad Request
Server: nginx/1.16.1
Date: Thu, 14 May 2020 07:59:15 GMT
Content-Type: text/html
Content-Length: 157
Connection: close

<html>
<head><title>400 Bad Request</title></head>
<body>
<center><h1>400 Bad Request</h1></center>
<hr><center>nginx/1.16.1</center>
</body>
</html>
```

However when I do the same request with Postman I get the response properly.

I attach the AT log.

[at-log.txt](https://github.com/vshymanskyy/TinyGSM/files/4626829/at-log.txt)

SRGDamia1, 2020-05-14:
The library is working properly.  It's the format of your request that is the problem.

Postman is a great tool, but when you use Postman it creates the headers and does most of the work for you.  When you use this library, you need to do it all yourself.  Some servers are very touchy about the format of the headers and whether they come in as a single string or not.  Make sure you look at the "raw" request Postman is sending and you are using matching headers.  I also recommend using only one print statement for each header; that is, changing
```cpp
client.print("Content-Length: ");
client.println(jsonString.length());
```
to
```cpp
client.print("Content-Length: " + String(jsonString.length()));
```

closed 2020-05-14 by SRGDamia1

SRGDamia1, 2020-05-14:
I wrote up some more about in the ReadMe because it's come up a few times.