vshymanskyy/TinyGSM #393: modem.getBattStats -> chargeState -> not working

MiCentinela, 2020-05-12:
Hello,

When I'm using a module TTGO 800L, when I call this function: modem.getBattStats -> chargeState ,
it always returns to 0, independently if the origin of the power source is the battery or an USB port.

Thanks for your support.

SRGDamia1, 2020-05-13:
I know that board is popular, but I'm not familiar with it.  Whether or not that function works depends on the specific firmware (not all support it) and the wiring (many break-outs have separate voltage regulators upstream of the "battery" voltage that the SIM800 sees).  Can you post a clip of your AT log for that function?

w3st0, 2021-02-14:
Same issue here, just gets random numbers on batttery Voltage, Percent, ChargeState TTGGO T-Call SIM800L_IP5306_VERSION_20190610 

AT+CBC

+CBC: 0,76,4016

Percent it goes from 70 to 80. (Randomly  between those values)
ChargeState is always 0 

Voltage it ranges from 3998 to 4025.

SRGDamia1, 2021-02-15:
Whether or not the battery parameters are meaningful depends on the wiring of your board.  You'll need to look it up and figure out the connections yourself.  Almost all breakouts I've seen have power regulators upstream of the Vbat for the SIM800.  If your board does, then the battery values/states read by the SIM800 only show the condition of your power regulators, not the battery on  the other side of them.  A constant voltage/percent/state reading probably means your breakout has a separate regulator.

closed 2021-02-15 by SRGDamia1