vshymanskyy/TinyGSM #394: Communication problem between Sara-R410M and ESP32/Arduino

wavewolf, 2020-05-14:
Hi, everybody.
I'd like to solve a specific problem.
I have a "LTE IoT click" module from Mikroelectronika [https://www.mikroe.com/lte-iot-click](url) with the "SARA-R410M" chip. The SIM card I have installed is with the PIN deactivated.

I am connecting this module to both my ESP32 and Arduino modules and in both situations I find the same error.

The only thing I get from the serial monitor is that it returns an "OK" when I send it the "AT" command. For everything else, it returns "ERROR".

Because of this, I have downloaded a specific program from the SARA-R410 chip for Windows [https://www.u-blox.com/en/product/m-center](url) and by connecting it directly to the USB of my computer, I have been able to send an SMS without any problem.

I have several reports of what the serial monitor shows me when I install the "AT_Debug" test program:

`[6034] Trying baud rate 115200 ...
[7049] Modem responded at rate 115200
***********************************************************
 You can now send AT commands
 Enter "AT" (without quotes), and you should see "OK"
 If it doesn't work, select "Both NL & CR" in Serial Monitor
***********************************************************
AT

OK
`

If I try the "Diagnostics" code, I get the following from the monitor:

`Initializing modem... [fail]
************************
 Is your modem connected properly?
 Is your serial speed (baud rate) correct?
 Is your modem powered on?
 Do you use a good, stable power source?
 Try useing File -> Examples -> TinyGSM -> tools -> AT_Debug to find correct configuration
************************`

And if in the same code I change "modem.restart()" for "modem.init()", I get the following:

`Initializing modem...[6044] ### TinyGSM Version: 0.10.6
[6044] ### TinyGSM Compiled Module:  TinyGsmClientSaraR4
 [fail]
************************
 Is your modem connected properly?
 Is your serial speed (baud rate) correct?
 Is your modem powered on?
 Do you use a good, stable power source?
 Try useing File -> Examples -> TinyGSM -> tools -> AT_Debug to find correct configuration
************************
`

What could be happening?

Thanks.

SRGDamia1, 2020-05-14:
Oh, the *SARA* R410M, how unworthy you are of your name!  Or is it that you are worthy of it for being so very temperamental?

Anyway, what baud rate are you using in the diagnostics?  Are you attempting to auto-baud?  Actually, are you attempting to use any baud rate other than 115200?  The R410M baud rate settings are NOT persistent between resets - it _always_ reverts back to 115200.  I think it does support auto-bauding, but if TinyGSM attempts to match it with its own auto-baud routine and settles on anything other 115200 the two won't match up.  Try making sure everyone is settled on that baud rate and hopefully they'll all chat nicely.

Using an 8MHz board with the R410M is a huge pain because 115200 is too fast for it.  After every single reset you have to start the baud at 115200, throw out a command to the SARA to tell it to drop its baud, hope there weren't errors in the outgoing message because chances are there will be in the OK back (Tx is ususally better than Rx), and then restart your serial at a lower baud:
```cpp
// after ever power on or reset:
delay(5250);  // Must wait for UART port to become active. ~5s
 _modemSerial->begin(115200);
 gsmModem.setBaud(9600);
_modemSerial->end();
_modemSerial->begin(9600);
```

wavewolf, 2020-05-14:
> Oh, the _SARA_ R410M, how unworthy you are of your name! Or is it that you are worthy of it for being so very temperamental?
> 
> Anyway, what baud rate are you using in the diagnostics? Are you attempting to auto-baud? Actually, are you attempting to use any baud rate other than 115200? The R410M baud rate settings are NOT persistent between resets - it _always_ reverts back to 115200. I think it does support auto-bauding, but if TinyGSM attempts to match it with its own auto-baud routine and settles on anything other 115200 the two won't match up. Try making sure everyone is settled on that baud rate and hopefully they'll all chat nicely.
> 
> Using an 8MHz board with the R410M is a huge pain because 115200 is too fast for it. After every single reset you have to start the baud at 115200, throw out a command to the SARA to tell it to drop its baud, hope there weren't errors in the outgoing message because chances are there will be in the OK back (Tx is ususally better than Rx), and then restart your serial at a lower baud:
> 
> ```c++
> // after every power on or reset:
> delay(5250);  // Must wait for UART port to become active. ~5s
>  _modemSerial->begin(115200);
>  gsmModem.setBaud(9600);
> _modemSerial->end();
> _modemSerial->begin(9600);
> ```

I'm sorry, but it's not very clear what you're asking me to do.
I am currently using ESP32 with a speed of 80MHz and in any case I use a communication speed of 115200 bauds. Code I use is the one provided as an example in this library, I don't touch anything.

Thanks @SRGDamia1

SRGDamia1, 2020-05-14:
In tools/diagnostics make sure you change ```SerialAT.begin(9600);``` to ```SerialAT.begin(115200);``` in line 132.

SRGDamia1, 2020-05-14:
The ESP32 should have no problem with 115200,  You mentioned another Arduino so I thought I'd warn you in case it was a slow board.

wavewolf, 2020-05-14:
I always use 115200 bauds. 
I think problem must be something else. 
My suspicions are more focused on design of Mikroelectronika's LTE IoT click module. 
Even if it's using SARA R410M chip, there must be something causing this miscommunication.
I'll keep investigating in the meantime.

SRGDamia1, 2020-05-14:
You checked your code?  Because the program in the tools directory uses 9600 and you said you didn't change it.

Are you correctly switching your module on with a 0.15-3.2 second LOW pulse on the PWR_ON pin? You have to supply power, wait about half a second, and then pulse the PWR_ON pin for the module to begin responding.  That's tied to pin 1 according to the click schematic.  The same pulse could also turn it off, so watch LED2 for the current state.  You have to add all that code for turning the board on yourself - none of the examples have anything of the sort.

You have to be able to consistently get responses to AT commands before TinyGSM will ever work for you.  It depends on having a functional serial connection.

wavewolf, 2020-05-15:
I was able to figure out the mistake. But I'm still trying all examples in library. 
Now I can communicate without a problem.

Thinking about your comment and focusing on the click module, I realized that module was at a logical voltage level of 5V and the microcontroller is at 3V3. I have simply modified configuration of module to 3V3, and it already works correctly.

Now as you say, I have to program module to turn on correctly when it should.
Thanks again.

wavewolf, 2020-08-09:
Hi @SRGDamia1 , again. 
I just have a question from my ignorance. 
Sara-R410M supports "LTE Cat M1" and "Narrowband 1 (NB1)" technology. 
When we use this library, does the library use these module technologies by default? 
I mean, if I wanted to use 4G technology with any module compatible with this library and TCP or SSL/TLS, does it really work at 4G or just GSM?

Thank you.

SRGDamia1, 2020-08-09:
TinyGSM doesn't tell any of the modules which technology to use, so each would use whatever its own "factory default" is our whatever it was last programmed for. Unless you've ever changed it, the SARA R410M default is to first choose LTE-M and "fallback" to NB-IoT.