vshymanskyy/TinyGSM #395: can't connect to server

maromme, 2020-05-17:
Hi,
I'm stuck with connecting to server, following that example:
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800L/tree/master/examples/Arduino_TinyGSM

modem.isGprsConnected() is positive -> GPRS connection is OK

my defines
```
// ------ TinyGSM ------
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb
```

That's how I try to connect:
```
// Server details
const char server[] = "vsh.pp.ua";
const char resource[] = "/TinyGSM/togo.txt";
const int  port = 80;

 Serial.print("Connecting to ");
  Serial.print(server);
  if (!client.connect(server, port)) {
    Serial.println(" fail");
    delay(10000);
    return (false);
  }
  Serial.println(" success");
```
And that's the "result".
```
21:27:00.469 -> Connecting to vsh.pp.uaAT+CIPCLOSE=0,1
21:27:00.516 -> 
21:27:00.516 -> ERROR
21:27:00.516 -> AT+CIPSSL=0
21:27:00.562 -> 
21:27:00.562 -> OK
21:27:00.562 -> AT+CIPSTART=0,"TCP","vsh.pp.ua",80
21:27:00.562 -> 
21:27:00.562 -> OK
21:28:15.549 ->  fail
```
Tried also my own server - without success. 

I Hope anyone got a hint for me to solve that issue.
Thanks a lot,
Markus

maromme, 2020-05-18:
That one was "easy". my cellphone contract wasn't  activated for data...

closed 2020-05-18 by maromme

SRGDamia1, 2020-05-18:
Doh.