vshymanskyy/TinyGSM #396: Implement peek()

WeightScale, 2020-05-18:
Implement peek() like this 
int peek() override {
	    return (uint8_t)rx.peek();//add to TinyGsmFifo.h
    }

TinyGsmFifo.h
uint8_t peek(){
		return _b[_r];	
	}

SRGDamia1, 2020-05-18:
Could you please form this to a pull request?

WeightScale, 2020-05-18:
// TODO(SRGDamia1): Implement peek
    int peek() override {

IoTThinks, 2021-06-14:
I intent to raise an issue, too.
https://github.com/vshymanskyy/TinyGSM/blob/master/src/TinyGsmTCP.tpp#L245

Then I see this issue.
When I use TinyGSM to download a bin to do OTA by esp32 Updater.
esp32 updater keeps complaining the first bit is not 0xe9.
When I or the Updater do the peek() to check the first bit, it is always FF.

Now, I know it may due to the -1 here. -1 mean 255 or FF.

IoTThinks, 2021-06-14:
> Implement peek() like this
> int peek() override {
> return (uint8_t)rx.peek();//add to TinyGsmFifo.h
> }
> 
> TinyGsmFifo.h
> uint8_t peek(){
> return _b[_r];
> }

Let me try if this works in my program.
If works, then I will create pull request.

IoTThinks, 2021-06-15:
I created a pull request based on the suggestion.
https://github.com/vshymanskyy/TinyGSM/pull/554

It works for my case. I can peek() correctly the first bit of the binary file when downloading via TinyGSM.