vshymanskyy/TinyGSM #397: Problem with client.connect

arangu97, 2020-05-19:
Hello, 
This is weird. When I use the client.connect(server,port) with a server URL it gets a connection error with the following AT log:

Connected to GSM Network
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

10.193.3.234

OK
GPRS connected
Connecting to https://api.thingspeak.com
AT+CIPCLOSE=0,1

ERROR
AT+CIPSSL=0

OK
AT+CIPSTART=0,"TCP","https://api.thingspeak.com",443

OK

0, CONNECT FAIL
Connection failed
AT+CIPCLOSE=0,1

ERROR
Server disconnected
AT+CIPSHUT

SHUT OK
AT+CGATT=0

However, if I put an IP address in the server field it connects it properly.

Any idea of what can be happening?

SRGDamia1, 2020-05-19:
You should set your server value to just 'api.thingspeak.com', without the https:// in front of it.

Also, shouldn't you be using port 80?

braysonjohn148, 2022-07-18:
@arangu97 were you able to solve this issue?

asadi2021, 2022-10-29:
Hi ,
I have the same problem when connecting to server teknifa.com (my website) with port80 via TinyGSM library  :

Initializing modem...
Modem: SIM800 R14.18
Waiting for network... OK
Connecting to myapn OK
Local IP:212.234.46.15
Signal quality:25
card initialized.
Connecting to teknifa.com fail