vshymanskyy/TinyGSM #398: isGprsConnected on esp32 not working for core 0

vossie123, 2020-05-24:
Hello
I am using an esp32 and using core 0 to run all the gsm modem lines.
Working great also connected to gprs but isGprsConnected is not working always giving 0

on core 0:
bool stateGPRS = modem.isGprsConnected();
Serial.print("Modem c0 status: ");
Serial.println(stateGPRS);
gives: Modem c0 status: 0

on core 1:
bool stateGPRS = modem.isGprsConnected();
Serial.print("Modem c1 status: ");
Serial.println(stateGPRS);
gives: Modem c1 status: 1

nickb512, 2020-07-07:
Also came here to report this. Same behavior over here but I found if you run the whole code including the setup of the modem on core 0 it works fine.

```
void networkTask(void * parameter) {
  modemSetup();
  for( ;; ) {
    return modem.isGprsConnected();
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}
```

This works fine using xTaskCreatePinnedToCore on either 0 or 1 (obvs modemSetup() contains all the code to start the modem, wait for network, connect to GPRS etc).

nickb512, 2020-07-07:
Also it took me longer to realise than I care to admit so worth noting for others who might be making similar mistakes, when multitasking with the ESP32's two cores it's important to remember that the underlying AT serial protocol is sending commands and awaiting replies from the modem, so if you start firing off commands from all sorts of tasks running at different times and/or on different cores then things are going to end up out of sync with multiple commands being sent at the same time.

Potentially worth documenting that by its nature, the library does not support multi-tasking and a single task should handle all communication with the modem AND any clients that use the connection (eg I had my mqtt.loop() call in a different task on a different core and it took me FOREVER to work out why the modem comms kept stalling).

SRGDamia1, 2020-08-07:
I'm glad you figured it out!

parthbhat13, 2020-10-03:
> Also it took me longer to realise than I care to admit so worth noting for others who might be making similar mistakes, when multitasking with the ESP32's two cores it's important to remember that [...]

Hey,
your comments helped me a lot as i have been struggling a lot to do the same. But, when you say that different task on differet core for mqtt, how about on the same core and different task of the mqtt once we are sure that the network task is done and stable with gprs connected ? will that work? do you mind to share a sample code snippet on how mqtt and gsm tasks could be run please? 

thanks in advance.

nickb512, 2020-10-03:
Hi @parthbhat13 

Indeed that did end up being an issue too. Our device is basically a GPS tracker (using SIM7000) with some advanced logic to switch a HVAC unit on/off and to different modes etc. There is also a local LCD display and menu for monitoring/setup of the unit. All running on an ESP32.

We ended up with 1 task running on core 1 which manages the LCD display and menu. Another task on core 1 which does all the logic/switching and a final task running on core 0 which does all of the network/location work. The single task then runs through in squence getting the GPS location, reading the sensor data, making/checking the network connection and then publishing the sensor data to MQTT etc.

Basically anything that could result in any serial command being sent to the SIM7000 happens in that one sequential (blocking) task so we never ask the SIM7000 to do two things at once, but it leaves the UI/Menu free to run on the other core so there are no delays to user input.

Our code is quite complex so its quite a faff to post an example, but hopefully that helps somewhat? If you have any more specific questions feel free to ask away.

Nick

parthbhat13, 2020-10-03:
hey @nickb512 ,
thanks for that quick reply. i have been spending all the time since i posted a reply here. what i came across so far is that , if i run any Serial based task on the core 0 it happens to stall or it will ask for the WDT rest thing. apart from that , i even noticed if i run a basic flasher on the core 0 still it will ask for the wdt_task reset. i am not so familiar with the rtos and not even that fluent all of my complex codes on arduino have always been on the Millis logic ... 
but everything works so far when i pin it to core 1 and run it. so i dont know where exactly is the problem.

so far i am able to run something more like this i will post a dropbox link. feel free to check, let me tell you the code is not at all formatted and i am just working on it. but this is just step one, i even have to add OTA update and then RS485 communication as well as mqtt, plus a wifi manager to handle everything else from the basic config.  super stressed out about this as i thought using rtos will make everything run smoother and better but i am facing tons of issues. probably because i am a beginner in the rtos family. but any guidance from you will be super helpful and appreciated. 

the link [https://www.dropbox.com/sh/hye2zqhddo0kreh/AADZg3xaq8-Rl7iH1VC_iuUKa?dl=0](url) 

thanks a lot for the quick reply. thank you very much. have ben struggling a lot on this whole day.

nickb512, 2020-10-03:
Haven't got time to look through the code today but will have a look in the week. Have you got a vTaskDelay call within your task loop?

`vTaskDelay(10000 / portTICK_PERIOD_MS);`

parthbhat13, 2020-10-03:
Nope, not in the loop, in the void loop, yes i have that. I usually avoid using delays in my code. And yeah thanks, please let me know. It's not at all a lengthy code just very basic for starters.. 

Thanks a lot.

nickb512, 2020-10-03:
You need that in the task look to yield the CPU to allow other system functions to take place too. One of those other functions is the watchdog reset so maybe that is causing your WDT issues.

parthbhat13, 2020-10-03:
Alright, i will try that and let you know. Because whenever i restart the gsm, or where it's waiting for the network. Thats where the problem starts is what i feel. I don't know how can i do that in the gsm waiting loop. That's where the problem is what i feel.