vshymanskyy/TinyGSM #401: Never leaving network state REG_SEARCHING

Elmi77, 2020-05-26:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[X] Question or request for help

### What are you working with?

Modem: SIM7000G 
Main processor board: LilyGo SIM7000G 
TinyGSM version: 0.10.5 (latest version provided via ArduinoIDE) 
Code: initialisation based on code as shown at https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino 

### Scenario, steps to reproduce

I want to register at the GSM network and send data via mobile internet in a second step.

### Expected result

### Actual result

modem.waitForNetwork() needs a very long time before it returns true. modem.isNetworkConnected() never returns true and modem.getRegistrationStatus() never leaves state 2 (SIM_SEARCHING).

So there is something obviously wrong either in my code or with the SIM-card or the hardware. My question is: how can I find out what is wrong? Any additional debug possibilites?

### Debug and AT command log

Initialise modem
[433] ### TinyGSM Version: 0.10.5
[433] ### TinyGSM Compiled Module:  TinyGsmClientSIM7000
[5450] ### Unhandled: f`f
[5874] ### Modem: SIMCOM SIM7000G
SIMCOM SIM7000G
SIM7000G R1529
Reg-Status: 2
SIM-status: 1
Reg-Status: 2
[11660] ### Unhandled: OK
GSM pref mode
Reg-Status: 2
Preferred network modes:  ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
[14203] ### Unhandled: OK
GSM wait for network
Reg-Status: 2

Elmi77, 2020-05-26:
Update: I found the AT debugging function. This is what it returns while waiting for the network:

```
AT+CEREG?
+CEREG: 0,2
OK
AT+CGREG?
+CGREG: 0,2
OK
Reg-Status: 2
AT+CGNSINF
+CGNSINF: 1,0,,,,,,,0,,,,,,23,,,,,,
OK
AT+CIPSHUT
SHUT OK
AT+CGATT=0
OK
AT+SAPBR=3,1,"Contype","GPRS"
OK
AT+SAPBR=3,1,"APN","internet"
OK
AT+CGDCONT=1,"IP","internet"
OK
AT+CGACT=1,1
+CME ERROR: no network service
AT+SAPBR=1,1
+CME ERROR: operation not allowed
AT+SAPBR=2,1
+SAPBR: 1,3,"0.0.0.0"
OK
AT+CGATT=1
AT+CEREG?
```

Can anyone shed some light on this?

SRGDamia1, 2020-05-26:
If you're always getting the registration status of 2, it's probably a problem with your SIM or your power supply.  The other functions (like setting APN) won't work correctly until after the network accepts registration.  The "waitforNetwork()" function does just that, it queries the module for network registration status and doesn't complete until the status is either 1=registered home or 5=registered roaming, or it times out

In my testing with a SIM7000A, I found that when powered at ~500mA the module appeared to be fully functional but refused to register on the network - always giving a registration status of 2-searching.  Bumping the power supply to a steady 2A+ solved everything.  You might also want to check your signal strength/CSQ (getSignalQuality).  Sometimes the network won't accept registration from a module that has a very poor connection or it will take a really long time.  If you're CSQ is less than ~10, you might want to try a different location or a different antenna or just adjusting the positioning of your antenna.  The uF.L antenna connections are pretty fragile.  Also, do check you account and make sure your SIM is valid.

Elmi77, 2020-05-26:
Thanks for the feedback. The signal quality is 99 and the board is power-backed by a 3000 mAh battery, so all this should not be a problem.

When I try the SIM card together with an Android-device, I have to allow roaming in order to let mobile internet work. May be this is a problem? Do I need to enable something similaro for the SIM7000G?

Elmi77, 2020-05-27:
Solved: I had to use setPreferredMode(3), all other modes did not work

Elmi77, 2020-05-27:
OK, I have to come back to my problem: It has nothing to do with setPreferredMode(). Now some time later I still stumble upon the same problem. One thing I noticed: when it worked, It gave me a signal quality in range 23..26. Now that it is not working it gives me constantly a signal quality of 99.

SRGDamia1, 2020-05-28:
That's just an invalid value - basically zero signal.

If you've played with the settings for the band or the preferred mode, you might have accidentally selected a band or mode that doesn't match with your SIM card carrier - in which case you'll get no signal.

If you're sure you're on the right band and there is signal available, you probably need to fix your antenna (or possibly the power supply).  As I mentioned, those little U.FL antenna connections are really fragile - I've busted a number of them.

Elmi77, 2020-06-14:
OK, I have a new antenna now but still the same problem. Now I see two possibilities:

- something is not turned on in GSM module. I already do a modem.sendAT("+CFUN=1") right afer modem.init() - but is this really the correct command to enable radio functions?

- the board / SIM700G module has a damage :-/

SRGDamia1, 2020-06-15:
AT+CFUN=1 *should* put it at full functionality.

Have tried the AT+CBAND or AT+CBANDCFG commands to check that they're somehow not set incorrectly?

Have you ever gotten this board/SIM to work?

Elmi77, 2020-06-15:
> Have you ever gotten this board/SIM to work?

The SIM works properly in an Android tablet. And with my board it has worked - in some very rare, random case (what lets me think there is a problem with the hardware).

closed 2020-06-17 by Elmi77