vshymanskyy/TinyGSM #402: Check if SIM-card is available?

Elmi77, 2020-05-27:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[X] Question or request for help

### What are you working with?

Modem: SIM7000G 
Main processor board: LilyGO SIM7000G 
TinyGSM version: 0.10.5 via ArduinoIDE 
Code: 

Is there a way/function to recognise if a SIM-card is inserted? Restarting/initialising of modem takes quite a long time before it fails when there is no SIM card available, so I'd prefer to check for this first.

SRGDamia1, 2020-05-28:
Try the getSimStatus() function. 1=ready.

closed 2020-05-28 by SRGDamia1

Elmi77, 2020-05-28:
Works - Thanks!