vshymanskyy/TinyGSM #403: Permission Denied

edwardruthven, 2020-05-28:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: 
Main processor board: 
TinyGSM version: 
Code: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### Debug and AT command log

edwardruthven, 2020-05-28:
I get the following:-

Arduino: 1.8.12 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Users\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0043 -ide-version=10812 -build-path C:\Users\edwar\AppData\Local\Temp\arduino_build_400502 -warnings=none -build-cache C:\Users\edwar\AppData\Local\Temp\arduino_cache_245421 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose D:\Users\Documents\Arduino\libraries\TinyGSM\examples\AllFunctions\AllFunctions.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Users\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0043 -ide-version=10812 -build-path C:\Users\edwar\AppData\Local\Temp\arduino_build_400502 -warnings=none -build-cache C:\Users\edwar\AppData\Local\Temp\arduino_cache_245421 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose D:\Users\Documents\Arduino\libraries\TinyGSM\examples\AllFunctions\AllFunctions.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\edwar\\AppData\\Local\\Temp\\arduino_build_400502\\sketch\\AllFunctions.ino.cpp" -o nul
Alternatives for TinyGsmClient.h: [TinyGSM@0.10.5]
ResolveLibrary(TinyGsmClient.h)
  -> candidates: [TinyGSM@0.10.5]
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-ID:\\Users\\Documents\\Arduino\\libraries\\TinyGSM\\src" "C:\\Users\\edwar\\AppData\\Local\\Temp\\arduino_build_400502\\sketch\\AllFunctions.ino.cpp" -o nul
Alternatives for TinyGsmClient.h: [TinyGSM@0.10.5]
ResolveLibrary(TinyGsmClient.h)
  -> candidates: [TinyGSM@0.10.5]
AllFunctions:92:10: fatal error: D:\Users\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h: Permission denied

 #include <TinyGsmClient.h>

          ^~~~~~~~~~~~~~~~~

compilation terminated.

Using library TinyGSM at version 0.10.5 in folder: D:\Users\Documents\Arduino\libraries\TinyGSM 
exit status 1
D:\Users\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h: Permission denied

SRGDamia1, 2020-05-28:
That's a problem with your computer, not the library.

https://forum.arduino.cc/index.php?topic=497165.0

Mr-HaleYa, 2020-07-17:
@edwardruthven 
this needs to be closed as we can not fix it it is your computer

closed 2020-07-20 by SRGDamia1