vshymanskyy/TinyGSM #404: GET request -> sometimes no response

maromme, 2020-05-31:
Hi,
I'm using a TTGO T-call with SIM800L to send sensor data to my own webserver. Occasionally (it's not possibel to reproduce the failor) the data is send, but I've got an empty/no response. 

First I thought it's been after changing the resource name on the server (producing 404 failors for testing), but that wasn't the case. 

Power or the board aren't the problem. Changed the board and using bench supply.

Someone a clue whats's wrong?

That's my settings:
```
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb
#include <TinyGsmClient.h>
TinyGsmClient client(modem);
```
That's my code:
```
 SerialAT.begin(9600, SERIAL_8N1, MODEM_RX, MODEM_TX);
  delay(1000);

  Serial.print("Initializing modem... ");
  if (!modem.init()) {
    Serial.println(" [fail]");
    return (false);
  } else Serial.println(" [OK]");

  Serial.print("Waiting for network...");
  if (!modem.waitForNetwork(600000L)) {  
    Serial.println(" [fail]");
    return (false);
  } else {
    Serial.println(" [OK]");
  }

  Serial.print(F("Connecting to APN: "));
  Serial.print(apn);
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    Serial.println(" fail");
    delay(10000);
    return (false);
  }
  Serial.println(" [OK]");

Serial.print("Connecting to ");
  Serial.print(server);
  if (!client.connect(server, port)) {
    Serial.println(" fail");
    delay(10000);
    return (false);
  }
  Serial.println(" [OK]");

  // Make a HTTP GET request:
  Serial.println("Performing HTTP GET request...");
  client.print(String("GET ") + url + " HTTP/1.1\r\n");
  client.print(String("Host: ") + server + "\r\n");
  client.print("Connection: close\r\n\r\n");
  client.println();

 unsigned long timeout = millis();
  while (client.connected() && millis() - timeout < 10000L) {
    // Print available data
    while (client.available()) {
      char c = client.read();
      Serial.print(c);
      timeout = millis();
    }
  }

client.stop();
  Serial.println(F("Server disconnected"));

  modem.gprsDisconnect();
  Serial.println(F("GPRS disconnected"));
```

maromme, 2020-05-31:
I continued some testing, I still couldn't reproduce the problem, but got two logs, one with and one without the problem.

The first one without the server response, the second one with a response. 
[not suc.txt](https://github.com/vshymanskyy/TinyGSM/files/4708291/not.suc.txt)
[suc.txt](https://github.com/vshymanskyy/TinyGSM/files/4708290/suc.txt)

Found differences on lines 
147 `+SAPBR: 1,1,"10.45.80.18` vs. `+SAPBR: 1,1,"10.151.148.233"`
197 same different IPs
255 after 2 times `+CIPSTATUS: 0,0,"TCP","137.74.234.116","80","CONNECTED"` in the succesfull one there is an response. The not succesfull one tries 20 times without any response.

Is there an timeout or something like that? Any possibility to "catch" that failor?

maromme, 2020-06-01:
I increased the timeout to 60 000 ms but didn't help. If it's not working within the 10 000 ms it woun't afterwards.

But running my script a second (or thrid) time - when there wasn't any response - is always (as far as i could test it) succeding. I'm glad it's my own php script I'm calling and I can handel the dobble requests there.

Still I think there should be another solution. And the example https://github.com/vshymanskyy/TinyGSM/blob/master/examples/WebClient/WebClient.ino at line 216 - 223 is not sufficently handeling/waiting for server responses.

mrzrvr, 2022-08-06:
Hi Maromme,
i found the same issue using TinyGSM on a TTGO-Call-Sim800L. 
About a month ago I made a serie of successfull tests (get requests over HTTPS with some sensors data, each 15 minute, for about a day) whith a sketch based on [this ](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_HttpsClient) example.
But when a I went on with the project, about a week ago, I encountered the same problem you mentioned; following yours considerations, I also tried to change DNS values in tinygsm library, but this didn't change the things.
So, i would try to approach the problem as you wrote (running a second time the script when non working); can you explain with more details how did you do?

kaustubh-ke, 2023-01-13:
hey @maromme 
I am  also having same issue no response from the server but the client.connect(server,port ) is true,  and Http post was working with wifi on my own aws server, but now using gprs its not working not even getting any response from the server , if u found the solution please share .. it would be very helpful for me . Thankyou

anh-tuan, 2023-08-11:
> hey @maromme I am also having same issue no response from the server but the client.connect(server,port ) is true, and Http GET/POST was working with wifi on my own aws server, but now using gprs its not working not even getting any response from the server , if u found the solution please share .. it would be very helpful for me . Thankyou

me too, I'm using A7600C-L1, this is my code

`
SerialMon.println("-----------------------------");
SerialMon.println("START using GPRS to send Data");
SerialMon.println("-----------------------------");

SerialMon.print("[GPRS] *** Connecting to server: ");
SerialMon.println(server);

 if (!client.connect(server, port)) {
    SerialMon.println("*** Connect to server FAILED!");
    delay(3000);
    return;
}
SerialMon.println("*** Connect to server OK!");    

SerialMon.println("*** Performing HTTP POST request...");

client.print(String("POST ") + resource + " HTTP/1.1\r\n");
client.print(String("Host: ") + server + "\r\n");
client.println("Connection: close");
client.println("Content-Type: application/json");
client.print("Content-Length: ");
client.println(postData.length());
client.println();
client.println(postData);

//Get response
SerialMon.println("----------------------");
SerialMon.println("START Server response");
SerialMon.println("----------------------");

uint32_t timeout = millis();
while (millis() - timeout < 10000L) {
//while (client.connected() && millis() - timeout < 10000L) {
  // Print available data
  while (client.available()) {
    char c = client.read();
    SerialMon.print(c);
    timeout = millis();
  }
}

// Close client and disconnect
client.stop();
SerialMon.println("-------------------");
SerialMon.println("END Server response");
SerialMon.println("Server disconnected");
`

and here is my response

`
-----------------------------
START using GPRS to send Data
-----------------------------
*** Connecting to server: server.com

AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
AT+CIPRXGET=1

OK
AT+CIPOPEN=0,"TCP","server.com",80

OK

+CIPOPEN: 0,0
*** Connect to server OK!
*** Performing HTTP POST request...
AT+CIPSEND=0,32

>POST /endpoint.php HTTP/1.1

OK

+CIPSEND: 0,32,32
AT+CIPSEND=0,20

>Host: server.com

OK

+CIPSEND: 0,20,20
AT+CIPSEND=0,17

>Connection: close
OK

+CIPSEND: 0,17,17
AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2
AT+CIPSEND=0,30

>Content-Type: application/json
OK

+CIPSEND: 0,30,30
AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2
AT+CIPSEND=0,16

>Content-Length: 
OK

+CIPSEND: 0,16,16
AT+CIPSEND=0,3

>160
OK

+CIPSEND: 0,3,3
AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2
AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2
AT+CIPSEND=0,160

>{"key1":"value1","key2":"value2","key3":"value3","key4":"value4","key5":"value5"}
OK

+CIPSEND: 0,160,160
AT+CIPSEND=0,2

>

OK

+CIPSEND: 0,2,2
----------------------
START Server response
----------------------
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK

+CIPRXGET: 1,0

+IPCLOSE: 0,1
[40136] ### Closed:  0
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?
+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+IP ERROR: 11

ERROR
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
-------------------
END Server response
Server disconnected
`