vshymanskyy/TinyGSM #406: Arduino: use SIM800/A6 GSM with custom SSL library

tuxmartin, 2020-06-05:
Hi,
I would like to use GSM (GPRS) network in my project.

Plain unencrypted HTTP/MQTT is not problem, but I want to use SSL.

I found some examples of HTTPS with SIM800, but no MQTTS.

Is possible to use SIM800/A6 for "raw tcp" communication with remote server and create HTTPS/MQTTS socket/layer in arduino?

I do not mean 8bit MCU, like ATmega328. I plan to use ESP32 or STM32 which has enough power and memory for SSL.

I found these two SSL libraries for arduino: https://github.com/OPEnSLab-OSU/SSLClient and https://github.com/arduino-libraries/ArduinoBearSSL

Is it possible? I cannot find anything about it.

Thanks for help.

SRGDamia1, 2020-06-08:
Not all versions or firmwares of the SIM800 support SSL.  If the version you have is capable of using SSL, you should be able to do it with this library.  If not, you won't.  You can check if your board supports SSL with the AT command ```AT+CIPSSL=?```.  You will have to set the certificate outside of this library, or use the certificate built into the SIM800's firmware.  I believe there's an example in the examples/more section.

To my knowledge the A6 doesn't support SSL at all.

SRGDamia1, 2020-06-08:
Oh.  Sorry.  I misread what you wanted to do.  

I suppose if one of those other libraries handled the socket layer, TinyGSM could provide the lower TCP layer.  TinyGSM is using all transport-layer commands with the modules.  The only session-layer commands are those to open the TCP over SSL for modules that easily support it using built-in certificates.  (For the SIM800, if it's supported, the SSL socket only takes one more like to open than the standard socket.)  TinyGSM does NOT use any presentation or application layer commands, even if the modules support them.

Have you tried it yet?  What was the result?

tuxmartin, 2020-06-08:
I would like to use TinyGSM and modem A6/SIM800 only for "raw tcp socket". 
SSL encapsulation would be handled in arduino (esp32/stm32) using software library.
So modem and TinyGSM would only see "raw data" in socket and have no information about SSL.

I do not want to use SSL function of modem SIM800/A6. Complete SSL will be implemented in Arduino code.
It will solve problem with supported TLS version (AFAIK: SIM800 support only TLS 1.0 - that is problem with AWS IoT).

Is it possible? And how?

better-adapted, 2020-08-02:
@tuxmartin 
see my fork for simple securemqtt working on Auzure tls1.2

Inkomidwastaken, 2020-12-29:
@tuxmartin
any updates?
I'm actually trying to do the exact same thing right now, but haven't been very succesfull.

tuxmartin, 2020-12-29:
@Inkomidwastaken No. I didn't have time for that :-(

genotix, 2021-01-04:
I’m also looking into getting the SIM800 to work with AWS IoT.
Would be great if we could be able to use an existing TLS library over the TinyGSM library or PPPoS working with the SIM800.

FStefanni, 2021-01-19:
Hi,

just to contribute...

I have the same issue of TLS 1.2 with SIM800L right now.
I have tried ArduinoBearSSL and SSLClient, but without success.
I must admit it has been a quick try, so maybe I have done something wrong.
Actually, thanks to the Client interface, nesting of functionalities should be possible... but maybe there is a corner case which inhibits this particular use case.

Regards.

horellana, 2021-01-27:
Hi, TinyGsmClient class can be used with SSLClient, it seems to work ok so far, i was able of connecting to an AWS API Gateway. I tried it in a esp32 with a sim800l.

tuxmartin, 2021-01-27:
@horellana 

It sounds interesting!

Can you please send an example of the code?

volant0098, 2021-09-14:
> 
> 
> Hi, TinyGsmClient class can be used with SSLClient, it seems to work ok so far, i was able of connecting to an AWS API Gateway. I tried it in a esp32 with a sim800l.

Hi, could you please send us the code?

genotix, 2021-09-14:
> Hi,
>
> just to contribute...
>
> I have the same issue of TLS 1.2 with SIM800L right now.
> I have tried ArduinoBearSSL and SSLClient, but without success.
> I must admit it has been a quick try, so maybe I [...]

SSLClient with BearSSL works with the SIM800L combined with PubSub.

I haven’t got the communication over 1024 kB per message working though so stay under that.

Also make sure to use larger antenna’s for the SIM800L because connection easily breaks and you’ll notice immediately on SSL breaking.

pperzyna, 2021-11-29:
@genotix Do you have piece of code to share?

SurajitGithub1234, 2022-06-21:
> Hi, TinyGsmClient class can be used with SSLClient, it seems to work ok so far, i was able of connecting to an AWS API Gateway. I tried it in a esp32 with a sim800l

Can you please share your code?

genotix, 2022-06-21:
I have, it is in my Git.

horellana, 2022-06-30:
https://github.com/OPEnSLab-OSU/SSLClient/blob/master/examples/EthernetAWSIoT/EthernetAWSIoT.ino

it is pretty much this example, but instead EthernetClient you have to use TinyGsmClient.

talhaahmii, 2022-09-21:
I have spent two days sending data to platform made on aws iot with tls 1.2 with sim800l still cant get succeeded with htts or mqtt

jelelectronica, 2025-05-03:
ok, thank you, I have already managed to connect with sim800l and AWS IoT core.

OmarAskar421, 2025-09-29:
> ok, thank you, I have already managed to connect with sim800l and AWS IoT core.
can you please share a code 
any help will mean the world to me