vshymanskyy/TinyGSM #410: How to use HTTPS rest api in sim800l?

biswajitjei02, 2020-06-13:
here is the example of HTTPS  GET /POST  as this data base  shows please check
https://firebase.google.com/docs/reference/rest/database.

how to ue https request using Tiny GSM library and this data base .

SRGDamia1, 2020-06-15:
Based on #407 it seems someone else was able to help you out.

In general, though, this isn't the place to ask someone to write out code for you.

closed 2020-06-15 by SRGDamia1

star297, 2021-02-20:
[image]

How to implement the RESTful API as shown on your web RREADME.md?
Are there any examples for using GET, POST, PUT and DELETE as used with Firebase?
Can this TinyGSM library actually handle the REST API functions?