vshymanskyy/TinyGSM #413: Check if SIM is allowed to send SMS?

Elmi77, 2020-06-21:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ X] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM700G 
Main processor board: ESP32 
TinyGSM version: 0.10.5 (latest in ArduinoIDE) 
Code: 

### Scenario, steps to reproduce

I try to send SMS via function modem.sendSMS()

### Expected result

SM is sent or the function returns more or less immediately.

### Actual result

I'm using a SIM card which can't send any SMS at the moment. The problem is, sendSMS() returns only after a long timeout with this SIM card. So my question: is there a possibility to check if SMS can be sent in advance so that I don't need to call this function which blocks for that long time? Or is there a possibility to let this function return faster?

### Debug and AT command log

SRGDamia1, 2020-06-21:
Hm.  I'm not sure if there are any AT commands to check this, but I haven't implemented anything for this in the library.

Elmi77, 2020-06-22:
When I turn on debugging, TinyGSM shows 

Unhandles: +CMS Error: 500

Sounds a bit like the sendSMS() command does not catch this case properly but aways waits for success until timeout occurs? Is there a possibility to workaround this?

SRGDamia1, 2020-06-22:
Do you get the CMS error with debugging on or off?

I added a catch for it when the debugging is on.

Elmi77, 2020-06-28:
I get it when the debug output is turned on, elsewhere nothign can be seen on the console.

SRGDamia1, 2020-06-29:
I mean, do you get a very long wait:?

Elmi77, 2020-06-30:
Yes, it needs about 30..40 seconds

SRGDamia1, 2020-08-07:
Are you still having this issue?  Can you post your AT log?

ohbhatt, 2024-09-17:
+1: I am having the same issue, but the device does indeed send an SMS.
Can we just modify the library to not wait for any feedback or just reduce the timeout?