vshymanskyy/TinyGSM #415: Missing example on long term connections

Zambiorix, 2020-06-24:
Hi,

Unless I'm mistaken I'm missing an example of how to use the library in a long term connected situation.

Most examples are: initialise, wait for network, connect gprs, connect client, send, receive, disconnect client, disconnect gprs and then loop indefinitely.

I'm totaly new to using gsm communication, so I'm struggling with the correct approach to use the library in case of an always-on situation. I understand this a place where issues should be reported. But I have to clue where to turn to, I cannot find a forum for TinyGSM users.

I am working on an application with Sara R410 LTE cat m1 and I need to send a small packet every 60 sec.

How do I keep going?

Now I do this (in a separate task):

1. Initialize (restart etc)
2. wait for network
3. connect gprs
4. connect client
5. loop (waiting for packets to send, checking on connection states of network, gprs & client)

In my loop, I tear down the client and/or gprs connection and reconnect if needed (or wait for the network again if it's gone)

Is this the correct approach?

Or is it better to connect/disconnect my client for each packet to send?
Same for grps connection, do I keep it alive or do I need to connect and disconnect for each packet to send?

And what if I need to send every 10 sec? or every 2 hours? Is the approach different in this situation?

If there is a longer time between sending packets, is there a way to put the device in sleep while still maintaining a connection? Of do I need to cleanup client and gprs connections?

Thank you for your help, much appreciated!

Cheers
Gerd

SRGDamia1, 2020-06-29:
There is a gitter discussion chat, but it's very lightly used:  https://gitter.im/tinygsm/Lobby

I haven't well tested options for keeping the connection open long term.  In my applications everything has been battery powered and data is only ever going out so I set the module down to lowest power disconnected state or cut its power between connections (usually once every 5-15 minutes).  That's easy and saves power, but it also means you miss all incoming data while your not powered and you have to go through the whole network registration with each power up.  With the SARA R410 if you're savvy you can use the eDRX (Extended Discontinuous Reception) or PSM (power save mode) features save power while staying connected or semi-connected to the network and able to respond to incoming requests.   None of that's implemented in this library, though.  If you want to implement it, most of the documentation on how to use both eDRX and PSM is in the system integration manual for the R4 series (get it on the u-blox website).  Look in section 1.4 on operating modes.  You use the commands ```AT+CPSMS``` and ```AT+CEDRXS``` to configure your timers.

Beyond power saving, there's no need to disconnect between outgoing packets.  If you want an instant response to incoming and outgoing data you should be able to stay fully connected all the time.

Zambiorix, 2020-07-08:
Thank you for the detailed feedback! I have already started writing my own custom driver for Sara R410 based on TinyGSM .. 

eDRX looks exactely what I need! 

Cheers

Mr-HaleYa, 2020-07-17:
@Zambiorix Close this now that you solved it

closed 2020-07-28 by Zambiorix