vshymanskyy/TinyGSM #416: Can't choose/see kind of network

SebastianWantia, 2020-07-02:
I can't choose kind of network. How can I choose or see how the module is connected with the network, is it only GPRS?

SRGDamia1, 2020-07-07:
So.. are you trying to see how you're connected to the network because you really understand the alphabet soup of connection protocols and want to know the details, or are you just worried about something 4G working on a library with functions named "GPRS"?  

If the module connects to the network with EPS (4G/LTE), then this library uses that.  If not, or if the module is in "fallback" it uses GPRS.

There's currently no function to check how you're connected to the network.  (ie, GPRS vs EPS)  It's up to the module and the network to determine how it can best connect.  TinyGSM queries each module for connection status based on the specs for the module.  Generally, GPRS is used by 2G/3G modules and TinyGSM checks for connection using `AT+CGREG?`.  4G/LTE/LTE-M/NB IoT modules use EPS and the `AT+CEREG?` command.

Mr-HaleYa, 2020-07-17:
you can set the preferred connection type with
 setNetworkMode() ( 2 Automatic , 13 GSM only , 38 LTE only , 51 GSM and LTE only )
 and setPreferredMode() ( 1 CAT-M , 2 NB-Iot , 3 CAT-M and NB-IoT)

or with AT commands 
 http://www.microchip.ua/simcom/LTE/SIM7000/SIM7000%20Series_AT%20Command%20Manual_V1.05.pdf